open62541pp/open62541pp #15: access level  control

vpashka, 2023-01-03:
Hello.

You have a great project, but there was a small question.
How can I set access level? For example when adding variables (addVariable).

By default 
`attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;`

At creation I cannot specify them in any way.
`addVariable(const NodeId& id, std::string_view browseName, Type type)`

Can I somehow set the access level after creation?

lukasberbuer, 2023-01-03:
Hi Pavel,

that's a good point. As I understand, it's not possible to modify the node attributes after creation.
So we need to pass the attributes to the `addVariable` method. I see following options:

1. Pass `accessLevel` and other common attributes as function arguments.. this might result in very long function signatures over time.
2. Pass the native `UA_VariableAttributes` [(doc)](https://www.open62541.org/doc/1.1/types.html#variableattributes) struct as a function parameter. `UA_VariableAttributes` has an additional flag `specifiedAttributes` flag. As far as I understand it, if you set the `accessLevel` you also have to set update the flag `specifiedAttributes |= UA_NODEATTRIBUTESMASK_ACCESSLEVEL`. This makes it quite complicated to work with.
3. Wrap the `UA_VariableAttributes` struct in a class with getters/setters and pass it to the method.

I don't have a clear favorit at the moment. What's your suggestion? If we decide on an option, I will propose a solution within this week.

vpashka, 2023-01-03:
Thanks for the quick response.
I think option 3 is a good idea and will be quite convenient. And if possible, it's worth providing the most commonly used defaults.

lukasberbuer, 2023-01-05:
Hi Pavel,

I was wrong, open62541 provides the function `UA_Server_writeAccessLevel` to set the access level attribute of node. Therefore you can set the access level after creation.
I added following methods to the `Node` class https://github.com/open62541pp/open62541pp/commit/8ecc0fca68d584cad1df3465e52aac8bc2b253a4:
- `Node::getAccessLevel`
- `Node::setAccessLevel`, e.g.: `node.setAccessLevel(UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE)`

Please check out the [latest doc of the `Node` class](https://open62541pp.github.io/open62541pp/classopcua_1_1Node.html).
I also changed the default access level to read-only `UA_ACCESSLEVELMASK_READ`.

Does that solve your issue?

vpashka, 2023-01-05:
Yes it's good. I'm quite happy with this option. Thank you so much!

lukasberbuer, 2023-01-05:
Great 👍 

Btw, you could also have used the native `UA_Server_writeAccessLevel` function with the native objects (available via the `handle()` methods):

```cpp
opcua::Server server;
auto node = server.getObjectsNode().addVariable({"test123", 1}, "test123", opcua::Type::Int32);

UA_Server* serverNative = server.handle();
UA_NodeId* nodeIdNative = node.getNodeId().handle();

const auto status = UA_Server_writeAccessLevel(
    serverNative,
    nodeIdNative,
    UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE
);
```

Please let me know if you have other question or issues. I'm currently ramping up this project and looking for feedback.

closed 2023-01-05 by lukasberbuer