open62541pp/open62541pp #20: Plans for the Client-implementation?

ErikTempelaarVO, 2023-04-05:
Hi,

I'm planning to start development using open62541 and I came across your library looking for a neat C++ wrapper.
I want to build a client and that is on your TODO-list.
Do you have any plans for it already? Some code in a draft-branch perhaps? Are you open to contributions?

lukasberbuer, 2023-04-06:
Hi @ErikTempelaarVO,

thanks for bringing that up. I made some drafts but didn't come up with I solution I was happy with. The main reason was, that I want to stick with the `Node` class. The `Node` class itself carries the `NodeId` and a `std::shared_ptr` to the underlying `Server` connection (to maybe extend the lifetime of the connection).
We need to introduce a `Client` class for sure. How can we adapt the `Node` class to handle both server and client connections? I was thinking of following solutions:
1. `NodeBase` base class with `NodeServer` and `NodeClient` implementations. Virtual functions will add some overhead. And we loose the beauty of returning `Node`s by value.
2. Use a `std::variant` to store either a server or client connection in the `Node`. Then we need dispatching in every `Node` method, keep the same interface but have no (?) performance penalties.
3. Introduce a `NodeClient` with the same or similar interface.

I saw that @duca started working on a client on [a fork](https://github.com/duca/open62541pp). It seems like he did choose option 3. Maybe we can share ideas and collaborate on this feature. I'll try to invite him to this discussion :)

lukasberbuer, 2023-04-06:
Draft of how option 2 might look like: [Link to godbolt](https://godbolt.org/z/78P7T54rM)
I like the approach with the free functions for both server and client - which might be useful for some use-cases.

```cpp
#include <iostream>
#include <utility>
#include <variant>

template<class... Ts> struct overload : Ts... { using Ts::operator()...; };
template<class... Ts> overload(Ts...) -> overload<Ts...>;

class Server {};
class Client {};
class NodeId {};
class Variant {};

// free functions for both Server and Client in separate namespace
namespace services {

void writeValue(Server& server, const NodeId& id, const Variant& var) {
    std::cout << "Server writeValue\n";
}

void writeValue(Client& client, const NodeId& id, const Variant& var) {
    std::cout << "Client writeValue\n";
}

}  // namespace services

class Node {
public:
    Node(std::variant<Server, Client> connection)
        : connection_(std::move(connection)) {}

    void writeValue(const Variant& var) {
        // std::visit(overload{
        //     [&](Server& c) { services::writeValue(c, id_, var); },
        //     [&](Client& c) { services::writeValue(c, id_, var); },
        // }, connection_);
        std::visit([&](auto& c) { services::writeValue(c, id_, var); }, connection_);
    }

private:
    NodeId id_;
    std::variant<Server, Client> connection_;
};

int main() {
    Server server;
    Client client;

    Node node(server);
    // Node node(client);
    node.writeValue({});
}
```

What option do you prefer?

duca, 2023-04-06:
> [CAUTION: This email originated from outside Oceaneering.]

Thanks for bringing me in Lukas,
I would very much like to collaborate on this. I am still muddling through the implementation trying to search for a solution I like, but I settled with it.

I was thinking on creating a common interface that a client and a server would adhere to and re-use the original Node. Free functions sounds fine and also looks like what Unified Automation went with, but I think they use global client/server instances.

The original plan was submit a PR once I had a usable client (right now I am lacking subscriptions and all the async variations).

Regards,
--

Eduardo Martins Lopes

Sr. Software Engineer
Technology Brazil
Direct: +55 21 2729 8647

***@***.******@***.***>

[cid:9652dddf-91f7-4347-ba46-1800b5750a02]​<http://www.oceaneering.com/>

Av. Rio Branco, 123 14º andar | Rio de Janeiro - RJ | 20040-005 Brasil| | www.oceaneering.com<http://www.oceaneering.com>

This email is confidential, may be privileged, and should be read or retained only by the intended recipient.

If you have received this email in error, please immediately notify me, delete it from your system and do not retain any copies.

Thank you for your cooperation.

lukasberbuer, 2023-04-06:
Great to have you on board @duca. I'm impressed how much you already implemented. 👍

I just though about a fourth solution with a templated `Node` class with two explicit instantiations `Node<Server>` and `Node<Client>`. We can also combine this solution with free functions. Should be the most performant solution and no memory size overhead of the `Node` class. With the explicit instantiations, we can keep the implementations in .cpp files. I think it's my favorite at the moment 🤔 

# Draft of option 4

[Link to godbolt](https://godbolt.org/z/Yo8ha7cbY)

```cpp
#include <iostream>
#include <utility>

class Server {};
class Client {};
class NodeId {};
class Variant {};

template <typename Connection = Server>
class Node {
public:
    Node(Connection connection)
        : connection_(std::move(connection)) {}

    void writeValue(const Variant& var);

private:
    NodeId id_;
    Connection connection_;
};

// explicit template instantiation
template class Node<Server>;
template class Node<Client>;

// implementation in cpp file

template <>
void Node<Server>::writeValue(const Variant& var) {
    std::cout << "Server writeValue\n";
}

template <>
void Node<Client>::writeValue(const Variant& var) {
    std::cout << "Client writeValue\n";
}

int main() {
    Server server;
    Client client;

    Node node(server);
    // Node node(client);
    node.writeValue({});
}
```

duca, 2023-04-07:
Coincidently, #4 was my ideal alternative when I first thought about generalizing Node class. One thing, though, how feasible would it be to remove Client/Server comms from Node and rely entirely on the free functions?

The reason I am asking this is that when I was implementing NodeClient, I noticed that most of the methods could easily be moved to my Client class. NodeClient was simplifying some things a bit, but not much.

It would make free functions more appealing, and simplify Node implementation, Node could embed its UA_ReferenceDescription (https://github.com/duca/open62541pp/blob/12d61f7a2472958e6de7cc62a4c0106e0d73faf7/include/open62541pp/ReferenceDescription.h), and make the Browser implementation also cleaner.
[https://opengraph.githubassets.com/bfc384ad006e7917b939e98ee70f9e5541c8857a5db052f223b3e26ad571754b/duca/open62541pp]<https://github.com/duca/open62541pp/blob/12d61f7a2472958e6de7cc62a4c0106e0d73faf7/include/open62541pp/ReferenceDescription.h>
open62541pp/ReferenceDescription.h at 12d61f7a2472958e6de7cc62a4c0106e0d73faf7 · duca/open62541pp<https://github.com/duca/open62541pp/blob/12d61f7a2472958e6de7cc62a4c0106e0d73faf7/include/open62541pp/ReferenceDescription.h>
C++ wrapper for open62541 (OPC UA library) licensed under Mozilla Public License v2.0 - open62541pp/ReferenceDescription.h at 12d61f7a2472958e6de7cc62a4c0106e0d73faf7 · duca/open62541pp
github.com

--

Eduardo Martins Lopes

Sr. Software Engineer
Technology Brazil
Direct: +55 21 2729 8647

***@***.******@***.***>

[cid:9467ee7e-4ec5-4bd5-9e9c-e22581134b52]​<http://www.oceaneering.com/>

Av. Rio Branco, 123 14º andar | Rio de Janeiro - RJ | 20040-005 Brasil| | www.oceaneering.com<http://www.oceaneering.com>

This email is confidential, may be privileged, and should be read or retained only by the intended recipient.

If you have received this email in error, please immediately notify me, delete it from your system and do not retain any copies.

Thank you for your cooperation.

lukasberbuer, 2023-04-07:
@duca: What do you mean with removing the Client/Server comms from `Node`? Should the Node class just embed the `NodeId`? What would be the benefit of using the `ReferenceDescription` instead of the `NodeId` for the Node class?

I totally agree on the argument for free functions. It's not much work for us to expose those free functions and keep the `Node` class for convenience just as another interface. In the end, the `Node` class avoids to specify the first two parameters of the free functions again and again:

```cpp
// free functions for both Server and Client in separate namespace
namespace services {

void writeValue(Server& server, const NodeId& id, const Variant& var);
void writeValue(Client& client, const NodeId& id, const Variant& var);

}  // namespace services

// Node class for convenience using the free functions

template <typename Connection = Server>
class Node {
public:
    // ...
    void writeValue(const Variant& var) {
        services::writeValue(connection_, id_, var);  //
    }
    // ...
private:
    NodeId id_;
    Connection connection_
}
```

I don't want to ditch the `Node` class. OPC UA is a complex beast and especially the API of the [Python library `asyncua`](https://github.com/FreeOpcUa/opcua-asyncio) was a big enabler for me. I could build the first OPC UA server/client within minutes. I think the idea of linked `Node`s having properties and methods is easy to understand and use. This is the main reason why I started this project - minimize the hurdle to start using OPC UA.

As you said @duca, it's sometimes hard to choose if a method belongs to the `Server`/`Client` or a `Node`. Let us try to go with the approach with free functions and try make the `Node` class a convenience class. We don't have to assign all functionality to either `Server`/`Client` or a `Node. So more sophisticated functionalities may stay just live as free functions.

## I propose following steps
1. Move the `Node` methods to free functions in a separate namespace and use those functions in the `Node` implementation
   - Naming of the namespace? I would propose `service` or `services`. The OPC UA specification uses the term [Services](https://reference.opcfoundation.org/Core/Part4/v104/docs/) for the different service sets like NodeManagement (add/delete nodes, add references, ...), Attribute (read/write node attributes), Method, Discovery, ...
   - File structure:
     ```
     services/Attribute.h/cpp
     services/NodeManagement.h/cpp
     ...
     ```
2. Implement a basic `Client` class
3. Implement option 4, the templated `Node` class with both `Node<Server>` and `Node<Client>`
4. Implement free functions for the `Client`

Any remarks or suggestions @duca, @ErikTempelaarVO?

I will start working on step 1 in a feature branch and invite you for the review.
@duca: Can you provide a basic `Client` class implementation similar to the current `Server` class?

duca, 2023-04-07:
> @duca: What do you mean with removing the Client/Server comms from `Node`? Should the Node class just embed the `NodeId`? What would be the benefit of using the `ReferenceDescription` instead of the `NodeId` for the Node class?
> 
> I totally agree on the argument for free functions. It's not much work for us to expose those free functions and keep the `Node` class for convenience just as another interface. In the end, the `Node` class avoids to specify the first two parameters of the free functions again and again:
> 
> ```c++
> // free functions for both Server and Client in separate namespace
> namespace services {
> 
> void writeValue(Server& server, const NodeId& id, const Variant& var);
> void writeValue(Client& client, const NodeId& id, const Variant& var);
> 
> }  // namespace services
> 
> // Node class for convenience using the free functions
> 
> template <typename Connection = Server>
> class Node {
> public:
>     // ...
>     void writeValue(const Variant& var) {
>         services::writeValue(connection_, id_, var);  //
>     }
>     // ...
> private:
>     NodeId id_;
>     Connection connection_
> }
> ```
> 
> I don't want to ditch the `Node` class. OPC UA is a complex beast and especially the API of the [Python library `asyncua`](https://github.com/FreeOpcUa/opcua-asyncio) was a big enabler for me. I could build the first OPC UA server/client within minutes. I think the idea of linked `Node`s having properties and methods is easy to understand and use. This is the main reason why I started this project - minimize the hurdle to start using OPC UA.
> 
> As you said @duca, it's sometimes hard to choose if a method belongs to the `Server`/`Client` or a `Node`. Let us try to go with the approach with free functions and try make the `Node` class a convenience class. We don't have to assign all functionality to either `Server`/`Client` or a `Node`. More sophisticated functionalities may just live as free functions.
> 
> ## I propose following steps
> 1. Move the `Node` methods to free functions in a separate namespace and use those functions in the `Node` implementation
>    
>    * Naming of the namespace? I would propose `service` or `services`. The OPC UA specification uses the term [Services](https://reference.opcfoundation.org/Core/Part4/v104/docs/) for the different service sets like NodeManagement (add/delete nodes, add references, ...), Attribute (read/write node attributes), Method, Discovery, ...
>    * File structure:
>      ```
>      services/Attribute.h/cpp
>      services/NodeManagement.h/cpp
>      ...
>      ```
> 2. Implement a basic `Client` class
> 3. Implement option 4, the templated `Node` class with both `Node<Server>` and `Node<Client>`
> 4. Implement free functions for the `Client`
> 
> Any remarks or suggestions @duca, @ErikTempelaarVO?
> 
> I will start working on step 1 in a feature branch and invite you for the review. @duca: Can you provide a basic `Client` class implementation similar to the current `Server` class?

@lukasberbuer like [this one](https://github.com/duca/open62541pp/blob/9630ad93abe42672ae37dbf2463a743fe5a19f48/include/open62541pp/Client.h) ?  I have not rebased my fork yet, so I may be missing updates you did to your Server class.

Regarding the proposition to remove Client/Server from the Node, please bear in mind I didn't have the time to figure out all the details yet, but the backstory is that one of my projects communicates with a couple of PLCs with a few hundreds of nodes each. Not massive, but it is not a small number either, so the idea was to invert the dependency. Having Client accept a Node and extract everything it needed to communicate with the server from the Node. 

Otherwise, I would have 500 nodes, each holding a reference or a pointer to the same Client instance (assuming I was not misusing the design).

To that end, Node would encompass NodeId, of course, but also ReferenceDescription, helper functions to convert from native data types to ours and whatever extra ticks needed to ease the subscription implementation later on. Node could also hold a list of NodeIds of its child nodes if it is a Reference type.

I didn't dive through that rabbit hole yet because I was trying not to diverge too much from your design, hoping to be able to merge our work together at some point.

Regarding your proposition, I agree to your points, with the exception that Node shouldn't take ownership of the Server or Client object. As per my current understanding, to meet my project needs I would need hundreds of nodes, and it would not be wise to have one client object for each of them.

lukasberbuer, 2023-04-10:
Thanks @duca for the feedback!

I didn't touch the `Server` class the last weeks so it should be up to date. Yes your referenced `Client` implementation would be a good starting point. Would you create a feature branch + pull request with this minimal implementation or should I?

I understand your concerns regarding the memory size of the `Node` class. At the moment, the `Node` class has following members:
- `NodeId nodeId_`: 24 bytes on a 64-bit system (same size as `UA_NodeId`)
- `Server server_`: 16 bytes on a 64-bit system due to the `std::shared_ptr` to the underlaying connection

So it's definitely an overhead storing `Node`s instead of just the `NodeId`s. We could reduce the the memory size of `Node` by 8 bytes if we just store a reference or pointer to the `Server` instance. But I guess, But that's still 8 bytes extra compared to plain `NodeId`s - which sums up for 500 nodes.

I see following approaches if you want to store a lot of nodes:
1. Store the `NodeId`s and if you want to use the `Node` later, just use the `Server::getNode(const NodeId& id)` method
2. Store the `NodeId`s and use the free functions. I you look at the `Node` implementation using the free functions, it's just as simple as using the `Node` interface: https://github.com/open62541pp/open62541pp/blob/feat/free-functions/include/open62541pp/Node.h
3. *Extreme*: Use integer-only `NodeId`s, just store the namespace index once and all integer ids. Create `NodeId`s on demand and use approach 1 or 2 for usage.

I think that the free-function approach should solve the large-scale problems with the `Node` class. Still, I see the `Node` interface as a beginner-friendly alternative to the free functions and don't want to drop it. Maybe we should switch from `shared_ptr` to a reference to save some memory...

PS: Can you explain for idea with the templated Node for native types idea a little bit?

duca, 2023-04-10:
> Thanks @duca for the feedback!
>
> I didn't touch the `Server` class the last weeks so it should be up to date. Yes your referenced `Client` implementation would be a good starting point. Would you [...]

Maybe I made myself harder to understand than needed, thanks for bearing with me. 

Everything I said about Node, I meant NodeClient as per current implementation. Our PLCs will be connected directly to a FactoryTalk instance running somewhere, and my application will connect to it. Memory consumption is not really a concert at this moment, either.

It was never my intention to dispense with the Node (or NodeClient), maybe have a single class that was agnostic whether is being used at the server or client side. This was just an attempt to describe what I was thinking on before I came to this discussion, and I understood your point that the current Node API is desirable from your end (and I am completely ok with it).

**Regarding building a feature branch:**

I can do that, but after a rebase against your current server, I noticed its API got smaller than the one I started with, would you like me to adjust the client to really match your current server API or it won't matter because of the changes we are planning to do?

**Regarding the templated Nodes** 

Not sure if that is too much work when using C++17, but one design that came to me was having stuff like
```Node<std::string>
Node<std::vector<uint16_t>>
Node<bool> 
Node<ReferenceType>
... so on
```
But this idea makes more sense if we had stuff such as:
`server.writeValue(Node<T> node, std::string value);`

With access to C++20 features, it would be very clean to implement `writeValue` for scalars, another for arrays of scalars, and so on. 

In any case, that is off the table because we agreed to move towards free functions and also keeping Node/NodeClient APIs as they are for convenience.

lukasberbuer, 2023-04-13:
> **Regarding the template Nodes**
> 
> Not sure if that is too much work when using C++17, but one design that came to me was having stuff like
> 
> ```
> Node<std::vector<uint16_t>>
> Node<bool> 
> Node<ReferenceType>
> ... so on
> ```
> 
> But this idea makes more sense if we had stuff such as:
> 
> ```
> server.writeValue(Node<std::string> const& node).
> client.readValue(Node<std::uint32_t> &node).
> ```
> 
> With access to C++20 features, it would be very clean to implement `writeValue` for scalars, another for arrays of scalars, and so on.

That's an interesting idea. We could move some errors (wrong node class) from runtime to compile time. On the other hand, we loose the capability to store `Node`s in a container. I let this idea rotate in my head :)

ErikTempelaarVO, 2023-04-17:
Pretty cool to see how this was picked up. Good effort on the in-progress & merged PRs and the nice work break down here too:
https://github.com/open62541pp/open62541pp/pull/23#issuecomment-1506646904

I hope to consume this recipe in the future via conan, so I've written a recipe for v0.1.0 and 0.2.0.
I'd be happy to contribute in other areas as well.
I'm still familiarizing myself with open62541 (without the pp), maybe I can help with subscriptions? Or browsing? Any suggestions for a "good first issue?"

lukasberbuer, 2023-04-18:
Hi @ErikTempelaarVO,

I'm happy to hear that you would like to contribute. Conan recipes would be a great addition. I heavily use Conan for dependencies but I never created a recipe myself. Would you create the recipe inside this repository or outside? Which approach more sense if we want to get it into conan-center-index later?

Another good issue might be #34 to get started. It's mainly about wrapping the `UA_ServerConfig` and `UA_ClientConfig` with a nice API. It's easy to get memory leaks but the `TypeWrapper` class and the `asWrapper` function will help.

Thanks for the note about the major changes in the client code 👍

lukasberbuer, 2023-04-22:
Update: I'm working on browsing in #40.

lukasberbuer, 2023-04-25:
@ErikTempelaarVO, browsing (#40) is merged into the master. Both `Client` and `Server` support now following service sets:
- node management (add/delete nodes and references)
- attribute (read/write attributes)
- view/browse + some higher-level browse methods in the `Node` class (`browseReferences`, `browseChild`, `browseChildren`, `browseParent`, ...)

Is the client ready-to-use for you? What features are important but still missing?

closed 2023-05-01 by lukasberbuer

ErikTempelaarVO, 2023-05-01:
Thanks, the new release looks promising.
I currently have a client built around open62541, and was working on the refactor to use the ++ client.
Just upgraded the submodule to v0.3.0.

My client will use a preconfigured set of subscriptions and monitored items and I'd want the stack to inform me of any changes.
So for me that's 'important", but that might be part of another issue.
(and since you closed this while is was typing this, it seems you agree  :-) )

lukasberbuer, 2023-05-01:
Thanks @ErikTempelaarVO! We had to introduce quite some breaking changes. But the API should be quite stable in the future. I'm currently working on subscriptions. Let's continue this discussion in following PR.

duca, 2023-05-05:
@lukasberbuer how far have you go implementing subscription? I was working on a similar feature

lukasberbuer, 2023-05-05:
Hi @duca, I've been working on this heavily the last days. You can see the current status in #45.
I plan to add a high-level `Subscription` in addition to the free functions.
What have you done so far? Do you have any remarks/issues/ideas with the state of #45?