open62541pp/open62541pp #30: Efficient wrapper of composed types

lukasberbuer, 2023-04-13:
Let's imagine we want to create a type-safe wrapper of  a composed type like `UA_ReferenceDescription`. The type is composed of builtin types, for which we have already wrappers implemented (`NodeId`, `ExpandedNodeId`, `QualifiedName`, `LocalizedText`).

```cpp
typedef struct {
    UA_NodeId referenceTypeId;
    UA_Boolean isForward;
    UA_ExpandedNodeId nodeId;
    UA_QualifiedName browseName;
    UA_LocalizedText displayName;
    UA_NodeClass nodeClass;
    UA_ExpandedNodeId typeDefinition;
} UA_ReferenceDescription;
```

How do we make a composed wrapper with convenient getters/setters? First try:

```cpp
class ReferenceDescription : public TypeWrapper<UA_ReferenceDescription, UA_TYPES_REFERENCEDESCRIPTION> {
public:
    ReferenceDescription(
        const NodeId& referenceTypeId,
        bool isForward,
        const ExpandedNodeId& nodeId,
        const QualifiedName& browseName,
        const LocalizedText& displayName,
        NodeClass nodeClass,
        const ExpandedNodeId& typeDefinition
    ) {
        // ...
    }

    NodeId getReferenceTypeId() {
        return NodeId(handle()->referenceTypeId);  // copy
    }

    void setReferenceTypeId(const NodeId& id) {
        TypeConverter<NodeId>::toNative(id, handle()->referenceTypeId);  // copy
    }

    // ...
};
```

Both in the getter and setter, the object is copied. What as waste.
**Wouldn't it be nice to just return a reference to the wrapped member?**:

```cpp
NodeId& getReferenceTypeId() noexcept;
const NodeId& getReferenceTypeId() const noexcept;
```

We don't need setters anymore because we can manipulate and assign through the reference.

## Possible implementation

According to the standard, `TypeWrapper<T>` and `T` are pointer-interconvertible, because:

> Two objects a and b are pointer-interconvertible if: one is a [standard-layout](https://en.cppreference.com/w/cpp/language/data_members#Standard_layout) class object and the other is the first non-static data member of that object or any base class subobject of that object
> -- [cppreference](https://en.cppreference.com/w/cpp/language/static_cast#pointer-interconvertible)

We can cast a native reference to it's wrapper type reference with a simple helper function:

```cpp
template <typename TWrapper, typename TNative = typename TWrapper::UaType>
constexpr TWrapper& asWrapper(TNative& native) noexcept {
    static_assert(std::is_standard_layout_v<TWrapper>);
    static_assert(sizeof(TWrapper) == sizeof(TNative));
    return *static_cast<TWrapper*>(static_cast<void*>(&native));
}

// same for const reference...
```

Then, the implementation of the `ReferenceDescription` getters could be like this:

```cpp
NodeId& getReferenceTypeId() noexcept {
    return asWrapper<NodeId>(handle()->referenceTypeId);
}

const NodeId& getReferenceTypeId() const noexcept {
    return asWrapper<NodeId>(handle()->referenceTypeId);
}
```

duca, 2023-04-13:
Awesome solution, I had the pedestrian version here https://github.com/duca/open62541pp/blob/5e5c6bbde035528664f2669cdd8f25614ae2ec22/include/open62541pp/types/ReferenceDescription.h

duca, 2023-04-22:
> ```c++
> ReferenceDescription(
>         const NodeId& referenceTypeId,
>         bool isForward,
>         const ExpandedNodeId& nodeId,
>         const QualifiedName& browseName,
>         const [...]

@lukasberbuer , when using clang 15 or gcc-12, I get this when using your asWrapper example
```bash
Binding reference of type 'opcua::NodeId' to value of type 'const opcua::NodeId' drops 'const' qualifier
```

The code is:

```cpp
NodeId& ReferenceDescription::getReferenceTypeId() const {
    return asWrapper<NodeId>(handle()->referenceTypeId);
}
NodeId& ReferenceDescription::getReferenceTypeId() {
    return asWrapper<NodeId>(handle()->referenceTypeId);
}
```

duca, 2023-04-22:
> Let's imagine we want to create a type-safe wrapper of a composed type like `UA_ReferenceDescription`. The type is composed of builtin types, for which we have already wrappers implemented (`NodeId`, `ExpandedNodeId`, `QualifiedName`, `LocalizedText`).
> 
> ```c++
> typedef struct {
>     UA_NodeId referenceTypeId;
>     UA_Boolean isForward;
>     UA_ExpandedNodeId nodeId;
>     UA_QualifiedName browseName;
>     UA_LocalizedText displayName;
>     UA_NodeClass nodeClass;
>     UA_ExpandedNodeId typeDefinition;
> } UA_ReferenceDescription;
> ```
> 
> How do we make a composed wrapper with convenient getters/setters? First try:
> 
> ```c++
> class ReferenceDescription : public TypeWrapper<UA_ReferenceDescription, UA_TYPES_REFERENCEDESCRIPTION> {
> public:
>     ReferenceDescription(
>         const NodeId& referenceTypeId,
>         bool isForward,
>         const ExpandedNodeId& nodeId,
>         const QualifiedName& browseName,
>         const LocalizedText& displayName,
>         NodeClass nodeClass,
>         const ExpandedNodeId& typeDefinition
>     ) {
>         // ...
>     }
> 
>     NodeId getReferenceTypeId() {
>         return NodeId(handle()->referenceTypeId);  // copy
>     }
> 
>     void setReferenceTypeId(const NodeId& id) {
>         TypeConverter<NodeId>::toNative(id, handle()->referenceTypeId);  // copy
>     }
> 
>     // ...
> };
> ```
> 
> Both in the getter and setter, the object is copied. What as waste. **Wouldn't it be nice to just return a reference to the wrapped member?**:
> 
> ```c++
> NodeId& getReferenceTypeId() noexcept;
> const NodeId& getReferenceTypeId() const noexcept;
> ```
> 
> We don't need setters anymore because we can manipulate and assign through the reference.
> 
> ## Possible implementation
> According to the standard, `TypeWrapper<T>` and `T` are pointer-interconvertible, because:
> 
> > Two objects a and b are pointer-interconvertible if: one is a [standard-layout](https://en.cppreference.com/w/cpp/language/data_members#Standard_layout) class object and the other is the first non-static data member of that object or any base class subobject of that object
> > -- [cppreference](https://en.cppreference.com/w/cpp/language/static_cast#pointer-interconvertible)
> 
> We can cast a native reference to it's wrapper type reference with a simple helper function:
> 
> ```c++
> template <typename TWrapper, typename TNative = typename TWrapper::UaType>
> constexpr TWrapper& asWrapper(TNative& native) noexcept {
>     static_assert(std::is_standard_layout_v<TWrapper>);
>     static_assert(sizeof(TWrapper) == sizeof(TNative));
>     return *static_cast<TWrapper*>(static_cast<void*>(&native));
> }
> 
> // same for const reference...
> ```
> 
> Then, the implementation of the `ReferenceDescription` getters could be like this:
> 
> ```c++
> NodeId& getReferenceTypeId() noexcept {
>     return asWrapper<NodeId>(handle()->referenceTypeId);
> }
> 
> const NodeId& getReferenceTypeId() const noexcept {
>     return asWrapper<NodeId>(handle()->referenceTypeId);
> }
> ```

@lukasberbuer, so I tried implemented the changes you suggested, and
```cpp
NodeId& ReferenceDescription::getReferenceTypeId() const noexcept {
    return asWrapper<NodeId>(handle()->referenceTypeId);
}

NodeId& ReferenceDescription::getReferenceTypeId() noexcept {
    return asWrapper<NodeId>(handle()->referenceTypeId);
}

bool& ReferenceDescription::isForward() const noexcept {
    return asWrapper<bool>(handle()->isForward);
}
```
 and got this:

```bash
open62541pp/src/types/ReferenceDescription.cpp:  In member function ‘opcua::NodeId& opcua::ReferenceDescription::getReferenceTypeId() const’:
open62541pp/src/types/ReferenceDescription.cpp:26:29: error: binding reference of type ‘opcua::NodeId&’ to ‘const opcua::NodeId’ discards qualifiers
   26 |     return asWrapper<NodeId>(handle()->referenceTypeId);
      |            ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
open62541pp/src/types/ReferenceDescription.cpp: In member function ‘bool& opcua::ReferenceDescription::isForward() const’:
open62541pp/src/types/ReferenceDescription.cpp:34:27: error: binding reference of type ‘bool&’ to ‘const bool’ discards qualifiers
   34 |     return asWrapper<bool>(handle()->isForward);
      |            ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
```
I had to do a slight change for it to work
```cpp
    NodeId const& getReferenceTypeId() const noexcept;
```

lukasberbuer, 2023-04-22:
Did you check this with the current master? The `asWrapper` implementation has a non-const and const overload.
The composed types are in [`types/Composed.h`](https://github.com/open62541pp/open62541pp/blob/master/include/open62541pp/types/Composed.h) and I use a macro to generate the code both for the non-const and const getter:

```cpp
#define UAPP_COMPOSED_WRAPPER_GETTER(WrapperType, member, getterName)                              \
    WrapperType& getterName() noexcept {                                                           \
        return asWrapper<WrapperType>(handle()->member);                                           \
    }                                                                                              \
    const WrapperType& getterName() const noexcept {                                               \
        return asWrapper<WrapperType>(handle()->member);                                           \
    }
```

I even got lazier in #40 and added more macros for different getter scenarios.