open62541pp/open62541pp #34: ServerConfig / ClientConfig class

lukasberbuer, 2023-04-18:
# Status quo

The `UA_ServerConfig` is created by the `Server` constructor. Some options can be set via the constructor or server methods:

```cpp
class Server {
public:
    Server::Server();
    explicit Server(uint16_t port);
    Server(uint16_t port, std::string_view certificate);

    // methods to change UA_ServerConfig
    void setLogger(Logger logger);
    void setCustomHostname(std::string_view hostname);
    void setApplicationName(std::string_view name);
    void setApplicationUri(std::string_view uri);
    void setProductUri(std::string_view uri);
    void setLogin(const std::vector<Login>& logins, bool allowAnonymous = true);
    // ...
};
```

# Issues

This API suggests, that the config can be changed at any time, but this can be dangerous if the server is already running.
Additionally, the interface of the `Server` gets bloated if we add methods for all config options over time.

# Possible solution
I would prefer a simpler approach with a separate `ServerConfig` class with is passed to the `Server` constructor.

```cpp
class ServerConfig {
public:
    ServerConfig()  // -> create default config
    // maybe some other useful constructors as starting points

    // methods to change UA_ServerConfig
    void setLogger(Logger logger);
    void setCustomHostname(std::string_view hostname);
    void setApplicationName(std::string_view name);
    void setApplicationUri(std::string_view uri);
    void setProductUri(std::string_view uri);
    void setLogin(const std::vector<Login>& logins, bool allowAnonymous = true);
    // ...
}

class Server {
public:
    Server();  // -> default config
    explicit Server(ServerConfig config);  // -> move config into server to prevent further changes
};
```

Any suggestions or ideas?

ErikTempelaarVO, 2023-04-19:
I think it would be nice to have a fluent interface here (i.e. [builder pattern](https://riptutorial.com/cplusplus/example/30166/builder-pattern-with-fluent-api))
That way you can easily recreate the config (if you keep the builder around) for a new instance of the server.

(Recreating the client is required in case of an irrecoverable error, I don't think that really applies to the server though.
The client recently got these _copy methods:  [open62541 PR](https://github.com/open62541/open62541/commit/38fc1ccdafe3f8e0963c3b548eb11c6e1077475c) )

I created a draft in my fork, [here](https://github.com/open62541pp/open62541pp/compare/master...ErikTempelaarVO:open62541pp:feat/builder-server-config)

lukasberbuer, 2023-04-20:
@ErikTempelaarVO, great job! I like the fluent interface. Some ideas / remarks:

1.  We should have a `UA_ServerClient` wrapper class to store something besides the actual config. At the moment, we only have a custom logger - but it's gonna be more in the future. The actual user function `Logger` is currently [stored in the `Server` class](https://github.com/open62541pp/open62541pp/blob/master/src/Server.cpp#L110). This should be moved to the config within this refactoring. The class interface can be very minimalistic, e.g.:

    ```cpp
    class ServerConfig {
    public:
        ServerConfig();
 
        ~ServerConfig() {
            UA_ServerConfig_clean(config_);
        }
    
        // prevent copy
        ServerConfig(const ServerConfig&) = delete;
        ServerConfig& operator=(const ServerConfig&) = delete;
    
        UA_ServerConfig* handle() noexcept {
            return config_;
        }
    
    protected:
        UA_ServerConfig config_{};
        CustomLogger logger_;
    }
    ```

2.  I would prefer to directly manipulate the `UA_ServerConfig` instead of storing the setup in C++ types/containers and then converting lazily to `UA_ServerConfig` in the build step. What is the advantage of the lazy building in your opinion?

lukasberbuer, 2023-04-20:
Just saw how the Rust guys did this with a similar fluent interface: https://github.com/locka99/opcua/blob/master/docs/server.md#serverbuilder

ErikTempelaarVO, 2023-04-20:
2. Because I was thinking of transferring ownership of the UA_ServerConfig to the actual UA_Server and didn't want to keep an object in memory unless it was actually built and consumed by the UA_Server. But this wrapper makes perfect sense and that will ensure the lifetime of the native UA_ServerConfig*.

I'm not sure if the _clean should be handled in this wrapper. Docs state:
> /* Creates a new server. Moves the config into the server with a shallow copy.
> * The config content is cleared together with the server. */
> UA_Server *
> UA_Server_newWithConfig(UA_ServerConfig *config);

And the lazy approach allowed me to make use of the existing initialization with 'UA_ServerConfig_setMinimalCustomBuffer'
It's just that some settings (like port) cannot be stored right away.
But I wasn't too happy with that either.

I was unaware of the rust-implementation, I'll take another good look at that.

lukasberbuer, 2023-04-21:
> I'm not sure if the _clean should be handled in this wrapper. Docs state:
> 
> > /* Creates a new server. Moves the config into the server with a shallow copy.
> > 
> > * The config content is cleared together with the server. */
> >   UA_Server *
> >   UA_Server_newWithConfig(UA_ServerConfig *config);

That's a tricky one. `UA_Server_delete` calls `UA_ServerConfig_clean`: https://github.com/open62541/open62541/blob/v1.3.5/src/server/ua_server.c#L186-L249.
And `UA_ServerConfig_clean` first checks if the pointer is `NULL`: https://github.com/open62541/open62541/blob/v1.3.5/src/server/ua_server_config.c#L12.

One solution might be to keep the cleanup in the `ServerConfig` destructor but set the `config` to `nullptr` in the `Server` destructor before `UA_Server_delete` is called so we avoid double free.

closed 2023-07-29 by lukasberbuer