open62541pp/open62541pp #41: test_package to verify installed cmake macros

duca, 2023-04-23:
> > Thanks @duca! I pushed some minor changes. For testing, I just installed it to a subdirectory and inspected the files. How do you test it?
> > Does it make sense to add a package test similar how Conan does it? https://github.com/conan-io/conan-center-index/tree/master/recipes/open62541/all/test_package
> 
> I think it does make sense, I built a stub project and optimized those lines and did some minor changes to make it work properly

@lukasberbuer any suggestions on how to integrate the test_package with the test fixture?

lukasberbuer, 2023-04-24:
I would keep the `test_package` application completely isolated from this project. They can live in the same repository but the tests `CMakeLists.txt` should not be configured by the main project.

It's harder to test locally on the PC but should be pretty easy in CI / GitHub actions:
1. build and install open62541pp
   1. `mkdir build` && `cd build`
   2. `cmake ..`
   3. `make && make install`
2. build test_package
   1. `mkdir build_test_package` && cd `build_test_package`
   2. `cmake ../test_package`
   3. `make`
3. succeeds if test compiles

I would like to create a test similar to the open62541 one: https://github.com/conan-io/conan-center-index/blob/master/recipes/open62541/all/test_package/test_package.c.

ErikTempelaarVO, 2023-04-24:
Maybe there's something of use here:
https://github.com/ErikTempelaarVO/conan-center-index/tree/add-recipe-open62541pp/recipes/open62541pp/all

lukasberbuer, 2023-04-24:
@ErikTempelaarVO uses the already existing example(s), that's smart!

I just looked how other libraries do it, e.g. `spdlog`: https://github.com/gabime/spdlog/blob/v1.x/example/CMakeLists.txt.
Those examples can be built standalone. In this case, we don't even have to create new tests but built the examples in-project or standalone:

lukasberbuer, 2023-04-24:
Sorry, I accidentally closed this issue. Will open a PR in a few minutes :)

lukasberbuer, 2023-04-24:
Seems to work fine. PR #42 already merged.

PR #38 (2023-04-24): CMake install targets and a small fix for the interface target
In this PR, I fixed the usage of the open62541pp_project_options target for modern cmake and added the installation of the infrastructure necessary to find_package this library when `lib/cmake/open62541pp` path is appended to the user's CMAKE_MODULE_PATH list

resolves #41
closed 2023-04-24 by lukasberbuer