open62541pp/open62541pp #46: Header open62541/nodeids.h not found when using UAPP_INTERNAL_OPEN62541

GMellar, 2023-05-10:
I tested version 0.3.0 and the compilation fails with the missing header. I think this is due to missing target_include_directories when using the internal open62541 library. The binary directory is not present in the include directory list.

lukasberbuer, 2023-05-10:
Hi @GMellar, thanks for reporting!

The header `open62541/nodeids.h` is auto-generated in `"${PROJECT_BINARY_DIR}/src_generated"` but is exported via `target_include_directories` within open62541: https://github.com/open62541/open62541/blob/v1.3.5/CMakeLists.txt#L1453-L1466
So this *should* work...

Did you set `UAPP_INTERNAL_OPEN62541` to `ON` or `OFF`?
Could you please provide a dump of your CMake cache variables with `cmake -LA`?

GMellar, 2023-05-10:
I used UAPP_INTERNAL_OPEN62541. Looking into the open62541 C library these includes should be exported so I don't understand that at the moment. 

[cache](https://github.com/open62541pp/open62541pp/files/11443317/ua.txt)

lukasberbuer, 2023-05-10:
Ah I see, you have `UA_ENABLE_AMALGAMATION` enabled. Did you set it intentionally? This will result in one huge header file `open62541/open62541.h`, so other header files are not found.

I need to add some simple checks in my includes to work with the amalgamated version as well. Fix will come the next days. I will link the PR to this issue so you can track it.

Are you ok do use turn set `UA_ENABLE_AMALGAMATION=OFF` meanwhile?

lukasberbuer, 2023-05-10:
We should definitely support amalgamation with open62541. Just had a looked at the code; this should be a 4-liner. PR is coming 😄

I personally would rather turn `UA_ENABLE_AMALGAMATION` off to improve the build time. open62541pp will only include the absolute minimum of open62541 headers needed. Everything else is forward declared or just used in the implementation. Because most people will build and use the static library , the result should be the same.

closed 2023-05-10 by lukasberbuer