open62541pp/open62541pp #51: MonitoredItems no longer work after re-connect

ErikTempelaarVO, 2023-05-19:
If a working connection is killed by restarting the server the connection seems to be restored (with a warning, about a sequence mismatch), but MonitoredItems no longer work for me.

[2023-05-19 19:08:33.627] [trace  ] Current server time (UTC): 13:42:14
[2023-05-19 19:08:34.660] [trace  ] Current server time (UTC): 13:42:15
[2023-05-19 19:08:35.175] [warning] OPCUA channel Connection 1380 | SecureChannel 4 | Receiving the response failed with StatusCode BadConnectionClosed 
[2023-05-19 19:08:35.175] [warning] OPCUA client Received Publish Response with code BadSecureChannelClosed 
[2023-05-19 19:08:35.175] [info   ] Skipped 9 duplicate messages..
[2023-05-19 19:08:35.175] [warning] OPCUA channel Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed 
[2023-05-19 19:08:35.175] [debug  ] OPCUA client Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good 
[2023-05-19 19:08:35.175] [info   ] OPCUA has been disconnected
[2023-05-19 19:08:35.689] [trace  ] OPCUA attempting to connect to opc.tcp://10.0.0.130:4840 anonymously
[2023-05-19 19:08:37.752] [warning] OPCUA network Connection to opc.tcp://10.0.0.130:4840 failed with error: No connection could be made because the target machine actively refused it.
 
[2023-05-19 19:08:37.752] [debug  ] OPCUA client Client Status: ChannelState: Fresh, SessionState: Created, ConnectStatus: BadDisconnect 
[2023-05-19 19:08:37.769] [warning] OPCUA failed to connect to opc.tcp://10.0.0.130:4840
[2023-05-19 19:08:51.175] [trace  ] OPCUA attempting to connect to opc.tcp://10.0.0.130:4840 anonymously
[2023-05-19 19:08:51.189] [debug  ] OPCUA channel Connection 1452 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 120.00s 
[2023-05-19 19:08:51.189] [debug  ] OPCUA client Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good 
[2023-05-19 19:08:51.189] [info   ] OPCUA is connected
[2023-05-19 19:08:51.192] [debug  ] OPCUA client Received a ServiceFault response 
[2023-05-19 19:08:51.192] [debug  ] OPCUA client The ServiceResult has the StatusCode BadSessionIdInvalid 
[2023-05-19 19:08:51.192] [error  ] OPCUA client ActivateSession failed with error code BadSessionIdInvalid 
[2023-05-19 19:08:51.192] [debug  ] OPCUA client Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good 
[2023-05-19 19:08:51.192] [trace  ] OPCUA session closed. Stopping subscriptions
[2023-05-19 19:08:51.192] [error  ] OPCUA client Session cannot be activated. Create a new Session. 
[2023-05-19 19:08:51.194] [debug  ] OPCUA client Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good 
[2023-05-19 19:08:51.196] [debug  ] OPCUA client Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good 
[2023-05-19 19:08:51.196] [trace  ] OPCUA session activated. Starting subscriptions
[2023-05-19 19:09:04.875] [debug  ] OPCUA userland Inactivity for client 
[2023-05-19 19:09:05.377] [debug  ] OPCUA client Error receiving the response with status code BadSecurityChecksFailed 
[2023-05-19 19:09:05.377] [warning] OPCUA channel Connection 1452 | SecureChannel 1 | Receiving the response failed with StatusCode BadSecurityChecksFailed 
[2023-05-19 19:09:05.377] [warning] OPCUA client Received Publish Response with code BadSecureChannelClosed 
[2023-05-19 19:09:05.377] [info   ] Skipped 9 duplicate messages..
[2023-05-19 19:09:05.377] [warning] OPCUA channel Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed 
[2023-05-19 19:09:05.377] [debug  ] OPCUA client Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good 
[2023-05-19 19:09:05.377] [info   ] OPCUA has been disconnected
[2023-05-19 19:09:06.410] [trace  ] OPCUA attempting to connect to opc.tcp://10.0.0.130:4840 anonymously
[2023-05-19 19:09:06.414] [debug  ] OPCUA channel Connection 1380 | SecureChannel 2 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 120.00s 
[2023-05-19 19:09:06.414] [debug  ] OPCUA client Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good 
[2023-05-19 19:09:06.414] [info   ] OPCUA is connected
[2023-05-19 19:09:06.416] [debug  ] OPCUA client Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good 
[2023-05-19 19:09:06.416] [trace  ] OPCUA session activated. Starting subscriptions
[2023-05-19 19:09:06.926] [warning] OPCUA client Invalid subscription sequence number: expected 9 but got 15 
[2023-05-19 19:10:37.160] [debug  ] OPCUA channel Connection 1380 | SecureChannel 2 | SecureChannel renewed with a revised lifetime of 120.00s 

The DataChanged-callback is still called, but requesting the NodeId/AttributeId fails with an exception.

`
throw BadStatus(UA_STATUSCODE_BADMONITOREDITEMIDINVALID);
`
I catch that error, but that only prevents a crash.

`  ::opcua::ReadValueId rvid;
  try {
    rvid = ::opcua::ReadValueId(item.getNodeId(), item.getAttributeId());
  } catch (const ::opcua::BadStatus& status) {
    // Unable to get NodeId and AttributeId?
    return;
  }
  `

Should I actively delete my subscriptions/monitored items from the client in case of an error? Or am I supposed to keep them around?

Thanks in advance. I appreciate the work being done for this wrapper-library.

lukasberbuer, 2023-05-20:
Hi @ErikTempelaarVO,
I had this error once but could not reproduce. The problem is, that I store some state when a MonitorenItem is created and delete it when the MonitoredItem is deleted. I rely on the delete callbacks of open62541 here. It seems that a mo itored item can get deleted but reactivated if a session is reactivated.
I am not sure how to handle the lifetime of the monitored items currently. When can the state of a monitored item be safely deleted?

Workaround for now: add a `client.disconnect()` in the catch block to avoid a reactivation of the session.

lukasberbuer, 2023-05-23:
@ErikTempelaarVO does the disconnect workaround work for you?

closed 2024-07-18 by lukasberbuer