open62541pp/open62541pp #52: MonitoredItemIds are not correct

ErikTempelaarVO, 2023-05-25:
When subscribing to multiple nodes the MonitoredItemIDs do not change.
This results in incorrect NodeIDs/AttributeIDs, when fetching from inside the callback.

`Data change notification:
- subscription id:   21
- monitored item id: 1
- node id:           i=2258
- attribute id:      13
Current server time (UTC): 09:28:22
Data change notification:
- subscription id:   21
- monitored item id: 1
- node id:           i=2258
- attribute id:      13
The Answer: 215`

The mixup happens here:

`
static const MonitoredItem<Client> monitoredItem(*connectionPtr, subId, monId);
callback(monitoredItem, value);
`

The resulting MonitoredItem<Client> is always monId '1'. Even if the passed monId = 2.
Removing the static helps.

[image]

lukasberbuer, 2023-05-26:
Hi @ErikTempelaarVO, thanks for find this bug. I am a little bit embarrassed that I didn't think of that. I will fix this and make a patch release on Monday.

lukasberbuer, 2023-05-26:
Or do you want to provide a pull request?

ErikTempelaarVO, 2023-05-26:
There are two possible solutions as I see it

1. Simply drop the static and recreate the MI<C> every callback
2. Store the MI<C> inside the ClientContexts MonitoredItem

Do you see a (performance) issue with 1. is the additional bookkeeping required for 2. worth it?

Any other approaches?

ErikTempelaarVO, 2023-05-26:
Maybe I should do 1. anyway, to get this fixed.
2. is a perfomance optimization that can go in later, that might take some more thinking in another PR.

lukasberbuer, 2023-05-27:
I would prefer solution 1. The construction of MonitoredItem is pretty cheap. With solution 2, we have to do a pointer dereference + map lookup. I guess it won't be faster.

PR #53 (2023-05-30): fix: incorrect monitoreditem ids
This fixes #52