open62541pp/open62541pp #54: MethodNode in client/server

mkuznecovs, 2023-06-02:
Hello,

I was wondering if adding method nodes to server is possible? (and calling them from a client)

Thank you!

Kind regards,
Maksims

lukasberbuer, 2023-06-02:
Hi @mkuznecovs, we did not implement method nodes yet. Currently you have to mix in the open62541 API. I did some first drafts and could finalize it in the next two weeks. Can you wait so long? 😀  Do you want to collaborate on this feature?

mkuznecovs, 2023-06-04:
Hi @lukasberbuer, The time is not an issue, I care much more about the quality. I would love to collaborate on this feature. Though I have never collaborated on github before, but I want to learn how to. Maybe you could give me a direction on where and how could I begin a collaboration with you? Thanks in advance!

lukasberbuer, 2023-06-05:
I will create a draft of the feature in a new branch, then we can discuss and collaborate on this branch until the feature is complete. The steps are following:
- fork the repository
- create a branch (usually for a specific feature, feature addition, bug fix, ...)
- work on the feature, bug fix, ...
- create a pull request to merge the changes into a branch of the open62541pp repository
- pull request will (eventually) be merged by the maintainer (me)

I will work on the prototype this week and notify you when we can start with the collaboration.

fabianhaerle, 2023-06-20:
I also am strongly interested in this feature. Do you have an idea about when it will be ready?

lukasberbuer, 2023-06-21:
Thanks for pushing this feature @fabianhaerle. Do you the `Server` or `Client` implementation or both?
I had some busy weeks and hopefully get some results soon 🤞

fabianhaerle, 2023-06-22:
I need a OPC UA server in my application so the client side is not so important to me. Thanks in advance!

lukasberbuer, 2023-06-24:
Just pushed the draft PR with `addMethod`. I hope the [example](https://github.com/open62541pp/open62541pp/blob/feat-method/examples/server_method.cpp) helps.
Do you have any suggestions regarding the API @mkuznecovs @fabianhaerle?

Next will be the `call` implementation for both `Client` and `Server` and finally tests.

lukasberbuer, 2023-06-27:
The `services::call` function and `Node::callMethod` method are implemented now, also with a small [`client_method.cpp` example](https://github.com/open62541pp/open62541pp/blob/feat-method/examples/client_method.cpp).

Do you have any issues/ideas regarding the API?

mkuznecovs, 2023-06-28:
Great job, @lukasberbuer! Regarding the API, server side looks sufficient in my opinion. On a client side it would be nice to add a timeout when calling method.

lukasberbuer, 2023-06-29:
Thanks for the feedback @mkuznecovs.
The timeout is a client-wide configuration in the [`UA_ClientConfig`](https://github.com/open62541/open62541/blob/v1.3.6/include/open62541/client.h#L75). I just made a PR to add the client method `Client::setTimeout` in #56.

closed 2023-06-29 by lukasberbuer