open62541pp/open62541pp #57: Cannot build shared library on Windows

fabianhaerle, 2023-06-29:
Hello,

I would like to build open62541pp as a shared library (dll) on Windows. I am using Visual Studio 2022 which is capable of working with cmake files. I can successfully build a static library but I select BUILD_SHARED_LIBS I get a lot of LNK1104 linker errors - the linker won't find the open62541pp.lib. The same happens when I create a Visual Studio project via CMake.

Do you know how I can solve this?

Best,

Fabian

lukasberbuer, 2023-06-29:
Hi @fabianhaerle,

Windows DLLs need explicit exports of symbols. I just added a PR #58 to fix this issue.
Can you please try again to compile the shared library on this branch?

lukasberbuer, 2023-07-12:
@fabianhaerle, can the issue be closed? Did you try it?

fabianhaerle, 2023-07-13:
No, it only partly works: the open62541pp.dll is built, the open62541pp.lib is not.

lukasberbuer, 2023-07-13:
That's weird. I just tried it out again and linking worked. Both `*.exp` and `*.lib` are created:
[image]

Did you try to delete the CMake cache?

lukasberbuer, 2023-07-13:
Just added both static/shared library builds on Windows to the CI pipeline: bc37bc0
Pipeline fails for shared library build before patch (https://github.com/open62541pp/open62541pp/actions/runs/5548424044) and passes afterwards (https://github.com/open62541pp/open62541pp/actions/runs/5548567943).

If clearing the cache doesn't work, we need to figure out how to reproduce your environment.

fabianhaerle, 2023-07-14:
To be sure I deleted the whole directory and cloned the repo again. Now it works. But for the examples to run one needs to copy the dll to the example directory manually.

lukasberbuer, 2023-07-14:
Yes, this is a Windows specific problem: There isn't anything like RPATH, which is used on most Unix platforms, to inject in other more appropriate paths. Windows will look in the same directory as the binary, and then search the directories in your PATH. 
(https://stackoverflow.com/questions/20433308/cmake-on-windows-doesnt-add-shared-library-paths-works-on-linux)
Btw., you will have the same issue with the open62541 examples.

Usually the workaround is to build everything to a single build directory. Otherwise you have to copy the runtime dlls to the same folder or specify their location in the `PATH` variable.

I was about to add the automatic copying of the runtime dlls to the examples `CMakeLists.txt`.  BUT I think it's not worth it. The examples will work but if you try to link the shared library to your own application, you will run into the same problems.

I post the solution here anyway `examples/CMakeLists.txt`:
```cmake
// ...
function(add_example source)
    # extract filename
    get_filename_component(name ${source} NAME_WE)
    # use open62541pp_example_ prefix for target to prevent naming collisions
    set(target_name "open62541pp_examples_${name}")
    add_executable(${target_name} ${source})
    target_link_libraries(
        ${target_name}
        PRIVATE
            open62541pp::open62541pp
            $<TARGET_NAME_IF_EXISTS:open62541pp_project_options>  # not available in stand-alone
    )
    set_target_properties(
        ${target_name}
        PROPERTIES
            OUTPUT_NAME ${name}
    )
    if(UAPP_ENABLE_PCH)
        target_precompile_headers(
            ${target_name}
            REUSE_FROM
                open62541pp::open62541pp
        )
    endif()
    # disable clang-tidy check to avoid try/except blocks in examples
    if(CMAKE_CXX_CLANG_TIDY)
        set(checks "-bugprone-exception-escape")  # allow uncaught exceptions in main
        set_target_properties(
            ${target_name}
            PROPERTIES
                CXX_CLANG_TIDY "${CMAKE_CXX_CLANG_TIDY};-checks=${checks}"
        )
    endif()
    # copy shared libraries to examples folder
    # https://stackoverflow.com/a/69736197/9967707
    if(BUILD_SHARED_LIBS)
        add_custom_command(
            TARGET ${target_name} POST_BUILD
            COMMAND ${CMAKE_COMMAND} -E copy $<TARGET_RUNTIME_DLLS:${target_name}> $<TARGET_FILE_DIR:${target_name}>
            COMMAND_EXPAND_LISTS
        )
    endif()
endfunction()
// ...
```

lukasberbuer, 2023-07-14:
Fixed with PR #58

closed 2023-07-14 by lukasberbuer