open62541pp/open62541pp #59: access level control at client

johanneskopf, 2023-07-02:
Hi,

I want to:
1.  create a **parent node** at the server from the client.
2. create a **int node** at the server from the client as a child node of **parent node** and set the attributes accordingly.

Here is a sample code:
```cpp
#include "open62541pp/open62541pp.h"

int main() {
    opcua::Client client;
    client.connect("opc.tcp://localhost:4840");
    auto objectsNode = client.getObjectsNode();
    auto parentNode = objectsNode.addObject({1, "parent.node"}, "parent node");
    auto integerNode = parentNode.addVariable({1, "int.node"}, "int node");
    integerNode.writeDataType(opcua::Type::Int32); // this line throws the error below
    integerNode.writeDisplayName({"en-US", "int node"});
    integerNode.writeDescription({"en-US", "int node"});
    integerNode.writeScalar<int32_t>(42);
}
```

This code always throws the following error:
> terminate called after throwing an instance of 'opcua::BadStatus'
> what():  BadUserAccessDenied

Due to the descriptive error, I figured I need to set the correct access level or write mask at the integer node to be able to write the attributes after creating the node at the server.

So I tried to add the following code in-between, but that resulted in the same error being thrown:
```cpp
    integerNode.writeAccessLevel(UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE);
    integerNode.writeUserAccessLevel(UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE);
    integerNode.writeWriteMask(~0U);
    integerNode.writeUserWriteMask(~0U);
```

How to achieve this?

lukasberbuer, 2023-07-04:
Hi @johanneskopf,
I can reproduce the issue but have to dig deeper.. 🕳️
Hopefully I can provide you a solution soon.

lukasberbuer, 2023-07-04:
A similar issue was also reported for open62541: https://github.com/open62541/open62541/issues/3910
The mentioned status code `2147614720` = `0x80200000` = `BadIdentityTokenInvalid` differs but is the adjacent one:
```
BadUserAccessDenied,0x801F0000,User does not have permission to perform the requested operation.
BadIdentityTokenInvalid,0x80200000,The user identity token is not valid.
```

lukasberbuer, 2023-07-17:
I could narrow down the problem: It makes a difference if the attributes are defined during construction of the node or afterwards. The `writeMask`'s default value is `0` (no permission):
[image]

A small modification of following snippet makes it work: https://github.com/open62541pp/open62541pp/blob/master/src/services/NodeManagement.cpp#L86-L106

```cpp
template <>
void addVariable<Client>(
    Client& client,
    const NodeId& parentId,
    const NodeId& id,
    std::string_view browseName,
    const NodeId& variableType,
    const NodeId& referenceType
) {
    auto attributes = UA_VariableAttributes_default;
    attributes.writeMask = ~0U;
    // attributes.userWriteMask = ~0U;
    const auto status = UA_Client_addVariableNode(
        client.handle(),
        id,
        parentId,
        referenceType,
        QualifiedName(id.getNamespaceIndex(), browseName),
        variableType,
        attributes,
        nullptr  // output new node id
    );
    detail::throwOnBadStatus(status);
}
```

So the only solution will be to change the function signature and allow passing initial node attributes to the function.
I am thinking of one generic `struct NodeAttributes` with all optional attributes and pass it by const reference:
```cpp
struct NodeAttributes {
    std::optional<LocalizedText> displayName;
    std::optional<LocalizedText> description;
    std::optional<uint32_t> writeMask;
    // ...
};

Any other ideas?

lukasberbuer, 2023-07-18:
Just pushed a draft PR #61 with a possible solution to provide initial attributes.

This would result in following client code:
```cpp
#include "open62541pp/open62541pp.h"

int main() {
    opcua::Client client;
    client.connect("opc.tcp://localhost:4840");
    auto objectsNode = client.getObjectsNode();
    auto parentNode = objectsNode.addObject({1, "parent.node"}, "parent node");

    // define initial attributes
    opcua::VariableAttributes attributes;
    attributes.setDisplayName({"en-US", "int node"});
    attributes.setDescription({"en-US", "int node"});
    attributes.setAccessLevel(UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE);
    attributes.setDataType(opcua::DataTypeId::Int32);

    auto integerNode = parentNode.addVariable(
        {1, "int.node"},
        "int node",
        opcua::VariableTypeId::BaseDataVariableType,
        opcua::ReferenceTypeId::HasComponent,
        attributes
    );

    integerNode.writeScalar<int32_t>(42);
}
```

What do you think?
I would rather prefer to have `attributes` as the first default argument for `addVariable`, so we don't need to specify `variableType` and `referenceType`. I guess it's a more frequent use-case to provide initial attributes than custom variable or reference types. BUT this would be a breaking change...

johanneskopf, 2023-07-23:
Thanks so much @lukasberbuer for finding out that the writemask is the culprit!
I would prefer being able to use a single method call instead of many to add a variable. This way, adding a variable is similar to write/read from it. I created a Pull Request (https://github.com/open62541pp/open62541pp/pull/68) to showcase that. Adding the variable would look like this:
```cpp
auto integerNode = objectsNode.addVariable<int32_t>({1, "int.node"}, "int.node", 42, UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE);
```
I also added an overloaded addVariable method so there is no breaking change. There, I put the `variableType` and `referenceType` as last arguments, so we don't need to specify them (I also prefer this like you do!). What do you think?

lukasberbuer, 2023-07-25:
One further idea. You said, that you would prefer to have a single method call. We could get rid of the extra variable declaration with a fluent interface but still keep the possibility to set all attributes initially:

```cpp
auto integerNode = parentNode.addVariable(
    {1, "int.node"},
    "int node",
    opcua::VariableTypeId::BaseDataVariableType,
    opcua::ReferenceTypeId::HasComponent,
    opcua::VariableAttributes{}
        .setDisplayName({"en-US", "int node"})
        .setDescription({"en-US", "int node"})
        .setAccessLevel(UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE)
        .setDataType(opcua::DataTypeId::Int32)
        .setValue(opcua::Variant::fromScalar(42))
);
```

What do you think @johanneskopf?

johanneskopf, 2023-07-25:
Looking at my approach, [named parameters](https://learn.microsoft.com/en-us/dotnet/csharp/programming-guide/classes-and-structs/named-and-optional-arguments) would be a great help, but unfortunately this feature does not exist for C++. However, CLion has a great feature here, [parameter hints](https://www.jetbrains.com/help/clion/parameter-hints.html), that improves readability. It always shows the names of the parameters when calling a method. But even that doesn't solve the problem that optimally we want to decide which variable attribute parameters to pass and which to omit when adding a variable. Your solution on the other hand solves that problem and improves readability. Therefore, I really like the fluent interface approach @lukasberbuer  !

I would still vote for overloading `addVariable` so that it doesn't require a value for all four arguments. Like this:
`addVariable(const NodeId& id, std::string_view browseName, const opcua::VariableAttributes& attributes)`
This would also not break backward compatibility. If this is an option for you, it would also be interesting to know if it makes sense to add the optional two parameters to the VariableAttributes as well. The existing `addVariable` call could then internally call the new, overloaded variant of the method.

Good idea to use this for the setter methods of the `Node` as well. I'm all for it!

lukasberbuer, 2023-07-26:
Happy to hear that the fluent interface suits your needs @johanneskopf! I would also be a big fan of named parameters...

Instead of overloading the `addVariable` method, I'd like to make the breaking change and **add the inital attributes as the first default parameter**. I think this is the most convenient way for most use-cases. Do you agree? Is there any convenient overload missing for?

johanneskopf, 2023-07-29:
I'm already using these changes and they work great!

Yes, I agree. Custom VariableAttributes are certainly needed more often than custom variableType or referenceType.

I have two ideas for convenient overloads that would be neat, but aren't essential:
1. When calling `setValue()` that I can also directly pass an `uint8_t`, `opcua::String`, etc., it automatically infers the type and creates the `opcua::Variant` with the matching datatype accordingly:
`opcua::VariableAttributes().setValue(static_cast<uint8_t>(0xAF));` -> same as -> `opcua::VariableAttributes().setValue(opcua::Variant::fromScalar(static_cast<uint8_t>(0xAF)));`
2. When calling `setDataType()` that there is an generic overload where I don't have to specify the DataTypeId but instead it is inferred from the generic type: 
`setDataType<int32_t>()` -> same as -> `setDataType(opcua::DataTypeId::Int32)`
`setDataType<opcua::String>()` -> same as -> `setDataType(opcua::DataTypeId::String)`

What do you think?

lukasberbuer, 2023-07-29:
Good ideas @johanneskopf!

1. Just added the overloads `VariableAttributes::setValueScalar` and `VariableAttributes::setValueArray`. I didn't want to overload `setValue` to avoid ambiguous overloads supporting both scalar and array values: https://github.com/open62541pp/open62541pp/blob/feat/initial-node-attributes/include/open62541pp/types/Composed.h#L212-L220.
2. I like the idea and will add this in a later PR. I just created the constexpr helper `details::guessDataType` in #76 which we should use for this solution as well

So I guess we are ready to merge?

closed 2023-07-29 by lukasberbuer

johanneskopf, 2023-07-30:
@lukasberbuer glad you like them!
1. `VariableAttributes::setValueScalar` is exactly what I wanted, shorter code while still maintaining readability. Just awesome, already using it.
2. Sounds great, looking forward to using that too.

Yes, ready to merge!