open62541pp/open62541pp #62: Variable value callbacks support

ussaka, 2023-07-19:
Hi,

Could value callbacks (`UA_Server_setVariableNode_valueCallback`) be added?

ussaka, 2023-07-20:
> Hi @ussaka, thank you for the request. Indeed, this feature needed to be added! Just added a draft implementation in PR #63.
> 
> I'm unsure if we should provide the `NodeId` as a parameter in the callback functions. Usually, it's easy to just capture the node id with lambdas. Any opinions?
> 
> Link to example: https://github.com/open62541pp/open62541pp/blob/feat/variable-value-callback/examples/server_valuecallback.cpp

Not really sure. I think current implementation is fine.

lukasberbuer, 2023-07-20:
This is how the example looks with the current solution:

```cpp
valueCallback.onBeforeRead = [&](const opcua::NodeId& id, const opcua::DataValue& value) {
    server.getNode(id).writeScalarCopy(opcua::DateTime::now());
};
```

Same example without `NodeId` parameter:

```cpp
valueCallback.onBeforeRead = [&](const opcua::DataValue& value) {
    currentTimeNode.writeScalarCopy(opcua::DateTime::now());
};
```

If we don't find a strong argument for providing the `NodeId` as callback parameter, I would go with the second/simpler option.
To make use of the `NodeId`, the `Server` instance needs so be captured anyway.