open62541pp/open62541pp #69: Server access control

johanneskopf, 2023-07-23:
Hi,
I need to be able to control the access to the server nodes, methods, etc. Hereby, open62541 provides many callbacks in the UA_AccessControl class. To achieve my goal, I need to overwrite them. Optimally, the wrapper would provide some functionality to do so.

For now, I tried to use the open62541 directly to achieve this, but those callbacks all have a `UA_Server *server` as argument. E.g.
```cpp
UA_UInt32 (*getUserRightsMask)(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_NodeId *nodeId, void *nodeContext);
```
How can I convert the `UA_Server *server` argument to an `opcua::Server` object? Getting the `UA_Server *server` to an `opcua::Server` variable is possible by using the method `UA_Server* handle() noexcept;`, but I need it the other way round. That way, I could use the opcua::Server functionality inside the callback.

Example code how to override the callbacks:
```cpp
#define UA_ACCESSLEVELMASK_ALL_BYTE 0xFF
#define UA_ACCESSLEVELMASK_ALL_UINT32 0xFFFFFFFF

static UA_UInt32 getUserRightsMaskOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_NodeId *nodeId, void *nodeContext) {
    printf("getUserRightsMaskOwn called\n");
    return UA_ACCESSLEVELMASK_ALL_UINT32;
}

static UA_Byte getUserAccessLevelOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_NodeId *nodeId, void *nodeContext) {
    printf("getUserAccessLevelOwn called\n");
    return UA_ACCESSLEVELMASK_ALL_BYTE;
}

static UA_Boolean getUserExecutableOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_NodeId *methodId, void *methodContext) {
    printf("getUserExecutableOwn called\n");
    return true;
}

static UA_Boolean getUserExecutableOnObjectOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_NodeId *methodId, void *methodContext, const UA_NodeId *objectId, void *objectContext) {
    printf("getUserExecutableOnObjectOwn\n");
    return true;
}

static UA_Boolean allowAddNodeOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_AddNodesItem *item) {
    printf("allowAddNodeOwn\n");
    return true;
}

static UA_Boolean allowAddReferenceOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_AddReferencesItem *item) {
    printf("allowAddReferenceOwn\n");
    return true;
}

static UA_Boolean allowDeleteNodeOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_DeleteNodesItem *item) {
    printf("allowDeleteNodeOwn\n");
    return true;
}

static UA_Boolean allowDeleteReferenceOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_DeleteReferencesItem *item) {
    printf("allowDeleteReferenceOwn\n");
    return true;
}

static UA_Boolean allowBrowseNodeOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_NodeId *nodeId, void *nodeContext) {
    printf("allowBrowseNodeOwn\n");
    return true;
}

int main(int argc, char *argv[]) {
    opcua::Server server(4840 /* port */);
    server.setApplicationName("ServerName");

    // Get server config
    UA_ServerConfig* config = getConfig(&server);

    // Access control for all nodes
    config->accessControl.getUserRightsMask = getUserRightsMaskOwn;
    // Additional access control for variable nodes
    config->accessControl.getUserAccessLevel = getUserAccessLevelOwn;
    // Additional access control for method nodes
    config->accessControl.getUserExecutable = getUserExecutableOwn;
    // Additional access control for calling a method node in the context of a specific object
    config->accessControl.getUserExecutableOnObject = getUserExecutableOnObjectOwn;
    // Allow adding a node
    config->accessControl.allowAddNode = allowAddNodeOwn;
    // Allow adding a reference
    config->accessControl.allowAddReference = allowAddReferenceOwn;
    // Allow deleting a node
    config->accessControl.allowDeleteNode = allowDeleteNodeOwn;
    // Allow deleting a reference
    config->accessControl.allowDeleteReference = allowDeleteReferenceOwn;
    // Allow browsing a node
    config->accessControl.allowBrowseNode = allowBrowseNodeOwn;
}
```

lukasberbuer, 2023-07-25:
Hi @johanneskopf,

you can attach anything you like to the `context` of `UA_AccessControl`:

```cp
struct UA_AccessControl {
    void *context;
    void (*clear)(UA_AccessControl *ac);
    // ...
};
```

To use the `Server` instance within the access callbacks, you can do following (didn't test it, might have some typos):

```cpp
auto* config = UA_Server_getConfig(server.handle());
config->accessControl->context = &server;

// use server instance in callback
static UA_Boolean allowAddNodeOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_AddNodesItem *item) {
    auto* server = static_cast<Server*>(ac->context);
    // ...
    return true;
}
```

I'm currently writing a `AccessControl` wrapper in PR #74 passing directly a `Server` reference to the callbacks. I hope this simplifies it for you.

johanneskopf, 2023-07-30:
Thanks for the tip with the use of `context` of `UA_AccessControl` @lukasberbuer  !

For reference, here your code tested and updated:
```cpp
auto* config = UA_Server_getConfig(server.handle());
config->accessControl.context = &server;

// use server instance in callback
static UA_Boolean allowAddNodeOwn(UA_Server *server, UA_AccessControl *ac, const UA_NodeId *sessionId, void *sessionContext, const UA_AddNodesItem *item) {
    auto* serverWrapper = static_cast<opcua::Server*>(ac->context);
    // ...
    return true;
}
```

The wrapper looks great so far! Also easy to use. Already tried to derive my custom class and that works great and frictionless. Looking forward to the implementation of the "Hook AccessControlBase& into Server" part.

closed 2023-08-15 by lukasberbuer