open62541pp/open62541pp #70: Add custom datatype

johanneskopf, 2023-07-23:
I need to add a custom data type to the server, and then use this custom data type to add a node of that type. It should be possible to read/write to/from this node from the server and client side. I have created an example (only server side) that is implemented using the open62541 library directly. Now I want to create an improved version using this wrapper, but I could not find out how to achieve a similar functionality. Is this possible?

```cpp
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <csignal>

char localeStr[] = "en-US";

static volatile UA_Boolean running = true;

static void stopHandler(int sig) {
    running = false;
}

static UA_NodeId generateNodeIdWithString(UA_UInt16 namespaceIndex, char *string) {
    UA_String uaString = {sizeof(string) - 1, reinterpret_cast<UA_Byte *>(string)};
    UA_NodeId node;
    node.namespaceIndex = namespaceIndex;
    node.identifierType = UA_NODEIDTYPE_STRING;
    node.identifier.string = uaString;
    return node;
}

#define ID_DATA_TYPE_TWO_STRINGS 4400
#define BINARY_ENCODING_ID_TWO_STRINGS  1
const UA_NodeId TwoStringsVariableTypeId = {1, UA_NODEIDTYPE_NUMERIC, {4401}};

static char twoStringsNodeIdStr[] = "TwoStrings";
UA_NodeId twoStringsNodeId = generateNodeIdWithString(1, twoStringsNodeIdStr);

typedef struct {
    UA_String a;
    UA_String b;
} TwoStrings;

static UA_DataTypeMember TwoStrings_members[2] = {
        {UA_TYPENAME("a") &UA_TYPES[UA_TYPES_STRING], 0, false, false},
        {UA_TYPENAME("b") &UA_TYPES[UA_TYPES_STRING], 0, false, false}
};

static const UA_DataType TwoStringsType = {
        UA_TYPENAME("TwoStrings") {1, UA_NODEIDTYPE_NUMERIC, {ID_DATA_TYPE_TWO_STRINGS}}, {1, UA_NODEIDTYPE_NUMERIC, {BINARY_ENCODING_ID_TWO_STRINGS}}, sizeof(TwoStrings), UA_DATATYPEKIND_STRUCTURE,
        false, false, 2, TwoStrings_members
};

static void addDataTypeTwoStrings(UA_Server *server) {
    UA_DataTypeAttributes attr = UA_DataTypeAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT(localeStr, "TwoStrings Type");
    UA_Server_addDataTypeNode(server, TwoStringsType.typeId, UA_NODEID_NUMERIC(0, UA_NS0ID_STRUCTURE), UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE), UA_QUALIFIEDNAME(1, "TwoStrings"), attr, nullptr,nullptr);
}

static void addVariableTypeTwoStrings(UA_Server *server) {
    UA_VariableTypeAttributes dattr = UA_VariableTypeAttributes_default;
    dattr.description = UA_LOCALIZEDTEXT(localeStr, "TwoStrings");
    dattr.displayName = UA_LOCALIZEDTEXT(localeStr, "TwoStrings");
    dattr.dataType = TwoStringsType.typeId;
    dattr.valueRank = UA_VALUERANK_ANY; // represents an array

    TwoStrings twoStrings;
    memset(&twoStrings, 0, sizeof(TwoStrings));
    twoStrings.a = UA_STRING("");
    twoStrings.b = UA_STRING("");
    UA_Variant_setScalar(&dattr.value, &twoStrings, &TwoStringsType);

    UA_Server_addVariableTypeNode(server, TwoStringsVariableTypeId, UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE), UA_QUALIFIEDNAME(1, "TwoStrings"),UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), dattr, nullptr, nullptr);
}

static void addNodeTwoStrings(UA_Server *server) {
    size_t array_size = 2;
    auto *twoStrings = (TwoStrings *) UA_Array_new(array_size, &TwoStringsType);
    twoStrings[0].a = UA_STRING_ALLOC("1:a");
    twoStrings[0].b = UA_STRING_ALLOC("1:b");
    twoStrings[1].a = UA_STRING_ALLOC("2:a");
    twoStrings[1].b = UA_STRING_ALLOC("2:b");
    UA_VariableAttributes vattr = UA_VariableAttributes_default;
    vattr.description = UA_LOCALIZEDTEXT(localeStr, twoStringsNodeIdStr);
    vattr.displayName = UA_LOCALIZEDTEXT(localeStr, twoStringsNodeIdStr);
    vattr.dataType = TwoStringsType.typeId;
    vattr.valueRank = UA_VALUERANK_ONE_OR_MORE_DIMENSIONS;
    vattr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    UA_UInt32 myArrayDimensions[1] = {array_size};
    vattr.value.arrayDimensions = myArrayDimensions;
    vattr.value.arrayDimensionsSize = 1;
    UA_Variant_setArray(&vattr.value, twoStrings, array_size, &TwoStringsType);

    UA_StatusCode ret = UA_Server_addVariableNode(server, twoStringsNodeId, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),UA_QUALIFIEDNAME(1, twoStringsNodeIdStr), TwoStringsVariableTypeId, vattr, nullptr, nullptr);
    if (ret != UA_STATUSCODE_GOOD) {
        printf("Error adding node");
    }
    UA_Array_delete(twoStrings, array_size, &TwoStringsType);
}

int main() {
    // set up stop-handler
    auto retSignal = signal(SIGINT, stopHandler);
    if (retSignal == SIG_ERR) {
        printf("signal(...) call failed\n");
        return EXIT_FAILURE;
    }
    retSignal = signal(SIGTERM, stopHandler);
    if (retSignal == SIG_ERR) {
        printf("signal(...) call failed\n");
        return EXIT_FAILURE;
    }

    // create server with default config
    UA_Server *server = UA_Server_new();
    UA_ServerConfig *config = UA_Server_getConfig(server);
    UA_ServerConfig_setDefault(config);

    // add custom datatype
    auto *types = (UA_DataType *) UA_malloc(1 * sizeof(UA_DataType));
    types[0] = TwoStringsType;
    auto *twoStringsMembers = (UA_DataTypeMember *) UA_malloc(sizeof(UA_DataTypeMember) * 2);
    twoStringsMembers[0] = TwoStrings_members[0];
    twoStringsMembers[1] = TwoStrings_members[1];
    types[0].members = twoStringsMembers;
    UA_DataTypeArray customDataTypes = {config->customDataTypes, 1, types};
    config->customDataTypes = &customDataTypes;
    addDataTypeTwoStrings(server);
    addVariableTypeTwoStrings(server);

    // add custom datatype node
    addNodeTwoStrings(server);

    // run server
    UA_StatusCode ret = UA_Server_run(server, &running);

    // cleanup server
    UA_Server_delete(server);

    // cleanup custom data types
    UA_free(twoStringsMembers);
    UA_free(types);

    return ret == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

Of course, I could for now just use the `getConfig` method and use the open62541 library directly, but then is it possible to read/write from the node using the new wrapper interface?

```cpp
inline static UA_ServerConfig* getConfig(opcua::Server* server) noexcept {
    return UA_Server_getConfig(server->handle());
}

int main() {
    opcua::Server server(4840 /* port */);
    server.setApplicationName("TwoStrings_Server");
    server.setLogger([](auto level, auto category, auto msg) {
        std::cout << "[" << opcua::getLogLevelName(level) << "] " << "[" << opcua::getLogCategoryName(category) << "] " << msg << std::endl;
    });

    // get server config
    UA_ServerConfig* config = getConfig(&server);

    // add custom datatype
    auto *types = (UA_DataType *) UA_malloc(1 * sizeof(UA_DataType));
    types[0] = TwoStringsType;
    auto *twoStringsMembers = (UA_DataTypeMember *) UA_malloc(sizeof(UA_DataTypeMember) * 2);
    twoStringsMembers[0] = TwoStrings_members[0];
    twoStringsMembers[1] = TwoStrings_members[1];
    types[0].members = twoStringsMembers;
    UA_DataTypeArray customDataTypes = {config->customDataTypes, 1, types};
    config->customDataTypes = &customDataTypes;
    addDataTypeTwoStrings(server.handle());
    addVariableTypeTwoStrings(server.handle());

    // add server node
    addNodeTwoStrings(server.handle());

    server.run();
}
```

lukasberbuer, 2023-07-29:
Hi @johanneskopf, I started the implementation in the draft PR #76.

My personal pain with custom types is the complex definition of  `UA_DataType` and the members of type `UA_DataTypeMember`. Most attributes can be derived from the type itself, including:
- `UA_DataType::memSize`
- `UA_DataType::typeKind`
- `UA_DataType::pointerFree`
- `UA_DataTypeMember::padding`
- `UA_DataTypeMember::isArray`
- `UA_DataTypeMember::isOptional`

If drafted a `DataTypeBuilder` so easily create those definitions, e.g.:
```cpp
struct TwoStrings {
    UA_String a;
    UA_String b;
};

auto dt = DataTypeBuilder<TwoStrings>::createStructure("TwoStrings", {1, 4401}, {1, 1})
    .addMember<&TwoStrings::a>("a", UA_TYPES[UA_TYPES_STRING])
    .addMember<&TwoStrings::b>("b", UA_TYPES[UA_TYPES_STRING])
    .build();
```

BUT we could also use the wrapper types `String` instead of `UA_String` without changing anything else. I wrote a little bit about using wrapper types in custom types in #76.

Next steps:
- [ ] `Node::addDataType` / `services::addDataType`
- [ ] `Server::addCustomTypes`
- [ ] `Client::addCustomTypes`
- [ ] Read/write custom data types (this is mainly a topic of optimizing the `ExtensionObject` interface)
- [ ] Example

I will try to push this topic and get it done quickly. Any feedback is welcome :)

lukasberbuer, 2023-07-31:
Hi @johanneskopf, the PR #76 is not ready to be tested in reviewed.
I reproduced the custom datatype examples of open62541. I guess they showcase the new functionalities quite good:
- [`custom_datatypes.h`](https://github.com/open62541pp/open62541pp/blob/feat/custom-datatype/examples/custom_datatypes/custom_datatypes.h) for data type definitions
- [`server_custom_datatypes.cpp`](https://github.com/open62541pp/open62541pp/blob/feat/custom-datatype/examples/custom_datatypes/server_custom_datatypes.cpp) for server-side implementation
- [`client_custom_datatypes.cpp`](https://github.com/open62541pp/open62541pp/blob/feat/custom-datatype/examples/custom_datatypes/client_custom_datatypes.cpp) for client-side implementation

Compared to the original open62541 implementation, the line number for the examples could be greatly reduced:
- [`custom_datatype.h`](https://github.com/open62541/open62541/blob/v1.3.6/examples/custom_datatype/custom_datatype.h) from 193 to 76 lines
- [`server_types_custom.c`](https://github.com/open62541/open62541/blob/v1.3.6/examples/custom_datatype/server_types_custom.c) from 310 to 115 lines
- [`client_types_custom.c`](https://github.com/open62541/open62541/blob/v1.3.6/examples/custom_datatype/client_types_custom.c) from 99 to 68 lines

I'm quite happy with the outcome and hope you like it too!

johanneskopf, 2023-08-01:
Hi @lukasberbuer ,
Great work again! The readability of the code I updated to use your wrapper has already improved a lot. I was also able to reduce my line count by quite a bit. I like it a lot!

I got a warning when compiling at DataType.cpp. I added a comment with the fix to the corresponding line at the PR https://github.com/open62541pp/open62541pp/pull/76/.

Currently, I try to add a node that represents an array of a custom datatype. Problem is I couldn't figure out how to assign e.g. a vector of the custom datatype to an opcua::Variant. It's quite easy to do so using the built-in datatypes though, but with the custom datatypes I believe it is missing an overloaded method with the just introduced DataType. As a result my example does not compile currently. Is it possible to add such an overload?

Here is the code:
```cpp
constexpr UA_UInt32 ID_DATA_TYPE_CUSTOM_DATA        = 4443;
constexpr UA_UInt32 ID_VARIABLE_TYPE_CUSTOM_DATA    = 4444;

struct CustomData{
    opcua::String string1;
    opcua::String string2;
};

const opcua::DataType& getCustomDataType() {
    static const opcua::DataType dt =
            opcua::DataTypeBuilder<CustomData>::createStructure("CustomData", {1, ID_DATA_TYPE_CUSTOM_DATA}, {1, 2})
                    .addField<&CustomData::string1>("string1")
                    .addField<&CustomData::string2>("string2")
                    .build();
    return dt;
}

// Get custom type definitions from common header
const auto& dataTypeCustomData = getCustomDataType();

// Provide custom data type definitions to server
server.setCustomDataTypes({dataTypeCustomData});

// Add data type node
auto nodeStructureDataType = server.getNode(opcua::DataTypeId::Structure);
nodeStructureDataType.addDataType(dataTypeCustomData.getTypeId(), "CustomData");

// Add variable type node
auto nodeBaseDataVariableType = server.getNode(opcua::VariableTypeId::BaseDataVariableType);
auto nodeVariableTypeCustomData = nodeBaseDataVariableType.addVariableType(
        {1, ID_VARIABLE_TYPE_CUSTOM_DATA},
        "CustomDataType",
        opcua::VariableTypeAttributes{}
                .setDataType(dataTypeCustomData.getTypeId())
                .setValueRank(opcua::ValueRank::Scalar)
                .setValueScalar(CustomData{}, dataTypeCustomData)
);

// Add variable node with a default value
const CustomData customData{
        opcua::String("string 1 data"),
        opcua::String("string 2 data"),
};

std::vector<CustomData> customDataVec {customData};

nodeObjects.addVariable(
        {1, "CustomDataVec"},
        "CustomData",
        opcua::VariableAttributes{}
                .setDataType(dataTypeCustomData.getTypeId())
                .setValueRank(opcua::ValueRank::OneDimension)
                .setValueArray(customDataVec, dataTypeCustomData), // this line throws the compile error
        nodeVariableTypeCustomData.getNodeId()
);
```

lukasberbuer, 2023-08-02:
Hi @johanneskopf, thanks for catching this bug.. two copy & paste errors in the new `Variant::fromArray` overloads. Also, the `Variant::fromArray` overload for `std::vector<T>` and a custom data type was missing.

Should be fixed in the lastest commit of #76. Can you please check again?

johanneskopf, 2023-08-02:
Hi @lukasberbuer , you are welcome! This fixed the compile error. Now I run into a runtime error here, will look into that later.

I tested some more and the `setArray()` and `setArrayCopy()` Variant seem to miss the same changes.
```cpp
opcua::Variant variant;
variant.setArray(customDataVec, dataTypeCustomData); // does not compile
variant.setArrayCopy(customDataVec, dataTypeCustomData); // does not compile
```

lukasberbuer, 2023-08-02:
Let me figure out the runtime issue. I just realized, that also the example `server_custom_datatypes` isn't working anymore. I think I messed something up with the master merges.

lukasberbuer, 2023-08-02:
> I tested some more and the `setArray()` and `setArrayCopy()` Variant seem to miss the same changes.
> 
> ```c++
> opcua::Variant variant;
> variant.setArray(customDataVec, dataTypeCustomData); // does not compile
> variant.setArrayCopy(customDataVec, dataTypeCustomData); // does not compile
> ```

Hi @johanneskopf, commit [66b0f0b](https://github.com/open62541pp/open62541pp/commit/66b0f0b6f05fba73f0b88a501bac776329199bdf) should fix this.

johanneskopf, 2023-08-02:
> Hi @johanneskopf, commit [66b0f0b](https://github.com/open62541pp/open62541pp/commit/66b0f0b6f05fba73f0b88a501bac776329199bdf) should fix this.

Hi @lukasberbuer ,
Thanks for the fix, this code compiles now!

johanneskopf, 2023-08-02:
I managed to track down the culprit that causes `BadTypeMismatch` in https://github.com/open62541pp/open62541pp/issues/70#issuecomment-1661867506. Namely, the array dimensions must be set accordingly. The following updated code does not throw a runtime error when creating the `CustomDataVec` node, regardless of whether you set the variable type to `ValueRank::Any` or `ValueRank::ScalarOrOneDimension`:
```cpp
const std::vector<CustomData> customDataVec{customData, customData};
    nodeObjects.addVariable(
            {1, "CustomDataVec"},
            "CustomDataVec",
            opcua::VariableAttributes{}
                    .setDataType(dataTypeCustom.getTypeId())
                    .setArrayDimensions({2})
                    .setValueRank(opcua::ValueRank::OneDimension)
                    .setValueArray(customDataVec, dataTypeCustom),
            nodeVariableTypeCustom.getNodeId()
    );
``` 
I also verified that the content of the `CustomDataVec` node is set correctly using UaExpect.

Now I try to update the client side. I ran into two issues here:
1.  How to verify that the elements of a variant array have a specific data type? I optimally want to make sure that there are CustomData struct elements inside the variant, before accessing the values.
2. I was not able to get any meaningful data at the client side from the data read from the `CustomDataVec` node. Any idea?

```cpp
#include <iostream>
#include "open62541pp/open62541pp.h"

struct CustomData {
    opcua::String string1;
    opcua::String string2;
};

const opcua::DataType& getCustomDataType() {
    static const opcua::DataType dt =
            opcua::DataTypeBuilder<CustomData>::createStructure("CustomData", {1, 5000}, {1, 2})
                    .addField<&CustomData::string1>("string1")
                    .addField<&CustomData::string2>("string2")
                    .build();
    return dt;
}

int main() {
    opcua::Client client;

    // Get custom type definitions from common header
    const auto& dataTypeCustom = getCustomDataType();

    // Provide custom data type definitions to client
    client.setCustomDataTypes({dataTypeCustom});

    // Connect to server
    client.connect("opc.tcp://localhost:4840");

    // Read custom data type node and print values
    auto nodeCustomDataVec = client.getNode({1, "CustomDataVec"});
    opcua::Variant variant;
    nodeCustomDataVec.readValue(variant);
    std::cout << "Variant.isType(dataTypeCustom) = " << variant.isType(dataTypeCustom) << std::endl;
    std::cout << "Variant.isArray() = " << variant.isArray() << std::endl;
    std::cout << "Variant.getArrayLength() = " << variant.getArrayLength() << std::endl;
    if (variant.isArray()) {  // TODO: How to make sure the type of the array elements is dataTypeCustom?
        auto* customDataArr = static_cast<CustomData*>(variant.getArray()); // TODO: Content inside here is invalid, why?
        std::vector<CustomData> customDataVec(customDataArr, customDataArr + variant.getArrayLength());
        std::cout << "CustomDataVec:" << std::endl;
        for (const auto& customData : customDataVec) {
            std::cout << "- string1 = " << customData.string1.get() << std::endl;
            std::cout << "- string2 = " << customData.string2.get() << std::endl;
        }
    }
}
```

lukasberbuer, 2023-08-02:
Nice discovery @johanneskopf. Setting the array dimensions makes sense as a prerequisite 💡

Regarding your client questions:

> 1. How to verify that the elements of a variant array have a specific data type? I optimally want to make sure that there are CustomData struct elements inside the variant, before accessing the values.

The variant class has a `isType` method with different overloads. You can just use following snippet:
```cpp
if (variant.isType(dataTypeCustom) {
    auto* customDataArr = static_cast<CustomData*>(variant.getArray());
   // ...
}
```

`variant.isArray()` is optional, because `Variant::getArray` will throw a exception anyway if the variant does not contain an array. `isArray` might be a good solution to avoid exceptions.

> 2. I was not able to get any meaningful data at the client side from the data read from the CustomDataVec node. Any idea?

I just added array of custom types to the example: https://github.com/open62541pp/open62541pp/pull/76/commits/307f4128be6a1aa933bf259f403392aeb26f65f5

The problem is, that the actual data is a array of `ExtensionObject` objects. They itself contain the custom data. So we don't have a contiguous memory region in memory with just the data. I don't know why open62541 unwraps the `ExtensionObject`s for scalar values but not for arrays. According to the OPC UA standard, a variant may only contain built-in types so the behaviour with scalar values seems to be out of the spec. Still I like it because it simplifies data handling. Maybe you have more insights?

> 3. Is there a more straight-forward way to convert the variant to a vector?

Currently not. Returning a vector always involves a copy so I didn't want this as the default behavior. What kind of API would you propose?

lukasberbuer, 2023-08-03:
I agree. Unwrapping an array with ExtensionObjects is a lot of boilerplate code. This should be handled by open62541pp more efficiently. Your implementation looks fine, although I would prefer not to introduce more method, but maybe extend an existing method or add an overload?

In the end, we want to have a copy of the underlying array, to `Variant::getArrayCopy` would be a good fit. The implementation could be extended by checking if the current data type is an ExtensionObject. If yes, do the unwrapping.

```cpp
template <typename T>
std::vector<T> Variant::getArrayCopy() const {
    checkIsArray();
    // if data type is extension object -> go another route
    checkReturnType<T>();
    return detail::fromNativeArray<T>(handle()->data, handle()->arrayLength, *getDataType());
}
```

I see one major problem: the custom data objects inside of the array might need manual deletion, e.g. if the custom data type is a struct with a `UA_String` member. If we unwrap the array elements and copy them in a vector, we will get a memory leak.
Ideally, the vector should be aware of the items types and do the clean up.

But maybe there is a solution in the future: I was already drafting a generic `Array` class for native open62541 arrays. My main goal was to eliminate copies in the getter methods of wrapper types, e.g. `std::vector<UserTokenPolicy> EndpointDescription::getUserIdentityTokens()`
For wrapper, we already have the `asWrapper` helper to return pointer/references to native objects as wrapper types. I want to have the same for array types. This `Array` class would be type aware and can to the cleanup.

**My proposal**: Postpone this feature until `Array` is ready. The ugly code snippet in the `client_custom_types.cpp` example will be a reminder :)

Are we ready to merge #76 so far?

johanneskopf, 2023-08-03:
@lukasberbuer  That `Array` feature is compelling and so are your points that led to it! I am all for exending the existing `getArrayCopy` method. Haha, yeah, this snippet is really not that nice to look at. You really do magic here to make the open62541 library more accessible.

Yes, ready to merge!

closed 2023-08-04 by lukasberbuer

johanneskopf, 2023-09-01:
@lukasberbuer awesome, that's a feature I am really looking forward to! Thank you for pointing this out. I am curious now when the 1.4 release will happen and how much work it will be to adapt your wrapper to the changes. But that's probably both hard to tell.