open62541pp/open62541pp #71: callbacks missing on server: onConnected/onDisconnected/onSessionActivated/onSessionClosed

johanneskopf, 2023-07-23:
Client side has the following callback implemented:
```cpp
void Client::onConnected(StateCallback callback) {
    setStateCallback(getContext(), ClientState::Connected, std::move(callback));
}

void Client::onDisconnected(StateCallback callback) {
    setStateCallback(getContext(), ClientState::Disconnected, std::move(callback));
}

void Client::onSessionActivated(StateCallback callback) {
    setStateCallback(getContext(), ClientState::SessionActivated, std::move(callback));
}

void Client::onSessionClosed(StateCallback callback) {
    setStateCallback(getContext(), ClientState::SessionClosed, std::move(callback));
}
```

Now I need those callbacks on the server side too. Is it possible to implement them? I looked at the wrapper code and found that a similar implementation as on the client side is not possible, as the client side has a state-callback (ua_client.c):
```cpp
if(client->config.stateCallback)
  client->config.stateCallback(client, client->channel.state, client->sessionState, client->connectStatus);
```
whereas on the server side there isn't. This seems to only work by overriding the callbacks in the access control of the config:
```cpp
config->accessControl.activateSession = ...;
config->accessControl.closeSession = ...;
```
It would be great to have those callbacks too in the wrapper.

johanneskopf, 2023-07-23:
Given this also is part of the server-side access-control, it is related to https://github.com/open62541pp/open62541pp/issues/69.

lukasberbuer, 2023-07-24:
Hi @johanneskopf. This is a tricky one. We could hook into the `UA_AccessControl` callbacks `activateSession` and `closeSession`, but those are usually used to provide access and clean-up session-related storage. We have to wrap the `UA_AccessControl` to call both the state callback and the actual access callback. But we can not prevent users from using the C-API and providing their own access control and un-hooking our wrapper.
Both `onConnected` and `onDisconnected` seem to be impossible - I don't see any spot where to hook in.
May I ask what you need those state callbacks for?

johanneskopf, 2023-07-25:
Thank you for taking a look @lukasberbuer . Basically, I need a counter on the server side that always tells how many clients are currently connected and still active. But just the activateSession and closeSession callbacks are not enough. I created an issue two months ago hoping to get more information/ideas in the open62541 repository:  https://github.com/open62541/open62541/issues/5788

lukasberbuer, 2023-08-06:
Hi @johanneskopf,
I just stumbled the `UA_Server_getStatistics` method: https://github.com/open62541/open62541/blob/v1.3.6/include/open62541/server.h#L1760-L1767

```cpp
// common.h
typedef struct {
    size_t currentSessionCount;
    size_t cumulatedSessionCount;
    size_t securityRejectedSessionCount; /* only used by servers */
    size_t rejectedSessionCount;
    size_t sessionTimeoutCount;          /* only used by servers */
    size_t sessionAbortCount;            /* only used by servers */
} UA_SessionStatistics;

// server.h
typedef struct {
   UA_NetworkStatistics ns;
   UA_SecureChannelStatistics scs;
   UA_SessionStatistics ss;
} UA_ServerStatistics;

UA_ServerStatistics UA_EXPORT
UA_Server_getStatistics(UA_Server *server);
```

This would provide you the number of active sessions without any callback mechanisms. Just as a 2nd option.. PR #74 will be finalized soon.

closed 2023-08-15 by lukasberbuer