open62541pp/open62541pp #104: UA_Range

lms-hfr, 2023-09-06:
Hello,

I have  question regarding the UA_Range.
I would like to add a range variable to my server. Adding a variable is well described in the /examples/server.cpp. 
But this does not work for the UA_Range because I do not have the NodeId of the Range nor is the Range defined in opcua::Type::.... .

How do I correctly add a range variable to my server? 

I added a Range by hardcoding the NodeId, buth this can't be the correct way..

`    // add Range
    const UA_Range initVal = {1.0, 4.0};
    const opcua::NodeId nid_range = {0, 884};
    opcua::NodeId nodeId_nr = {1, "node.range"};
    std::string_view browsename_nr = "range";
    auto nrange_node = myDRCNode.addVariable(nodeId_nr, browsename_nr);
    nrange_node.writeDataType(nid_range);
    nrange_node.writeDisplayName({"en-US", "range"});
    nrange_node.writeDescription({"en-US", "the range"});
    nrange_node.writeValueScalar(initVal);

`

Thanks in advance

lukasberbuer, 2023-09-06:
Hi @lms-hfr, good question!

I created a minimal example with two possible solutions:

```cpp
#include "open62541pp/open62541pp.h"

int main() {
    opcua::Server server;
    opcua::Node rangeNode = server.getObjectsNode().addVariable({1, "node.range"}, "range");

    // 1. option: use UA_DataType
    rangeNode.writeDataType(opcua::NodeId(UA_TYPES[UA_TYPES_RANGE].typeId));
    const auto rangeVariant = opcua::Variant::fromScalar(UA_Range{1.0, 4.0}, UA_TYPES[UA_TYPES_RANGE]);
    rangeNode.writeValue(rangeVariant);

    // 2. option: use TypeWrapper
    using Range = opcua::TypeWrapper<UA_Range, UA_TYPES_RANGE>;
    rangeNode.writeDataType<Range>();
    Range rangeWrapper(UA_Range{1.0, 4.0});
    // or set values after construction
    rangeWrapper->low = 1.0;
    rangeWrapper->high = 4.0;
    rangeNode.writeValueScalar(rangeWrapper);

    server.run();
}
```

`Variant` provides overloads to specify the data type, e.g.:
- `Variant::fromScalar(T &value, const UA_DataType &dataType)`
- `Variant::setScalarCopy (const T &value, const UA_DataType &dataType)`

`Node` does have those overloads for `Node::writeValueScalar` at the moment This could be a good addition. What do you think?

lukasberbuer, 2023-09-07:
Btw, there is a nicer way for option 1 using `opcua::DataTypeId` (numeric node ids of namespace 0, implicit conversion defined for `NodeId`):

```cpp
    // 1. option: use UA_DataType
    rangeNode.writeDataType(opcua::DataTypeId::Range);
    const auto rangeVariant = opcua::Variant::fromScalar(UA_Range{1.0, 4.0}, UA_TYPES[UA_TYPES_RANGE]);
    rangeNode.writeValue(rangeVariant);
```

lms-hfr, 2023-09-07:
Hi Lukas,

thanks for your very good answer. 

Concerning the overload for the Node: 
Does a Node always have a defined type? Then there shouldn't be any need for a Node::writeValueScalar(const T &value, const UA_DataType &dataType) overload.

I don't think we need one method to set the datatype and the value at once. That can be done in two separate lines. It could be confusing if one would write data of type 'A' in a Node that already has type 'B'.

Can you think of a other situation where the overloads come in handy?

Greetings
Harald

lukasberbuer, 2023-09-07:
Hi Harald,

a variable node can have value attributes of different types if unspecified. Only if the data type is explicitly defined, only value attributes of that specific type can be written. Following example showcases this behaviour:

```cpp
#include "open62541pp/open62541pp.h"

int main() {
    opcua::Server server;
    opcua::Node node = server.getObjectsNode().addVariable({1, "Example"}, "Example");

    // the data type of the variable node is undefined -> values of different can be written to the node
    node.writeValueScalar(42);  // int
    node.writeValueScalar(11.11);  // double
    node.writeValueScalar("test123");

    // now we define the variable data type to be of type string
    node.writeDataType<opcua::String>();

    // any other value types written to the node will generate an exception
    node.writeValueScalar(42);  // throws BadTypeMismatch

    server.run();
}
```

But we need to differentiate between the data type of the node and the `UA_DataType` necessary to write an unknown type to a variant (which then can be written to a node). The `UA_DataType` described the memory layout of the type and how to encode/decode this. A `Variant` always stores the scalar/array and the corresponding `UA_DataType`.
Open62541 stores the `UA_DataType` definitions for all defined types in the array `UA_TYPES`. Each type has an associated type index, e.g. `UA_TYPES_STRING`. The data type can be accessed with `UA_TYPES[UA_TYPES_STRING]`.

The `TypeWrapper` class takes two template types, the wrapped type and the corresponding type index. So if we pass the wrapped type, e.g. `TypeWrapper<UA_String, UA_TYPES_STRING>` to the method `Node::writeValueScalar`, the data type can be retrieved from the wrapper class. If you just pass `UA_String` to the method, you manually need to specify the data type.

I hope this helps.

lms-hfr, 2023-09-13:
Hi Lukas,

Sorry for the late reply.

You said: 
> But we need to differentiate between the data type of the node and the UA_DataType necessary to write an unknown type to a variant (which then can be written to a node).

I thought a node has always the type _variant_. And the _variant_ can contain any datatype as you described.

Even, when writing a scalar to a node like in your example

`node.writeValueScalar(42);  // int`

the `writeValueScalar()` method creates a _variant_ internally.

This seems to work with _UA_Range_ as well, but I don’t understand how the

`constautovariant=Variant::fromScalar<T>(const_cast<T&>(value));`

can work with a complex datatype like _UA_Range_.

We can of course add two methods like:

```
/// Write scalar to variable node with UA_DataType.
///@returnCurrent node instance to chain multiple methods (fluent interface)
template<typenameT>
Node&writeValueScalar(constT&value,UA_DataType &type){
    // NOLINTNEXTLINE, variant isn't modified, try to avoid copy
    constauto variant=Variant::fromScalar<T>(const_cast<T&>(value),type);
    writeValue(variant);
    return*this;
}
```

But what would be the difference? If the variant can be created "automatically”.

Greatings 
Harald

lukasberbuer, 2023-09-13:
Hi Harald,

OPC UA allows us to contrain the Value attribute of a variable node ([link to specification](https://reference.opcfoundation.org/Core/Part3/v105/docs/5.6.2)):
- `DataType`: Only the specified data type will be accepted
- `ValueRank`: Accept scalar values or arrays or arrays of a specific dimension or anything...
- `ArrayDimensions`: Specify (maximum) supported length of each array dimension

The `Value` attribute of the variable node is always a variant and we must always create a variant within the write request.

According to the OPC UA specification, a variant can only hold builtin types: https://reference.opcfoundation.org/Core/Part6/v105/docs/5.1.2
Complex/composed types like `UA_Range` must therefore be wrapped in an `ExtensionObject`.

open62541 hides this complexity for us:

> The standard mandates that variants contain built-in data types only. If the value is not of a builtin type, it is wrapped into an ExtensionObject. open62541 hides this wrapping transparently in the encoding layer. If the data type is unknown to the receiver, the variant contains the original ExtensionObject in binary or XML encoding.
>
> https://www.open62541.org/doc/1.3/types.html#variant

Now open62541 provides functions to write to a `UA_Variant`, e.g.:

```c
void UA_Variant_setScalar(UA_Variant *v, void *p, const UA_DataType *type);
void UA_Variant_setArray(UA_Variant *v, void *array, size_t arraySize, const UA_DataType *type);
```

`UA_DataType` is a open62541-specific data type:
> All information about a (builtin/structured) data type is stored in a UA_DataType. The array UA_TYPES contains the description of all standard-defined types
>
> https://www.open62541.org/doc/1.3/types.html#generic-type-handling

The `UA_DataType` of a specific type can be retrieved from the `UA_TYPES` with the corresponding index `UA_TYPES_*`, e.g. `UA_TYPES[UA_TYPES_INT32]`.

open62541 simplifies this by providing a mapping of the types and the type indexes with specializations of the `TypeConverter` struct. In the `TypeConverterNative.h` header, conversions for all known types are generated, also for `UA_Range`: https://github.com/open62541pp/open62541pp/blob/master/include/open62541pp/TypeConverterNative.h#L455-L457

Yes, you a right, we don't need the overload for `UA_Range`. I forgot, that open62541pp can also deduce the type index for all native types ;)

lukasberbuer, 2023-09-21:
Are there any open question @lms-hfr?

closed 2023-10-17 by lukasberbuer