open62541pp/open62541pp #109: Cross-compiling for ARM for GCC 7.4.1

piperoc, 2023-09-15:
I was wondering if anyone has attempted cross-compilation recently.

I am building for an ARM target using the Linaro toolchains.

If I'm using the **GCC 11.4.0 arm-linux-gnueabihf**  compilers, it build fine.
When I use **GCC 7.4.1 arm-linux-gnueabihf** (because that's what my QA has tested, so they would prefer me to use it), the compilation fails saying:

```
In file included from /home/cos/git/EdgeXConnect2/deps/open62541pp/include/open62541pp/types/NodeId.h:13:0,
[build]                  from /home/cos/git/EdgeXConnect2/deps/open62541pp/include/open62541pp/Session.h:3,
[build]                  from /home/cos/git/EdgeXConnect2/deps/open62541pp/include/open62541pp/AccessControl.h:6,
[build]                  from /home/cos/git/EdgeXConnect2/deps/open62541pp/src/AccessControl.cpp:1:
[build] /home/cos/git/EdgeXConnect2/deps/open62541pp/include/open62541pp/types/Builtin.h:5:10: fatal error: filesystem: No such file or directory
[build]  #include <filesystem>
[build]           ^~~~~~~~~~~~
[build] compilation terminated.
```

repeatedly.

I'm not sure where to look into since the code builds with the newer toolchain (or with the x86 compilers). I wonder if there are versions requirements (I thought GCC 7.x supports CPP17...).
Thanks

lukasberbuer, 2023-09-15:
Looking at https://en.cppreference.com/w/cpp/compiler_support/17 it seems like GCC 7 supports all the C++17 core language features. The only missing library feature we are using seems to filesystem.
What error do you get after using experimental/filesystem?

piperoc, 2023-09-15:
I should have been more careful, as I initially commented out the C++ 17 options in CMake ...

So, what I did just now was to install a fresh Ubuntu 18.04 image (which ships with GCC 7.5 by default) and cloned the repo.

At first it complained, then I just changed the offending include and nothing else.

The only file that use `filesystem` are:
`include/open62541pp/types/Builtin.h`
`src/types/Builtin.cpp`

**Once I changed those everything seems to build and run fine.**

Here's the modified versions of the files (which you can easily diff to see the few changes):
<details>

<summary><b>Builtin.h</b></summary>

```cpp
#pragma once

#include <array>
#include <cstdint>
#include <experimental/filesystem>
#include <iosfwd>  // forward declare ostream
#include <string>
#include <string_view>
#include <vector>

#include "open62541pp/ErrorHandling.h"
#include "open62541pp/TypeWrapper.h"
#include "open62541pp/open62541.h"

namespace opcua {

/**
 * UA_StatusCode wrapper class.
 * This type is not derived from TypeWrapper. @ref UA_StatusCode is just an alias for `uint32_t`.
 * StatusCode can be used interchangeably with @ref UA_StatusCode due to implicit conversions
 * (without any overhead) but provides some methods to simplify the handling with status codes.
 * @see statuscodes.h
 * @see https://reference.opcfoundation.org/Core/Part4/v105/docs/7.39
 * @ingroup TypeWrapper
 */
class StatusCode {
public:
    using NativeType = UA_StatusCode;

    /// Create a StatusCode with the default status code `UA_STATUSCODE_GOOD`.
    constexpr StatusCode() noexcept = default;

    constexpr StatusCode(UA_StatusCode code) noexcept  // NOLINT, implicit wanted
        : code_(code) {}

    /// Implicit conversion to UA_StatusCode.
    constexpr operator UA_StatusCode() const noexcept {  // NOLINT, implicit wanted
        return code_;
    }

    /// Explicitly get underlying UA_StatusCode.
    constexpr UA_StatusCode get() const noexcept {
        return code_;
    }

    /// Get human-readable name of the StatusCode.
    /// This feature might be disabled to create a smaller binary with the
    /// `UA_ENABLE_STATUSCODE_DESCRIPTIONS` build-flag. Then the function returns an empty string
    /// for every StatusCode.
    std::string_view name() const noexcept {
        return {UA_StatusCode_name(code_)};
    }
[... 549 lines cut ...]

std::string NumericRange::toString() const {
    std::ostringstream ss;
    for (size_t i = 0; i < dimensions_.size(); ++i) {
        const auto& dimension = dimensions_.at(i);
        ss << dimension.min;
        if (dimension.min != dimension.max) {
            ss << ":" << dimension.max;
        }
        if (i < dimensions_.size() - 1) {
            ss << ",";
        }
    }
    return ss.str();
}

}  // namespace opcua

```

</details>

The examples compile with the exception of `examples/typeconversion.cpp` which might be a similar "experimental" issue. I did not have time to look into this.

Thanks again for helping.

lukasberbuer, 2023-09-17:
@piperoc, this is something which we can merge into master. I made a draft PR #110. Can you please review and try if this works with your toolchain?

closed 2023-09-18 by lukasberbuer