open62541pp/open62541pp #112: How to set up custom data type

ipa-vsp, 2023-09-20:
Hello,
I am trying to set up a custom data type to communicate with the OPC-UA Server running in the Omron NJ Controller. Here is the screenshot of the UaExpert OPC-UA Client. 

[image]

[image]

Custom data type: 
```C++
#ifndef MC_ABSOLUTEMOVE_HPP
#define MC_ABSOLUTEMOVE_HPP

#include "open62541pp/open62541pp.h"
#include <cstdint>

/**
 * @brief Structure for reading the status of the MC_MoveAbsolute from the server to client
 */
struct MC_MoveAbsoluteOUT
{
    bool busy;
    bool done;
    bool active;
    bool error;
    int errorID;
    bool commandAborted;
};

/**
 * @brief Structure for writing the status of the MC_MoveAbsolute to the server from client
 */
struct MC_MoveAbsoluteIN
{
    bool execute;
    double position;
    double velocity;
    double acceleration;
    double deceleration;
    double jerk;
};

const opcua::DataType &getMcMoveAbsoluteOutDataType(const char *type_name, opcua::NodeId typeId,
                                                    opcua::NodeId binaryTypeID)
{
    static const opcua::DataType dt =
        opcua::DataTypeBuilder<MC_MoveAbsoluteOUT>::createStructure(type_name, typeId, binaryTypeID)
            .addField<&MC_MoveAbsoluteOUT::done>("Done")
            .addField<&MC_MoveAbsoluteOUT::busy>("Busy")
            .addField<&MC_MoveAbsoluteOUT::active>("Active")
            .addField<&MC_MoveAbsoluteOUT::error>("Error")
            .addField<&MC_MoveAbsoluteOUT::errorID>("Error_ID")
            .addField<&MC_MoveAbsoluteOUT::commandAborted>("CommandAborted")
            .build();
    return dt;
}

const opcua::DataType &getMcMoveAbsoluteInDataType(const char *type_name, opcua::NodeId typeId,
                                                   opcua::NodeId binaryTypeID)
{
    static const opcua::DataType dt =
        opcua::DataTypeBuilder<MC_MoveAbsoluteIN>::createStructure(type_name, typeId, binaryTypeID)
            .addField<&MC_MoveAbsoluteIN::execute>("Execute")
            .addField<&MC_MoveAbsoluteIN::position>("Position")
            .addField<&MC_MoveAbsoluteIN::velocity>("Velocity")
            .addField<&MC_MoveAbsoluteIN::acceleration>("Acceleration")
            .addField<&MC_MoveAbsoluteIN::deceleration>("Decceleration")
            .addField<&MC_MoveAbsoluteIN::jerk>("Jerk")
            .build();
    return dt;
}

#endif // MC_ABSOLUTEMOVE_HPP
```
Main.cpp:

```C++
#include <iostream>

#include "mc_absolutemove.hpp"
#include "open62541pp/open62541pp.h"
#include "rclcpp/rclcpp.hpp"

int main(int argc, char *argv[])
{
    rclcpp::init(argc, argv);

    rclcpp::NodeOptions options;
    options.automatically_declare_parameters_from_overrides(true);
    auto node_ = rclcpp::Node::make_shared("opcua_real_hardware_node", options);
    auto exec = std::make_shared<rclcpp::executors::SingleThreadedExecutor>();

    exec->add_node(node_);

    opcua::Client client;

    const auto &mc_move_absolute_write_data_type = getMcMoveAbsoluteInDataType("MC_MOVE_ABS_IN", {4, 5001}, {4, 1});
    const auto &mc_move_absolute_read_data_type = getMcMoveAbsoluteOutDataType("MC_MOVE_ABS_OUT", {4, 5004}, {4, 2});

    // Provide custom data type definitions to client
    client.setCustomDataTypes({mc_move_absolute_write_data_type});
    client.setCustomDataTypes({mc_move_absolute_read_data_type});

    client.connect("opc.tcp://192.168.250.1:4840");
    RCLCPP_INFO(node_->get_logger(), "Connected to OPC UA server");
    RCLCPP_INFO(node_->get_logger(), "Getting node");
    auto mcMoveAbsoluteWriteNode = client.getNode({4, "Push_Cylinder_ABS_IN"});
    auto mcMoveAbsoluteReadNode = client.getNode({4, "Push_Cylinder_ABS_OUT"});
    
    RCLCPP_INFO(node_->get_logger(), "Got node");
    // Initialize the struct
    MC_MoveAbsoluteIN moveAbsoluteData;
    moveAbsoluteData.execute = true;
    moveAbsoluteData.position = 99.0;
    moveAbsoluteData.velocity = 5.0;
    moveAbsoluteData.acceleration = 2.0;
    moveAbsoluteData.deceleration = 2.0;
    moveAbsoluteData.jerk = 1.0;

    // // Write custom variables
    RCLCPP_INFO(node_->get_logger(), "Writing to node");
    auto mcMoveAbsoluteWriteVar = opcua::Variant::fromScalar(moveAbsoluteData, mc_move_absolute_write_data_type);
    RCLCPP_INFO(node_->get_logger(), "Converted to variant");
    mcMoveAbsoluteWriteNode.writeValue(mcMoveAbsoluteWriteVar);
    RCLCPP_INFO(node_->get_logger(), "Wrote to node");

    std::this_thread::sleep_for(std::chrono::seconds(1));

    auto mcMoveAbsoluteReadVar = mcMoveAbsoluteReadNode.readValue();

    if (mcMoveAbsoluteReadVar.isType(mc_move_absolute_read_data_type))
    {
        RCLCPP_INFO(node_->get_logger(), "Reading from node");
    }
    else{
        RCLCPP_INFO(node_->get_logger(), "Not reading from node");
    }
    
    auto *mcMoveAbsoluteRead = static_cast<MC_MoveAbsoluteOUT *>(mcMoveAbsoluteReadVar.data());
    std::cout << "busy: " << mcMoveAbsoluteRead->busy << std::endl;

    client.disconnect();
    rclcpp::shutdown();
    return 0;
}
```

Unfortunately, I am getting the following error: 
```bash
[2023-09-20 10:19:52.224 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[INFO] [1695197992.224539438] [opcua_real_hardware_node]: Connected to OPC UA server
[INFO] [1695197992.224643701] [opcua_real_hardware_node]: Getting node
[INFO] [1695197992.224666924] [opcua_real_hardware_node]: Got node
[INFO] [1695197992.224673379] [opcua_real_hardware_node]: Writing to node
[INFO] [1695197992.224680134] [opcua_real_hardware_node]: Converted to variant
terminate called after throwing an instance of 'opcua::BadStatus'
  what():  BadInvalidArgument
[ros2run]: Aborted
```

However, if I comment `mcMoveAbsoluteWriteNode.writeValue(mcMoveAbsoluteWriteVar);` line of code, then I am able to read the value from the client. I am trying to figure out how I am not converting the data type correctly. Can you please point out my mistake or help to fix this issue? Thank you.

lukasberbuer, 2023-09-21:
Hi @ipa-vsp, thank you for the detailed description. This helps a lot!

I strongly guess, that the problem is in following lines:
```cpp
    // Provide custom data type definitions to client
    client.setCustomDataTypes({mc_move_absolute_write_data_type});
    client.setCustomDataTypes({mc_move_absolute_read_data_type});
```

The `Client::setCustomDataTypes` method accepts an array. The internal data type definitions are overwritten whenever you call the method again. So you should provide an array with all custom data types needed in one call:

```cpp
    // Provide custom data type definitions to client
    client.setCustomDataTypes({mc_move_absolute_write_data_type, mc_move_absolute_read_data_type});
```

Can you please verify if this works? And I should update the documentation to make the behaviour of `Client::setCustomDataTypes` clear 😄

ipa-vsp, 2023-09-21:
Hi @lukasberbuer,
Thank you for your answer. 

Yes, I also did the testing with `client.setCustomDataTypes({mc_move_absolute_write_data_type, mc_move_absolute_read_data_type});` similar result. 

During the debugging, I discovered that the variant type is **ExtentionObject** instead of defined **MC_MOVE_ABS_IN** or **MC_MOVE_ABS_OUT**. 

Current code:
```C++
#include <iostream>

#include "open62541pp/open62541pp.h"
#include "rclcpp/rclcpp.hpp"

struct MC_MoveAbsoluteIN
{
    bool execute;
    float position;
    float velocity;
    float acceleration;
    float deceleration;
    float jerk;
}__attribute__((packed));

namespace opcua
{
    template <>
    struct TypeConverter<MC_MoveAbsoluteIN>
    {
        using ValueType = MC_MoveAbsoluteIN;
        using NativeType = UA_ExtensionObject;
        using ValidTypes = TypeIndexList<UA_TYPES_EXTENSIONOBJECT>;
        
        static void fromNative(const NativeType &src, ValueType &dst)
        {

            // Check the size first
            if(src.encoding != UA_EXTENSIONOBJECT_ENCODED_BYTESTRING) {
                throw std::runtime_error("Invalid encoding type for UA_ExtensionObject");
            }
            std::cout << "src.content.encoded.body.length: " << src.content.encoded.body.length << std::endl;
            std::cout << "sizeof(MC_MoveAbsoluteIN): " << sizeof(MC_MoveAbsoluteIN) << std::endl;
            if(src.content.encoded.body.length != sizeof(MC_MoveAbsoluteIN)) {
                throw std::runtime_error("Size mismatch between MC_MoveAbsoluteIN and UA_ExtensionObject body length");
            }

            memcpy(&dst, src.content.encoded.body.data, sizeof(MC_MoveAbsoluteIN));
        }

        static void toNative(const ValueType &src, NativeType &dst)
        {
            // Handle buffer allocation
            if(UA_ByteString_allocBuffer(&dst.content.encoded.body, sizeof(MC_MoveAbsoluteIN)) != UA_STATUSCODE_GOOD) {
                throw std::runtime_error("Failed to allocate memory for UA_ExtensionObject");
            }

            memcpy(dst.content.encoded.body.data, &src, sizeof(MC_MoveAbsoluteIN));

            // Set the encoding type
            dst.encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
        }
    };
}

int main(int argc, char *argv[])
{
    rclcpp::init(argc, argv);

    rclcpp::NodeOptions options;
    options.automatically_declare_parameters_from_overrides(true);
    auto node_ = rclcpp::Node::make_shared("opcua_real_hardware_client_node", options);
    auto exec = std::make_shared<rclcpp::executors::SingleThreadedExecutor>();

    exec->add_node(node_);
    opcua::Client client;

    // Connect to server
    client.connect("opc.tcp://192.168.250.1:4840");

    auto pullInNode = client.getNode({4, "Pull_Cylinder_ABS_IN"});

    auto varient = pullInNode.readValue();

    auto pull_abs_in = varient.getScalarCopy<MC_MoveAbsoluteIN>();
    std::cout << "Pull_Cylinder_ABS_IN: " << std::endl;
    std::cout << "Execute: " << pull_abs_in.execute << std::endl;
    std::cout << "Position: " << pull_abs_in.position << std::endl;
    std::cout << "Velocity: " << pull_abs_in.velocity << std::endl;
    std::cout << "Acceleration: " << pull_abs_in.acceleration << std::endl;
    std::cout << "Deceleration: " << pull_abs_in.deceleration << std::endl;
    std::cout << "Jerk: " << pull_abs_in.jerk << std::endl;

    MC_MoveAbsoluteIN pull_abs_in_new;
    pull_abs_in_new.execute = true;
    pull_abs_in_new.position = 101.0;
    pull_abs_in_new.velocity = 102.0;
    pull_abs_in_new.acceleration = 103.0;
    pull_abs_in_new.deceleration = 104.1;
    pull_abs_in_new.jerk = 105.5;

    opcua::DataValue pullInNode_write_value;
    opcua::Variant var;
    var.setScalarCopy<MC_MoveAbsoluteIN>(pull_abs_in_new);
    pullInNode.writeValue(var);

    client.disconnect();
    rclcpp::shutdown();
    return 0;
}
```

Result:

```bash
src.content.encoded.body.length: 21
sizeof(MC_MoveAbsoluteIN): 21
Pull_Cylinder_ABS_IN: 
Execute: 1
Position: 10.1
Velocity: 0
Acceleration: 0
Deceleration: 155.222
Jerk: 0
terminate called after throwing an instance of 'opcua::BadStatus'
  what():  BadInvalidArgument
[ros2run]: Aborted
```
Screenshot of debug,
[image]

I am still stuck on how to write to the server. But reading from the server works well with `TypeConverter`.

Thank you.

lukasberbuer, 2023-09-25:
Hi @ipa-vsp, I created a minimal example on branch `issue#112-minimal-example` to reproduce and understand the problem: https://github.com/open62541pp/open62541pp/tree/issue%23112-minimal-example/examples/custom_datatypes

If both server and client share the same data type definitions, we have no issues and open62541 can automatically decode the custom types. The only real unknown of the data type definitions is the `binaryEncodingId`, which I guess you randomly set. You can't see it in the client like UAExpert. If the `binaryEncodingId` is wrong, open62541 won't be able to decode the structure. Sadly, it's quite hard to get the binary encoding id if you don't have access to the server information model:
https://github.com/open62541/open62541/issues/4222

You can reproduce this issue if you build the example above and change the `binaryEncodingId` and run the server and client with different binary encoding ids.

lukasberbuer, 2023-10-02:
Hi @ipa-vsp, could you find out the `binaryEncodingId` and verify if it works then?

ipa-vsp, 2023-10-06:
Hi @lukasberbuer, 

Thank you so much for your answer! I was able to resolve my issue writing `TypeConverter`

```C++
struct MC_MoveAbsoluteIN
{
    bool execute; // Offset: 2
    float position; // Offset: 6
    float velocity; // Offset: 10
    float acceleration; // Offset: 14
    float deceleration; // Offset: 18
    float jerk; // Offset: 22 --> total size: 22
}__attribute__((packed));

struct NodeIdInfo {
    uint16_t namespaceIndex;
    uint32_t identifier;  // Assuming you are using UA_NODEIDTYPE_STRING for simplicity.
};

NodeIdInfo nodeIdInfoIN;
NodeIdInfo nodeIdInfoOUT;

namespace opcua
{
    template <>
    struct TypeConverter<MC_MoveAbsoluteIN>
    {
        using ValueType = MC_MoveAbsoluteIN;
        using NativeType = UA_ExtensionObject;
        using ValidTypes = TypeIndexList<UA_TYPES_EXTENSIONOBJECT>;
        
        static void fromNative(const NativeType &src, ValueType &dst)
        {
            // Check the encoding type
            if(src.encoding != UA_EXTENSIONOBJECT_ENCODED_BYTESTRING) {
                throw std::runtime_error("Invalid encoding type for UA_ExtensionObject");
            }
            
            // Print Type ID
            // std::cout << "src.content.encoded.typeId: " << nodeIdToString(src.content.encoded.typeId) << std::endl;
            nodeIdInfoIN.namespaceIndex = src.content.encoded.typeId.namespaceIndex;
            nodeIdInfoIN.identifier = src.content.encoded.typeId.identifier.numeric;
            
            // Check size
            std::cout << "src.content.encoded.body.length: " << src.content.encoded.body.length << std::endl;
            std::cout << "sizeof(MC_MoveAbsoluteIN): " << sizeof(MC_MoveAbsoluteIN) << std::endl;
            if(src.content.encoded.body.length != sizeof(MC_MoveAbsoluteIN)) {
                throw std::runtime_error("Size mismatch between MC_MoveAbsoluteIN and UA_ExtensionObject body length");
            }

            memcpy(&dst, src.content.encoded.body.data, sizeof(MC_MoveAbsoluteIN));
        }

        static void toNative(const ValueType &src, NativeType &dst)
        {
            std::cout << "dst.content.encoded.body.length: " << dst.content.encoded.body.length << std::endl;
            std::cout << "sizeof(MC_MoveAbsoluteIN): " << sizeof(MC_MoveAbsoluteIN) << std::endl;
            
            // Allocate memory buffer
            if(UA_ByteString_allocBuffer(&dst.content.encoded.body, sizeof(MC_MoveAbsoluteIN)) != UA_STATUSCODE_GOOD) {
                throw std::runtime_error("Failed to allocate memory for UA_ExtensionObject");
            }
            
            // Initialize memory
            memset(dst.content.encoded.body.data, 0, sizeof(MC_MoveAbsoluteIN));
            
            // Copy data
            memcpy(dst.content.encoded.body.data, &src, sizeof(MC_MoveAbsoluteIN));
            
            // Set the encoding type
            dst.encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
            
            // Set Type ID (if necessary, example given below)
            dst.content.encoded.typeId = UA_NODEID_NUMERIC(nodeIdInfoIN.namespaceIndex, nodeIdInfoIN.identifier);
        }
    };
} 
```
In my last code, TypeId was missing. And now I am able to read the OPC-UA server data. Is there a way to create a server which encodes the structure data? Right now I am to read and write real hardware but it will be nice to have if can create a mock server.

lukasberbuer, 2023-10-11:
Hi @ipa-vsp, now you are using a `TypeConverter` specialization to encode/decode the types from an `ExtensionObject`. open62541 will encode/decode your data type automatically if the type is known by the client or server. BUT you need to have the correct `binaryEncodingId` from the server as described here: https://github.com/open62541/open62541/issues/4222
The `ExtensionObject` is just the fall-back if the client/server is not able to decode the received data type, usually because the server does not know the type.
So I would strongly recommend that you try to find out the `binaryEncodingId`s of the servers custom data types. This should be included in the information model. Otherwise you have to spy those e.g. with WireShark.

lukasberbuer, 2023-10-11:
Btw, I already created a mock server of your example here: https://github.com/open62541pp/open62541pp/blob/issue%23112-minimal-example/examples/custom_datatypes/server_custom_datatypes.cpp

lukasberbuer, 2023-12-11:
Hi @ipa-vsp, with OPC UA specification v1.04, you can read the `binaryEncodingId` with your client. The `DataType` node has a `DataTypeDefinition` attribute:

[image]

pemartin2, 2024-07-25:
@lukasberbuer: I am trying to read the datatype and binaryEncodingId automatically with your library. Is this possible?
I got to the datatype easily.
Is it possible to read out that `binaryEncodingId` with open62541pp? If yes could you guide me on how to do so? 

What I got so far using your point custom datatype example on the client side:

before main
```
struct StructureDefinition {
    ::opcua::NodeId encodingId;
};

const opcua::DataType& getStructureDefinitionDatatype() {
    static const opcua::DataType dt =
        opcua::DataTypeBuilder<StructureDefinition>::createStructure(
            "StructureDefinition", {0, 99}, {1, 1}
        )
            .addField<&StructureDefinition::encodingId>("encodingId")
            .build();
    return dt;
}
```
then in main:

 ```
const auto& dataTypeStructureDefiniton = getStructureDefinitionDatatype();

client.setCustomDataTypes({
    dataTypePoint,
    dataTypeMeasurements,
    dataTypeOpt,
    dataTypeUni,
    dataTypeColor,
    dataTypeStructureDefiniton, //if I put this in first place something seems to break completely (->memory alignment(?))
});

[...]

    auto datatype = client.getNode({1, "Point"}).readDataType();
    auto attributeDataValue = ::opcua::services::readAttribute(
        client, datatype, ::opcua::AttributeId::DataTypeDefinition
    );
    assert(attributeDataValue.hasValue());
    auto dataValue = attributeDataValue.value();
    opcua::Variant structureDefAsVaraint = dataValue.getValue();
    if (structureDefAsVaraint.isType(dataTypeStructureDefiniton)) {
        const auto* p = static_cast<opcua::NodeId*>(variant.data());
       // -->I get here but p is a nullptr    
    }
```
I am able to get the `DataTypeDefinition` attribute of the custom datatype. and get it as a variant. However, I don't really have a clue how to extract the encoding as a value out of it... I was trying to create a custom datatype for it but probably am failing to do so...

Any help appreciated.

Hope it's fine that I capture this thread here or would you like me to start a new one?

lukasberbuer, 2024-07-29:
Hi @pemartin2, reading the `DataTypeDefinition` attribute with open62541 is possible but wasn't implemented yet.
I implemented it in this PR https://github.com/open62541pp/open62541pp/pull/145
Please review 😊

The return type is a `Variant` and can be of type `StructureDefinition` (with a `defaultEncodingId` member).
Following test showcases how to read the `defaultEncodingId` (= `binaryEncodingId`): https://github.com/open62541pp/open62541pp/blob/edefbb662d27a3cb36e25938cc77cf0f94b77367/tests/Services.cpp#L470-L481