open62541pp/open62541pp #113: Custom DataTypes on Server

MioStarkid, 2023-10-09:
I try to add multiple custom datatypes to a ua server as follow:

```c
 /**
   * ChannelCfg
   * ^^^^^^^^^^^^^^^^^^^^^^^
   */
  struct UA_CHANNELCFGDataType {
    UA_Byte messart;
    UA_Byte messbereich;
  };

  /**
   * ComponentConfig
   * ^^^^^^^^^^^^^^^^^^^^^^^
   */
  struct UA_COMPONENTCONFIGDataType {
    UA_SByte cmpId;
    UA_Byte cmpType;
    UA_Int16 aliasIndex;
    UA_Int32 dataDb;
    UA_Int32 configDb;
    UA_Byte fbActive;
    UA_UInt32 fbStateDbAddr;
  };
     
  /**
   * ConfigTexts
   * ^^^^^^^^^^^^^^^^^^^^^^^
   */
  struct UA_CONFIGTEXTSDataType {
    UA_SByte cmpId;
    UA_Byte idType;
    UA_Int16 identifier;
    UA_SByte subIndex;
    UA_SByte configSet;
    UA_String text;
  }; 

  /**
   * PersSignal
   * ^^^^^^^^^^^^^^^^^^^^^^^
   */
  struct UA_PERSSIGNALDataType {
    UA_Int32 identifier;
    UA_Double value;
  };

  /**
   * PersData
   * ^^^^^^^^^^^^^^^^^^^^^^^
   */
  struct UA_PERSDATADataType {
    size_t persSignalSize;
    UA_PERSSIGNALDataType* persSignals;// [60];
    UA_Int16 groupId1;
    size_t aliases1Size;
    UA_Int16* aliases1;//[20];
    UA_Int16 groupId2;
    size_t aliases2Size;
    UA_Int32* aliases2;// [20];
  };
  
  const opcua::DataType& getChannelCfgType()
  {
    // ChannelCfg
    static const opcua::DataType channelCfg =
      opcua::DataTypeBuilder<UA_CHANNELCFGDataType>::createStructure("ChannelCfg", { 3, "DT_\"ChannelCfg\"" }, { 3, "TE_\"ChannelCfg\"" })
      .addField<&UA_CHANNELCFGDataType::messart>("Messart")
      .addField<&UA_CHANNELCFGDataType::messbereich>("Messbereich")
      .build();
    return channelCfg;
  }

  const opcua::DataType& getComponentConfigType()
  {
    // ComponentConfig
    static const opcua::DataType componentConfig =
      opcua::DataTypeBuilder<UA_COMPONENTCONFIGDataType>::createStructure("ComponentConfig", { 3, "DT_\"ComponentConfig\"" }, { 3, "TE_\"ComponentConfig\"" })
      .addField<&UA_COMPONENTCONFIGDataType::cmpId>("CmpId")
      .addField<&UA_COMPONENTCONFIGDataType::cmpType>("CmpType")
      .addField<&UA_COMPONENTCONFIGDataType::aliasIndex>("AliasIndex")
      .addField<&UA_COMPONENTCONFIGDataType::dataDb>("DataDb")
      .addField<&UA_COMPONENTCONFIGDataType::configDb>("ConfigDb")
      .addField<&UA_COMPONENTCONFIGDataType::fbActive>("FbActive")
      .addField<&UA_COMPONENTCONFIGDataType::fbStateDbAddr>("FbStateDbAddr")
      .build();
    return componentConfig;
  }
   
  const opcua::DataType& getPersSignalType()
  {
    // PersSignal
    static const opcua::DataType persSignal =
      opcua::DataTypeBuilder<UA_PERSSIGNALDataType>::createStructure("PersSignal", { 3, "DT_\"PersSignal\"" }, { 3, "TE_\"PersSignal\"" })
      .addField<&UA_PERSSIGNALDataType::identifier>("Identifier")
      .addField<&UA_PERSSIGNALDataType::value>("Value")
      .build();
    return persSignal;
  }

  const opcua::DataType& getPersDataType()
  {
    // PersData
    static const opcua::DataType persData =
      opcua::DataTypeBuilder<UA_PERSDATADataType>::createStructure("PersData", { 3, "DT_\"PersData\"" }, { 3, "TE_\"PersData\"" })
      .addField<&UA_PERSDATADataType::persSignalSize, &UA_PERSDATADataType::persSignals>("PersSignals")
      .addField<&UA_PERSDATADataType::groupId1>("GroupId1")
      .addField<&UA_PERSDATADataType::aliases1Size, &UA_PERSDATADataType::aliases1>("Aliases1")
      .addField<&UA_PERSDATADataType::groupId2>("GroupId2")
      .addField<&UA_PERSDATADataType::aliases2Size, &UA_PERSDATADataType::aliases2>("Aliases2")
      .build();
    return persData;
  }  
```

```c
int main() {
    opcua::Server server(4840 /* port */);

    server.setApplicationName("open62541pp server example");
    server.setLogger([](auto level, auto category, auto msg) {
        std::cout << "[" << opcua::getLogLevelName(level) << "] "
                  << "[" << opcua::getLogCategoryName(category) << "] " << msg << std::endl;
    });

    const auto& dataTypes1 = getChannelCfgType();
    const auto& dataTypes2 = getComponentConfigType();
    const auto& dataTypes3 = getPersSignalType();
    const auto& dataTypes4 = getPersDataType();

    // Add a variable node to the Objects node
    auto parentNode = server.getObjectsNode();
    auto myIntegerNode = parentNode.addVariable(
        {1, "the.answer"},
        "the answer",
        opcua::VariableAttributes{}
            .setDataType<int>()
            .setDisplayName({"en-US", "the answer"})
            .setDescription({"en-US", "the answer"})
    );

    // Write a value (attribute) to the node
    myIntegerNode.writeValueScalar(42);

    // Read the value (attribute) from the node
    std::cout << "The answer is: " << myIntegerNode.readValueScalar<int>() << std::endl;

    server.run();
}
```

When I call getPersSignalType() or getPersDataType(), I get the exception "Debug Error" abort() as been called.
The exception happens in the DataTypeBuilder.h
```c
template <typename T, typename Tag>
template <auto T::*field>
DataTypeBuilder<T, Tag>& DataTypeBuilder<T, Tag>::addField(
    const char* fieldName, const UA_DataType& fieldType
) {
    using TMember = detail::MemberTypeT<decltype(field)>;
    static_assert(
        std::is_same_v<Tag, detail::TagDataTypeStruct>,
        "Built type must be a struct or class to add members"
    );
    assert(sizeof(detail::UnqualifiedT<TMember>) == fieldType.memSize);
    if (std::is_pointer_v<TMember>) {
        dataType_.setTypeKind(UA_DATATYPEKIND_OPTSTRUCT);
    }
    if (std::is_pointer_v<TMember> || !fieldType.pointerFree) {
        dataType_.setPointerFree(false);
    }
    fields_.push_back({
        sizeof(TMember),
        detail::offsetOfMember(field),
        detail::createDataTypeMember(
            fieldName,
            fieldType,
            {},  // calculate padding between members later
            false,
            std::is_pointer_v<TMember>
        ),
    });
    return *this;
}
```
At the row  assert(sizeof(detail::UnqualifiedT<TMember>) == fieldType.memSize);

I work with the version v0.9.0.
Do I miss anything here?

lukasberbuer, 2023-10-10:
Hi @MioStarkid, thanks for reporting! Which compiler and operating system are you using?

I will provide an example for your types. We might need to add overloads to allow nested data types like `UA_PERSDATADataType` containing a custom member array of type `UA_PERSSIGNALDataType`

lukasberbuer, 2023-10-10:
I could not reproduce the assertion error, but compilation failed. Following modification was necessary to make it compile and run on my machine: You need to specify the data type of the custom type member `UA_PERSDATADataType::persSignals` manually.

```cpp

const opcua::DataType& getPersDataType() {
    static const opcua::DataType persData =
        opcua::DataTypeBuilder<UA_PERSDATADataType>::createStructure("PersData", {1, "DT_\"PersData\""}, {1, "TE_\"PersData\""})
            .addField<&UA_PERSDATADataType::persSignalSize, &UA_PERSDATADataType::persSignals>(
                "PersSignals",
                getPersSignalType()  // nested custom data type needs explicit data type parameter
            )
            .addField<&UA_PERSDATADataType::groupId1>("GroupId1")
            .addField<&UA_PERSDATADataType::aliases1Size, &UA_PERSDATADataType::aliases1>("Aliases1")
            .addField<&UA_PERSDATADataType::groupId2>("GroupId2")
            .addField<&UA_PERSDATADataType::aliases2Size, &UA_PERSDATADataType::aliases2>("Aliases2")
            .build();
    return persData;
}
```

MioStarkid, 2023-10-11:
Hi @lukasberbuer, thanks for your answer and your example.
I guess the libraries I built had somehow a problem. However I never got any compilation error.
I, now compiled the library on a diffrent pc and now the server "custom datatype example" works just fine.

1. PC was with Windows 10 and Visual Studio 15 2017 compiled for x86 and x64. With both versions I had the problem, that the function
```c
template <auto memberPtr>
constexpr const UA_DataType& guessMemberDataType() {
    return guessDataType<detail::MemberTypeT<decltype(memberPtr)>>();
}
```
intepreted a UA_TYPES[UA_TYPES_STRING] as a UA_TYPES[UA_TYPES_FLOAT] somehow. And then failed at
```c
template <typename T, typename Tag>
template <auto T::*field>
DataTypeBuilder<T, Tag>& DataTypeBuilder<T, Tag>::addField(
    const char* fieldName, const UA_DataType& fieldType
) {
    using TMember = detail::MemberTypeT<decltype(field)>;
    static_assert(
        std::is_same_v<Tag, detail::TagDataTypeStruct>,
        "Built type must be a struct or class to add members"
    );
    assert(sizeof(detail::UnqualifiedT<TMember>) == fieldType.memSize);
    if (std::is_pointer_v<TMember>) {
        dataType_.setTypeKind(UA_DATATYPEKIND_OPTSTRUCT);
    }
    if (std::is_pointer_v<TMember> || !fieldType.pointerFree) {
        dataType_.setPointerFree(false);
    }
    fields_.push_back({
        sizeof(TMember),
        detail::offsetOfMember(field),
        detail::createDataTypeMember(
            fieldName,
            fieldType,
            {},  // calculate padding between members later
            false,
            std::is_pointer_v<TMember>
        ),
    });
    return *this;
}
```

2. PC was with Windows 11 and Visual Studio 2022. Here the example project works just fine.

lukasberbuer, 2023-10-11:
Ok this is strange. Can you reproduce the error on the first machine if you start with a clean build directory?
VisualStudio 15 is not part of the CI pipeline, but version 16 and 17 are.

lukasberbuer, 2023-10-17:
@MioStarkid, do you have any updates?

MioStarkid, 2023-11-07:
On other odd thing I discovered with custom datatypes is, when you add a custom variable to the server and read it back.
The variant is not a opcua::ExtensionObject. However ua expert shows the variable as ExtensionObject. After you perform a write operation via client (ua expert and own client application), if you then read the node in the server application it is now a opcua::ExtensiontObject.
```c

    std::vector<float> measurementsValues{19.1F, 20.2F, 19.7F};
    const Measurements measurements{
        opcua::String("Test description"),
        measurementsValues.size(),
        measurementsValues.data(),
    };
    auto var = nodeObjects.addVariable(
        {1, "Measurements"},
        "Measurements",
        opcua::VariableAttributes{}
            .setDataType(dataTypeMeasurements.getTypeId())
            .setValueRank(opcua::ValueRank::Scalar)
            .setValueScalar(measurements, dataTypeMeasurements),
        nodeVariableTypeMeasurement.getNodeId()
    );

    auto variant = var.readValue();
    bool isExtensionObj = variant.isType(opcua::Type::ExtensionObject); // is false. Becomes true after a client writes to it.
```

This bug can be reproducet via custom datatype server example

lukasberbuer, 2023-11-07:
Hi @MioStarkid, I am not sure if I understand this correctly. If I run the `server_custom_datatype` example, the `Measurements` value looks like this in UaExpert:
[image]

Open62541 has some non-standard behavioral regarding custom data types and ExtensionObjects, maybe that is the reason for this confusion:

> The standard mandates that variants contain built-in data types only. If the value is not of a builtin type, it is wrapped into an [ExtensionObject](https://www.open62541.org/doc/master/types.html#extensionobject). open62541 hides this wrapping transparently in the encoding layer. If the data type is unknown to the receiver, the variant contains the original ExtensionObject in binary or XML encoding.

Found here: https://www.open62541.org/doc/master/types.html#variant

MioStarkid, 2023-11-07:
If you subscribe the variable uaexpert shows custom datatype variables as ExtensionObject.
[image]

Follwing is a short test program which shows the odd behaviour.
If you start the server app and call the method "CheckDatatype" the variant value of the node "PointVec" is not a ExtensionObject Type. However if you change the value of the "PointVec" variable and call "CheckDatatype" again, it will now be of type ExtensionObject, and needs to be handled as such otherwise the data gets wrong decoded.

[image]
 

```c
int main() {
    opcua::Server server;

    // Get custom type definitions from common header
    const auto& dataTypePoint = getPointDataType();

    // Provide custom data type definitions to server
    server.setCustomDataTypes({dataTypePoint});

    // Add data type nodes
    auto nodeStructureDataType = server.getNode(opcua::DataTypeId::Structure);
    nodeStructureDataType.addDataType(dataTypePoint.getTypeId(), "Point");

    // Add variable type nodes
    auto nodeBaseDataVariableType = server.getNode(opcua::VariableTypeId::BaseDataVariableType);
    auto nodeVariableTypePoint = nodeBaseDataVariableType.addVariableType(
        {1, 4243},
        "PointType",
        opcua::VariableTypeAttributes{}
            .setDataType(dataTypePoint.getTypeId())
            .setValueRank(opcua::ValueRank::ScalarOrOneDimension)
            .setValueScalar(Point{1, 2, 3}, dataTypePoint)
    );

    // Add variable nodes with some values
    auto nodeObjects = server.getObjectsNode();
    const std::vector<Point> pointVec{{1.0, 2.0, 3.0}, {4.0, 5.0, 6.0}};
    auto test = nodeObjects.addVariable(
        {1, "PointVec"},
        "PointVec",
        opcua::VariableAttributes{}
            .setAccessLevel(UA_ACCESSLEVELMASK_WRITE | UA_ACCESSLEVELMASK_READ)
            .setDataType(dataTypePoint.getTypeId())
            .setArrayDimensions({0})  // single dimension but unknown in size
            .setValueRank(opcua::ValueRank::OneDimension)
            .setValueArray(pointVec, dataTypePoint),
        nodeVariableTypePoint.getNodeId()
    );

    // Add a method to check datatype of variable
    nodeObjects.addMethod(
        {1, 1000},
        "CheckDatatype",
        [&server](opcua::Span<const opcua::Variant> input, opcua::Span<opcua::Variant> output) {
            static size_t methodCallNr = 0;

            std::cout << "------------------- " << ++methodCallNr << ". Method Call--------------------------\n";
            opcua::Variant value = server.getNode({1, "PointVec"}).readValue();
            if (value.isType(opcua::Type::ExtensionObject))
            {
                std::cout << "Value is ExpenstionObject Type\n";
                size_t i = 0;
                std::cout << "Handle it as ExtenstionObject:\n";
                for (auto&& extObj : value.getArray<opcua::ExtensionObject>()) {
                    const auto* p = static_cast<Point*>(extObj.getDecodedData());
                    std::cout << "PointVec[" << i++ << "]:\n";
                    std::cout << "- x = " << p->x << "\n";
                    std::cout << "- y = " << p->y << "\n";
                    std::cout << "- z = " << p->z << "\n";
                }

                std::cout << std::endl << "Handle it as non ExtenstionObject:\n";
                const auto* p = static_cast<Point*>(value.data());
                for (size_t i = 0; i < value.getArrayLength(); i++) {
                    std::cout << "PointVec[" << i << "]:\n";
                    std::cout << "- x = " << p[i].x << "\n";
                    std::cout << "- y = " << p[i].y << "\n";
                    std::cout << "- z = " << p[i].z << "\n";
                }
            } else {
                std::cout << "Value is not ExpenstionObject Type\n";
                std::cout << "Handle it as non ExtenstionObject:\n";                
                const auto* p = static_cast<Point*>(value.data());
                for (size_t i = 0; i < value.getArrayLength(); i++) {
                    std::cout << "PointVec[" << i << "]:\n";
                    std::cout << "- x = " << p[i].x << "\n";
                    std::cout << "- y = " << p[i].y << "\n";
                    std::cout << "- z = " << p[i].z << "\n";
                }
            }
        },
        {},
        {}
    );
    server.run();
}
```

lukasberbuer, 2023-11-07:
This is interesting, thanks for the great example @MioStarkid. I get the same results as you.
But if you also read the `PointVec` node with the slightly modified `client_custom_datatypes`, the results are correct on the client-side:

Output server:
```
------------------- 2. Method Call--------------------------
Value is ExpenstionObject Type
Handle it as ExtenstionObject:
PointVec[0]:
- x = 1
- y = 2
- z = 3
PointVec[1]:
- x = 10
- y = 5
- z = 6

Handle it as non ExtenstionObject:
PointVec[0]:
- x = 4.2039e-45
- y = 0
- z = 1.80218e-40
PointVec[1]:
- x = 3.45953e-41
- y = 5.21709e-40
- z = 3.44832e-41
```

Output client:
```
PointVec[0]:
- x = 1
- y = 2
- z = 3
PointVec[1]:
- x = 10
- y = 5
- z = 6
```

So the problem seems to be, that there is a mismatch on the server side between the returned data type and the actual unwrapping. If yes, this is a issue of open62541. Unwrapping of arrays of structs is currently WIP and planned vor open62541 v1.4: https://github.com/open62541/open62541/pull/4814

closed 2024-09-15 by lukasberbuer