open62541pp/open62541pp #114: mbedtls

johanneskopf, 2023-10-15:
Hi @lukasberbuer ,

Do you have any experience/hints on how to set up and compile an open62541pp project with mbedtls using cmake? Ideally, I also want to have it as a git subrepo of my project so that I can quickly navigate to its source code as well. 

In the future, I will also need to reduce the memory footprint of mbedtls to make it run on an embedded system. I haven't worked with mbedtls before. The best information I could find related to that topic is here: https://mbed-tls.readthedocs.io/en/latest/kb/how-to/reduce-polarssl-memory-and-storage-footprint/

lukasberbuer, 2023-10-17:
Hi @johanneskopf,

if you really need mbedtls as a submodule, you need to change the `FindMbedTLS.cmake` file: https://github.com/open62541pp/open62541pp/blob/master/3rdparty/open62541/tools/cmake/FindMbedTLS.cmake

Currently, open62541 searches for the compiled library and header files. So you have to build and install mbedtls first to make open62541 use it. Please let me know if you need help to rewrite the CMake file.

johanneskopf, 2023-10-20:
Hi @lukasberbuer ,

It would be great to have it as a submodule later on to make it even faster to iterate while adapting the mbedTLS flags.

As a first step though, I tried to compile any open62541pp example with mbedTLS enabled. I was able to clone, build and install mbedTLS just fine on Windows using CLion. Next, I tried to adapt my existing project (using open62541pp as a submodule) accordingly. Thus, I adapted the CMake options to the following:

```
-DUA_ARCHITECTURE:STRING=win32
-DUA_ENABLE_AMALGAMATION:BOOL=OFF
-DUA_ENABLE_ENCRYPTION:STRING=MBEDTLS
-DMBEDTLS_INCLUDE_DIRS="C:\Program Files (x86)\mbed TLS\include"
-DMBEDTLS_LIBRARY="C:\Program Files (x86)\mbed TLS\lib"
-DMBEDX509_LIBRARY="C:\Program Files (x86)\mbed TLS\lib"
-DMBEDCRYPTO_LIBRARY="C:\Program Files (x86)\mbed TLS\lib"
```

My run configurations have been working fine so far. But since I changed the CMake options, I now get the following error when building one of my configurations:
`LINK : fatal error LNK1104: cannot open file 'C:\Program Files (x86)\mbed TLS\lib.obj'`

So far I found this here which could be relevant to solve the issue:
[https://github.com/Mbed-TLS/mbedtls/issues/1130](url)
But I used the following flags (set per default like this) so I should not run into this issue according to the replies:
`-DUSE_SHARED_MBEDTLS_LIBRARY=Off -DUSE_STATIC_MBEDTLS_LIBRARY=On`

Any idea how to fix this?

johanneskopf, 2023-10-30:
Hi @lukasberbuer ,

Thanks, your reply helped a lot! That was exactly the problem.

Per default mbedTLS is set to `-DUSE_STATIC_MBEDTLS_LIBRARY=On`, thus generating static libraries, i.e. .lib files on Windows. 
During the CMake installation it wrote the following lines to the output:
```
-- Installing: C:/Program Files (x86)/Mbed TLS/lib/mbedtls.lib
-- Installing: C:/Program Files (x86)/Mbed TLS/lib/mbedx509.lib
-- Installing: C:/Program Files (x86)/Mbed TLS/lib/mbedcrypto.lib
```
Thus I adapted the paths just as you suggested (but using .lib ending) and now it compiles fine.
For reference, my library paths are now set like this:
```
-DMBEDTLS_LIBRARY="C:\Program Files (x86)\mbed TLS\lib\mbedtls.lib"
-DMBEDX509_LIBRARY="C:\Program Files (x86)\mbed TLS\lib\mbedx509.lib"
-DMBEDCRYPTO_LIBRARY="C:\Program Files (x86)\mbed TLS\lib\mbedcrypto.lib"
```

For completeness sake, here also the output of the `tree 'C:\Program Files (x86)\Mbed TLS\' /F` command:
```
C:\PROGRAM FILES (X86)\MBED TLS
├───bin
│       aead_demo.exe
│       benchmark.exe
│       cert_app.exe
│       cert_req.exe
│       cert_write.exe
│       cipher_aead_demo.exe
│       crl_app.exe
│       crypto_examples.exe
│       crypt_and_hash.exe
│       dh_client.exe
│       dh_genprime.exe
│       dh_server.exe
│       dtls_client.exe
│       dtls_server.exe
│       ecdh_curve25519.exe
│       ecdsa.exe
│       generic_sum.exe
│       gen_entropy.exe
│       gen_key.exe
│       gen_random_ctr_drbg.exe
│       hello.exe
│       hmac_demo.exe
│       key_app.exe
│       key_app_writer.exe
│       key_ladder_demo.exe
│       key_ladder_demo.sh
│       load_roots.exe
│       mbedcrypto.dll
│       mbedtls.dll
│       mbedx509.dll
│       md_hmac_demo.exe
│       mini_client.exe
│       mpi_demo.exe
│       pem2der.exe
│       pk_decrypt.exe
│       pk_encrypt.exe
│       pk_sign.exe
│       pk_verify.exe
│       psa_constant_names.exe
│       query_compile_time_config.exe
│       query_included_headers.exe
│       req_app.exe
│       rsa_decrypt.exe
│       rsa_encrypt.exe
│       rsa_genkey.exe
│       rsa_sign.exe
│       rsa_sign_pss.exe
[... 132 lines cut ...]
│           crypto_driver_contexts_key_derivation.h
│           crypto_driver_contexts_primitives.h
│           crypto_extra.h
│           crypto_legacy.h
│           crypto_platform.h
│           crypto_se_driver.h
│           crypto_sizes.h
│           crypto_struct.h
│           crypto_types.h
│           crypto_values.h
│
└───lib
    │   everest.lib
    │   mbedcrypto.lib
    │   mbedtls.lib
    │   mbedx509.lib
    │   p256m.lib
    │
    └───cmake
        └───MbedTLS
                MbedTLSConfig.cmake
                MbedTLSConfigVersion.cmake
                MbedTLSTargets-debug.cmake
                MbedTLSTargets.cmake
```

lukasberbuer, 2023-10-30:
Glad this worked! 😄

lukasberbuer, 2023-10-30:
Please reopen this issue if there are any questions/problems with mbedtls integration

closed 2023-10-30 by lukasberbuer