open62541pp/open62541pp #116: Using random NodeIds for Variables and Objects is not possible

lbckmnn, 2023-10-30:
First of all: Thanks for this great library!
Using plain open62541 in C++ is quite painful and this project makes it so much easier.
However, I have found a small Problem:

In plain open62541 it is possible to let the library select a node id when adding objects or variables with functions like `UA_Server_addObjectNode` by passing a node id with 0 as numeric identifier (See https://www.open62541.org/doc/master/server.html#node-addition-and-deletion).

However the following Code throws a `BadParentNodeIdInvalid`
``` C++
#include "open62541pp/open62541pp.h"

int main() {
    opcua::Server server;

    // Add a variable node to the Objects node
    opcua::Node parentNode = server.getObjectsNode();
    opcua::Node object = parentNode.addObject({1, 0}, "DummyObject");
    opcua::Node myIntegerNode = object.addVariable({1, 1000}, "TheAnswer");
    // Write some node attributes
    myIntegerNode.writeDisplayName({"en-US", "The Answer"})
        .writeDataType(opcua::DataTypeId::Int32)
        .writeValueScalar(42);

    server.run();
}
```
The reason for this is, that `parentNode.addObject()` creates its return value (`Node`) by using the passed node id, and not the actual node id of the created object. 

It is possible to retrieve the actual node id by using the last parameter (`UA_NodeId *outNewNodeId`) of `UA_Server_addObjectNode`. 

So my suggestion is to use this to return a `Node` that is constructed with the actual node id. 
I created a quick draft of this in https://github.com/open62541pp/open62541pp/compare/master...lbckmnn:open62541pp:feat/random_node_ids and can do a proper PR with tests for this if welcome.

What do you think?

lukasberbuer, 2023-10-31:
Hi @lbckmnn, thank you for reporting 👍 
I didn't know about the special behavior of node id 0. I will create a PR based on your draft.

closed 2023-10-31 by lukasberbuer

lukasberbuer, 2023-11-01:
Thanks a lot for your contribution @lbckmnn! It just got released with version [v0.11.0](https://github.com/open62541pp/open62541pp/releases/tag/v0.11.0). I just have let you create the PR so you get listed as a contributer, sorry. I hope it's not the last contribution from your side :)

lbckmnn, 2023-11-03:
No worries, 
Probably not, also feel free to ask if you want help with anything :)

PR #118 (2023-10-31): fix: return output node id
Fixes #116.