open62541pp/open62541pp #117: Set operational limits of server

MioStarkid, 2023-10-31:
I cant find a way to set the operational limits of a server.
Like MaxNodesPerRead {0, 11705} or MaxNodesPerWrites {0, 11707}
with open62541 c library, this is done by setting the sever configruation.

With
```c
opcua::Server server;
```
maxnodesperread and maxnodesperwrite are 0

lukasberbuer, 2023-10-31:
Hi @MioStarkid, you can always use the `handle()` methods to get access to the underlying `UA_*` types, for example:

```cpp
#include "open62541pp/open62541pp.h"
#include "open62541/server.h"  // UA_Server_getConfig

int main() {
    opcua::Server server;
    UA_ServerConfig* config = UA_Server_getConfig(server.handle());
    config->maxNodesPerRead = 10;
    config->maxNodesPerWrite = 10;

    server.run();
}
```

MioStarkid, 2023-11-07:
Well that somehow doesnt work.
Do I have to write the config back to the server?

lukasberbuer, 2023-11-07:
How do you check if this is working?
I just checked the `ServerCapabilities` with UaExpert and `MaxNodesToRead` seems to be `0` instead of `10`. BUT I get error messages with `BadTooManyOperations`. So maybe only the value `MaxNodesToRead` is not updated?
[image]

Also tried pure open62541 with the same results:

```cpp
#include "open62541/server.h"
#include "open62541/server_config_default.h"

int main() {
    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    UA_ServerConfig* config = UA_Server_getConfig(server);
    config->maxNodesPerRead = 10;
    config->maxNodesPerWrite = 10;

    bool running = true;
    UA_Server_run(server, &running);
}
```

MioStarkid, 2023-11-07:
Ah okay our client app reads those values to split up large read and write operations.
So I guess this there is an underlaying bug in the open62541 library

lukasberbuer, 2023-11-09:
I just created an issue for open62541 https://github.com/open62541/open62541/issues/6119 and will close this issue. Please feel free to comment or open another issue if you have any questions.

closed 2023-11-09 by lukasberbuer