open62541pp/open62541pp #126: Valgrind reporting memory leak when setting the logger

Harry-George, 2023-11-08:
open62541pp version: 0.9.0
open62541 version: 1.3.8

I have the following code

```c++
#include "open62541pp/open62541pp.h"

auto Log(opcua::LogLevel, opcua::LogCategory, std::string_view) -> void {}

int main()
{
	opcua::Client client;
	client.setLogger(&Log);
	return 0;
}
```

When run under valgrind I get the following error

```
Leak_DefinitelyLost
vg_replace_malloc.c
4 bytes in 1 blocks are definitely lost in loss record 1 of 1
0x484D47B calloc vg_replace_malloc.c:1328 
0x1925FD UA_Log_Stdout_withLevel 
0x18243D UA_Client_new 
0x11BCFC opcua::Client::Connection::Connection 
0x119A0C opcua::Client::Client 
0x119801 main src/Nodes/OPC_UA/test/testOPC_UASource.cpp:500 
```

After digging in I changed my code so that before setting the logger it would clean up the old handle if it was there

```c++
#include "open62541pp/open62541pp.h"
#include "open62541/client.h"

auto Log(opcua::LogLevel, opcua::LogCategory, std::string_view) -> void{}

int main()
{
	opcua::Client client;
	auto *client_config = UA_Client_getConfig(client.handle());
	if(client_config->logger.clear) {
		client_config->logger.clear(client_config->logger.context);
	}
	client_config->logger.log = nullptr;
	client_config->logger.clear = nullptr;
	client.setLogger(&Log);
	return 0;
}
```
And it no longer reports any issues.

Am I doing something wrong or should setLogger clear the previous log handle?

lukasberbuer, 2023-11-08:
Hi @Harry-George, good catch! This free should be handled automatically.
I just provided a fix in #127. Can you please double check if this works?

Harry-George, 2023-11-08:
Thank you. I will be able to test it first thing tomorrow and will get back to you. Looking at the commit, is the stdout logging supposed to be in the custom logger destructor or is that accidently left in debugging? I would request not adding it or adding a way for the user to disable it so users can control what is output from their application.

lukasberbuer, 2023-11-08:
Thanks! Yes the stdout was for debugging, just removed it 🥴

Harry-George, 2023-11-09:
Have tested the new version and can confirm it fixed my problem. Thank you very much.

PR #127 (2023-11-09): fix: setLogger memory leak
Fixes #126
closed 2023-11-09 by lukasberbuer