open62541pp/open62541pp #129: Documentation: roadmap, timeline, completeness

jeroendoggen, 2023-11-14:
Could you add some information related to the following topics to the documentation: roadmap, timeline, completeness, project state

Roadmap & timeline:
- You recently released several versions in fast succession. Are there any plans for a 1.0.0 release or do you expect breaking API changes before a first release is possible?

Completeness & development stage

- Do you consider the library as nearly feature complete or are there some crucial missing pieces.
- Where would you place the library in the software life cycle? (alpha, beta, release candidate, ready for release?)

lukasberbuer, 2023-11-18:
Thanks @jeroendoggen for bringing this up 👍

Open62541++ is currently in `beta` stage but I am planning to release `v1.0.0` in the beginning of next year and mark it as production ready. I just created a milestone with the missing features (from my perspective) that need to be addressed until then: https://github.com/open62541pp/open62541pp/milestone/1
I expect no breaking changes. Some methods have been marked deprecated that will be removed in `v1.0.0`. But there are always alternatives with slightly different signatures, that can be easily adapted.

Do you see any missing pieces, that are not addressed in the milestone?

jeroendoggen, 2023-11-28:
Minor remarks: 
- Explain that/why the Open62541 1.3.x branch is currently used
- Do you plan to support 1.4.x in the future? 
- How do you plan to synchronize with Open62541 releases?

lukasberbuer, 2023-11-28:
All open62541 releases since `v1.0.0` are supported and tested in a CI pipeline with debug/release build and as static/dynamic library: https://github.com/open62541pp/open62541pp/blob/master/.github/workflows/open62541-compatibility.yml

Using the latest stable open62541 version gives us all the improvements open62541 gained over time. For example, more functions have been made and marked thread-safe over time. Do you see any disadvantage in using the latest version @jeroendoggen?

Open62541++ comes with the latest stable open62541 release (currently `v1.3.8`) as a submodule. Depending on the value of the CMake flag `UAPP_INTERNAL_OPEN62541`, the submodule or an external open62541 installation is used. So its up to you what open62541 version you want to use.

Open62541++ will follow the releases of open62541 with a little delay if some changes are necessary.

I will add some of those information condensed in the README. Thanks for your input!

jeroendoggen, 2023-11-29:
Thanks for the update, I did not know that these tests existed in the CI pipeline. Maybe you could also add a condensed note related to the CI setup in the README.

>  Do you see any disadvantage in using the latest version @jeroendoggen?
 
--> No

PR #141 (2023-12-08): docs: mention supported open62541 versions and CI pipeline
Closes #129.
closed 2023-12-08 by lukasberbuer