open62541pp/open62541pp #137: Integrate OPC Foundation/UA-Nodeset

jeroendoggen, 2023-11-29:
The OPC Foundation provides various node sets: https://github.com/OPCFoundation/UA-Nodeset

Whats is the recommended workflow to integrate these nodes when using open62541pp?
Should they be handled as custom datatypes?

I would be nice to have a minimal example where one of general models is used together with open62541. (i.e. [Device integration model](https://github.com/OPCFoundation/UA-Nodeset/tree/latest/DI))

lukasberbuer, 2023-12-08:
Hi @jeroendoggen, that's a very good point.

The new open6241 way to load nodesets is during runtime with the `UA_Server_loadNodeset` function. BUT this is only available in the upcoming `v1.4`. I prefer this to the old code-generation approach using Python/CMake scripts.
https://github.com/open62541/open62541/blob/v1.4.0-rc1/examples/nodeset_loader/nodeset_loader.c
Let's wait for `v1.4` and integrate it as fast as possible and provide an example.

There is one remaining issue: Custom types...
For example, the DI nodeset comes with a few custom types which are defined in the `.bsd` or `.xsd` file: https://github.com/OPCFoundation/UA-Nodeset/blob/latest/DI/Opc.Ua.Di.Types.bsd

We definately need code generation for this. Let's take the example of `ParameterResultDataType` type.
```xml
  <opc:StructuredType Name="ParameterResultDataType" BaseType="ua:ExtensionObject">
    <opc:Field Name="NoOfNodePath" TypeName="opc:Int32" />
    <opc:Field Name="NodePath" TypeName="ua:QualifiedName" LengthField="NoOfNodePath" />
    <opc:Field Name="StatusCode" TypeName="ua:StatusCode" />
    <opc:Field Name="Diagnostics" TypeName="ua:DiagnosticInfo" />
  </opc:StructuredType>
```

The open62541 generated type looks like this:
```c
typedef struct {
    size_t nodePathSize;
    UA_QualifiedName *nodePath;
    UA_StatusCode statusCode;
    UA_DiagnosticInfo diagnostics;
} UA_ParameterResultDataType;
```

Additionally, `UA_DataType`s a generated for each type and stored in the `UA_TYPES_DI` array.

I see following solution for open62541++:

### 1. Use the open62541 generated types

The same problem as with the builtin types.. it's easy to get memory leaks. But they can be wrapped with the `TypeWrapper` class.

ToDos for this approach:
- Extend `TypeWrapper` to suppose other data type arrays as `UA_TYPES`.

### 2. Generate RAII types

All builtin types are available as wrapped types, e.g. `String` for `UA_String`, ... If a `struct` only has pointer-free or wrapper types, it's safe.

A open62541++ struct could look like this:

```cpp
namespace opcua {
struct ParameterResultDataType {
    Array<QualifiedName> nodePath;
    StatusCode statusCode;  // or UA_StatusCode, it's just a uint32_t
    DiagnosticInfo diagnostics;
};
}  // namespace opcua
```

A struct may also contain optional fields (pointers). Both arrays and optional fields should be wrapped to be automatically destructed.

ToDos for this approach:
- Implement `Array`, `Optional` and `Union` wrapper
- Code generator to parse `.bsd` files and generate the header files
- Add CMake function to easily use code generation is own project

What's your take on this @jeroendoggen?

piperoc, 2024-04-16:
@lukasberbuer , just a reminder that open62541 1.4 has now been released. That has the much awaited nodeset loading feature. Maybe I can help if I get a few pointers (I have patched the NodesetLoader but never worked on this codebase)

piperoc, 2024-05-21:
@jeroendoggen  @lukasberbuer  why did you write "We definitely need code generation for this..." . I thought the current open62541 release does support importing custom types and structures.

This is very important for companion specs compliance. 
Let me know if I am not understanding this right.

lukasberbuer, 2024-05-21:
Since open62541 v1.4, custom nodesets can be loaded to populate the address space of a server.
Nodesets contain usually following files:
- `NodeSet2.xml` - The formal definition of the Nodes defined by the specification
- `Types.xsd` - The XML schema for the DataTypes defined by the specification
- `Types.bsd` - The OPC Binary schema for the DataTypes defined by the specification (obsolete)
- `NodeIds.csv` - A CSV file containing the NodeIds assigned to Nodes defined by the specification

The new `UA_Server_loadNodeset` function has following signature:
```c
UA_StatusCode UA_Server_loadNodeset(UA_Server *server, const char *nodeset2XmlFilePath, UA_NodeSetLoaderOptions *options);
```

You see, that the only input is the `NodeSet2.xml` file. So the servers address space will be populated with all the nodes.
BUT if you actually want to write a data type to a e.g. variable node, you must recreate the memory layout of the target data structure. Those information are part of the `Types.bsd` file which is used to generate the C structs/enums and the `UA_TYPES` array in open62541.

piperoc, 2024-05-21:
@lukasberbuer thank you so much, now I understand your previous comment. I actually never tested beyond just loading the address space.

aadiljaleel, 2024-08-23:
where are we at on this? :)

lukasberbuer, 2025-04-08:
#551 added `opcua::loadNodeset` to load `NodeSet2.xml` files during runtime. This works also with open62541 v1.3.

closed 2025-04-08 by lukasberbuer