open62541pp/open62541pp #142: Intended implementation of an enumeration in open62541pp

tony-p, 2023-12-09:
Is there a intended process for implementing an enum type in this library? I didn't find any examples, although open62541 has the following examples

https://github.com/open62541/open62541/issues/2032

lukasberbuer, 2023-12-10:
Hi @tony-p, yes `DataTypeBuilder::createEnum` is the right way. Did it work for you?

lukasberbuer, 2023-12-10:
Hi @tony-p, the `DataTypeBuilder::createEnum` method is for creating `UA_DataType`s of enum (classes). The `UA_DataType` describes how the type can be converted to OPC UA binary format. The enum field  names doesn't matter. So it's only important to get the underlying data type (usually `uint32_t`) for encoding/decoding.

On the other hand, the data type is described in the server's information model as a `DataType`. For example, the enumeration `ApplicationType` or `HistoryUpdateType` looks like this:
[image]

So the DataType node itself of an enumeration may have an property called `EnumStrings` (of type `LocalizedText`) or `EnumValues` (of type `EnumValueType`) to describe the meaning of the integer values.

How to add DataType nodes is described in the example: https://github.com/open62541pp/open62541pp/blob/master/examples/custom_datatypes/server_custom_datatypes.cpp#L17-L22

Just add the property `EnumStrings`/`EnumValues` to the node:
```cpp
    server.getNode(opcua::DataTypeId::Enumeration).addDataType(dataTypeColor.getTypeId(), "Color")
        .addProperty(
            {0, 0},  // auto-generate node id
            "EnumValues",
            opcua::VariableAttributes{}
                .setDataType<opcua::EnumValueType>()
                .setValueRank(opcua::ValueRank::OneDimension)
                .setArrayDimensions({0})
                .setValueArray(opcua::Span<const opcua::EnumValueType>{
                    {0, {"", "Red"}, {}},
                    {1, {"", "Green"}, {}},
                    {2, {"", "Yellow"}, {}},
                })
        )
        .addModellingRule(opcua::ModellingRule::Mandatory);
```

I will update the `server_custom_datatype` example and link this in this issue.

Btw, there is an error with `DataTypeBuilder::createEnum`.. fix will be here in a few minutes :)

tony-p, 2023-12-10:
Thanks. I have managed to get it minimally working using the branch.

lukasberbuer, 2023-12-11:
Just merged #143 it into master. Please let me know if something is missing or not working.

I know it's a bit painful to manually define the enum values in the information model, but I don't see any other more efficient way.

closed 2023-12-11 by lukasberbuer

tony-p, 2023-12-11:
Yeah I have looked into it a bit and was planning on trying magic enum https://github.com/Neargye/magic_enum it uses some tricks with modern compilers to produce a static reflection which would be enough in most instances I think to generate the names

lukasberbuer, 2023-12-12:
This is interesting @tony-p.

I just extended the `server_custom_datatypes` example to reproduce:

```cpp
    auto nodeColor = nodeObjects.addVariable(
        {1, "Color"},
        "Color",
        opcua::VariableAttributes{}
            .setDataType(dataTypeColor.getTypeId())
            .setValueRank(opcua::ValueRank::Scalar)
            .setValueScalar(Color::Green, dataTypeColor)
    );

    // read custom enum type
    {
        auto variant = nodeColor.readValue();
        std::cout << "typeId: " << opcua::NodeId(variant.getDataType()->typeId).toString() << std::endl;
        // typeId: ns=1;i=4946
        std::cout << "value: " << *static_cast<int32_t*>(variant.data()) << std::endl;
        // value: 1
    }

    // write & read enum as int
    {
        nodeColor.writeValueScalar(static_cast<int32_t>(Color::Yellow));

        auto variant = nodeColor.readValue();
        std::cout << "typeId: " << opcua::NodeId(variant.getDataType()->typeId).toString() << std::endl;
        // typeId: i=6
        std::cout << "value: " << *static_cast<int32_t*>(variant.data()) << std::endl;
        // value: 2
    }
```

Enumerations are always treated as unsigned integers of a specific length: https://reference.opcfoundation.org/Core/Part5/v105/docs/C.2.4

So we don't need to write those as custom datatypes but can instead use the underlying data type (`int32_t` in this case).

lukasberbuer, 2023-12-19:
Hi @tony-p,

I tried to implement the modern way using the `DataTypeDefinition` attribute and `EnumDefinition` in #145.
But the `DataTypeDefinition` AttributeId is currently not supported in open62541 v1.3 and also v1.4.

So currently, we have to stick with the `EnumValueType` property...

lbckmnn, 2025-02-06:
> Hi [@tony-p](https://github.com/tony-p),
>
> I tried to implement the modern way using the `DataTypeDefinition` attribute and `EnumDefinition` in [...]

Hey @lukasberbuer 
Do you know if this is still the case?

I am trying to get an asyncua client to work with an open62541pp server with a custom data structure which contains a custom enum.

However, asyncua seems to only support the `DataTypeDefinition` based enums, when used in custom structures...
The `EnumValueType` based enums are supported with `client.load_enums()`, but apparently not in custom structures (`client.load_data_type_definitions()` results in a `RuntimeError: Unknown datatype for field..` in this case).

tony-p, 2025-02-06:
> Hey [@lukasberbuer](https://github.com/lukasberbuer) Do you know if this is still the case?
>
> I am trying to get an asyncua client to work with an open62541pp server with a custom data structure [...]

(I haven’t done any experimentation) but some recent work with asyncua I came across that they still support both legacy and 1.04 data structures. There is a get data structures for each specification and both will only get the custom structures for that specification, however the legacy version will output a deprecation warning to terminal