open62541pp/open62541pp #144: Problems while building and installing

Razuox, 2023-12-18:
Hey there,
im new to using open62541pp as well as C++. When trying to build and install open62541pp for Visual Studio 15 2017 following [the guide](https://github.com/open62541pp/open62541pp#build-and-install) I face the problem (which I believe is) the lib is not getting build successful.

I guess im missing something essential which is not obvious to me.

Running ```cmake -DCMAKE_BUILD_TYPE=Debug -DUAPP_BUILD_EXAMPLES=ON -DUAPP_BUILD_TESTS=ON ..```, ```cmake --build .``` and ```ctest --output-on-failure``` returns the output found in [log.txt](https://github.com/open62541pp/open62541pp/files/13701957/log.txt).

First of all: I was not able to get the command prompt to output all logs in english, so I translated the parts from my original language by hand using DeepL.

Output from first command is from line 1 to 32:
I am not sure if the skipped checks for CXX Compiler and C Compiler are fine or what I can do about it.

Output from second command is from line 33 to 606:
What do the C4100 warnings mean or do I need to pay further attention to them?

Output from third command is from line 607 to 1244
Why do the "connecting" tests fail? Do I need to configure them to a test OPC UA server so they won't fail?

Am I right in assuming that as soon as the execution of the commands works properly, the library in C++ can also be used to its full extent?

Please let me know if you need more information to answer this question.
Thanks in advance,
Razuox

lukasberbuer, 2023-12-18:
Hi @Razuox,
the compilation of the library was successful, but some client/server tests are failing.
The error message for most tests is something like this:

```
C:\WORK\CppLibs\open62541pp\tests\Client.cpp(20): ERROR: test case THREW exception: BadCommunicationError
```

The `BadCommunicationError` is described as: `A low level communication error occurred`. I guess this has something to do with your firewall. The client/server tests are based on a running OPC UA server instance.
You can try to start one of the examples `./examples/server_minimal.exe` and see if you get any firewall warning and if you can access the server with some client like UaExpert.

Razuox, 2023-12-18:
Hi @lukasberbuer,
thanks for this very fast response.

I tried executing ```./examples/server_minimal.exe``` and it failed due to an error while binding a socket. I believe this happens due to restrictions from my IT provider.
Well this is not that important, because i don't want to use the server side of the SDK in the future.

So am i right believing that the lib has been build successfully and my next step would be to set the right ```Additional Include Directories```, ```Additional Library Directories``` and ```Additional Dependencies``` in the Visual Studio Project?

Regarding the last step ```cmake --install .```: when doing so, the output is the following:
```
-- Install configuration: "Release"
CMake Error at cmake_install.cmake:39 (file):
  file INSTALL cannot find
  "C:/WORK/CppLibs/open62541pp/build/bin/Release/open62541pp.lib": No error.
``` 
To me this is quite confusing, because I built the lib for Debug and now the output states "Release". And the next line states: "Error.... No error.".
Pardon me if this question is nonsensical or I don't understand something obvious.

lukasberbuer, 2023-12-18:
Hi @Razuox,

you probably want to add the `--config Debug|Release` flag for the `cmake --install` command:
https://cmake.org/cmake/help/latest/manual/cmake.1.html#cmdoption-cmake-install-config

I am not a Visual Studio user, but I should be as simple as that:
- provide the include directory for the header files
- provide the path to the compiled library

Razuox, 2023-12-19:
Hi @lukasberbuer,

using the ```--config Debug``` helps installing it successful, thanks.

Providing the include directory as well as the lib directory to Visual Studio results in Link Errors
```
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol ___WSAFDIsSet@8 referenced in function _UA_ClientConnectionTCP_poll
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__accept@12 referenced in function _ServerNetworkLayerTCP_listen
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__bind@12 referenced in function _addServerSocket
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__closesocket@4 referenced in function _purgeFirstConnectionWithoutChannel
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__connect@12 referenced in function _UA_ClientConnectionTCP_poll
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__getsockname@12 referenced in function _addServerSocket
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__getsockopt@20 referenced in function _UA_ClientConnectionTCP_poll
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__htonl@4 referenced in function _addServerSocket
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__listen@8 referenced in function _addServerSocket
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__ntohs@4 referenced in function _addServerSocket
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__recv@16 referenced in function _connection_recv
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__select@20 referenced in function _UA_ClientConnectionTCP_poll
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__send@16 referenced in function _connection_write
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__setsockopt@20 referenced in function _ServerNetworkLayerTCP_add
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__shutdown@8 referenced in function _ServerNetworkLayerTCP_close
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__socket@12 referenced in function _UA_ClientConnectionTCP_poll
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__gethostname@8 referenced in function _ServerNetworkLayerTCP_start
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__WSAGetLastError@0 referenced in function _UA_ClientConnectionTCP_init
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__WSAPoll@12 referenced in function _connection_write
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__getaddrinfo@16 referenced in function _UA_ClientConnectionTCP_init
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__freeaddrinfo@4 referenced in function _ServerNetworkLayerTCP_start
1>open62541.lib(network_tcp.obj) : error LNK2019: unresolved external symbol __imp__getnameinfo@28 referenced in function _ServerNetworkLayerTCP_add
```

Even though every error is about server or client connection again, I believe this is not caused by the firewall, because I tested it on another system which has another security provider. The tests while installing get passed but the errors above are still the same.

Since you are not using VS I suppose you don't know how to fix these. Maybe some further Google research or some other user can help solving this problem.

Cheers Razuox

lukasberbuer, 2023-12-19:
Hi @Razuox,
the linking errors occur for the open62541 functions. You need to specify also the include/library directory of open62541.
Please let me know if this fixes the linking problems.

Razuox, 2023-12-19:
Hi @lukasberbuer,

so I have built open62541 as well as open62541pp regarding the instructions. This worked totally fine. I have added the include directories as well as the library directories to my project as seen below:
[image]
[image]
Doing so results in following link errors while building:
<details>
<summary>Error List</summary>

```
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol _UA_StatusCode_name referenced in function "public: virtual char const * __thiscall opcua::BadStatus::what(void)const " (?what@BadStatus@opcua@@UBEPBDXZ)
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol _UA_copy referenced in function "class opcua::DateTime __cdecl opcua::detail::copy<class opcua::DateTime>(class opcua::DateTime const &,struct UA_DataType const &)" (??$copy@VDateTime@opcua@@@detail@opcua@@YA?AVDateTime@1@ABV21@ABUUA_DataType@@@Z)
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol _UA_clear referenced in function "void __cdecl opcua::detail::clear<struct UA_DataValue>(struct UA_DataValue &,struct UA_DataType const &)" (??$clear@UUA_DataValue@@@detail@opcua@@YAXAAUUA_DataValue@@ABUUA_DataType@@@Z)
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol "public: class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __thiscall opcua::DateTime::format(class std::basic_string_view<char,struct std::char_traits<char> >,bool)const " (?format@DateTime@opcua@@QBE?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$basic_string_view@DU?$char_traits@D@std@@@4@_N@Z) referenced in function _main
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol "public: struct UA_DataType const * __thiscall opcua::Variant::getDataType(void)const " (?getDataType@Variant@opcua@@QBEPBUUA_DataType@@XZ) referenced in function "private: void __thiscall opcua::Variant::checkIsDataType<class opcua::DateTime>(void)const " (??$checkIsDataType@VDateTime@opcua@@@Variant@opcua@@ABEXXZ)
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol "private: void __thiscall opcua::Variant::checkIsScalar(void)const " (?checkIsScalar@Variant@opcua@@ABEXXZ) referenced in function "public: class opcua::DateTime const & __thiscall opcua::Variant::getScalar<class opcua::DateTime>(void)const " (??$getScalar@VDateTime@opcua@@@Variant@opcua@@QBEABVDateTime@1@XZ)
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol "public: class opcua::Variant & __thiscall opcua::DataValue::getValue(void)" (?getValue@DataValue@opcua@@QAEAAVVariant@2@XZ) referenced in function "class opcua::Variant __cdecl opcua::services::readValue<class opcua::Client>(class opcua::Client &,class opcua::NodeId const &)" (??$readValue@VClient@opcua@@@services@opcua@@YA?AVVariant@1@AAVClient@1@ABVNodeId@1@@Z)
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol "public: __thiscall opcua::NodeId::NodeId(unsigned short,unsigned int)" (??0NodeId@opcua@@QAE@GI@Z) referenced in function "public: __thiscall opcua::NodeId::NodeId(enum opcua::VariableId)" (??0NodeId@opcua@@QAE@W4VariableId@1@@Z)
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol "public: __thiscall opcua::Client::Client(void)" (??0Client@opcua@@QAE@XZ) referenced in function _main
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol "public: void __thiscall opcua::Client::connect(class std::basic_string_view<char,struct std::char_traits<char> >)" (?connect@Client@opcua@@QAEXV?$basic_string_view@DU?$char_traits@D@std@@@std@@@Z) referenced in function _main
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol "public: class opcua::Node<class opcua::Client> __thiscall opcua::Client::getNode(class opcua::NodeId)" (?getNode@Client@opcua@@QAE?AV?$Node@VClient@opcua@@@2@VNodeId@2@@Z) referenced in function _main
1>Open62541ppTest.obj : error LNK2019: unresolved external symbol "class opcua::DataValue __cdecl opcua::services::readAttribute<class opcua::Client>(class opcua::Client &,class opcua::NodeId const &,enum opcua::AttributeId,enum opcua::TimestampsToReturn)" (??$readAttribute@VClient@opcua@@@services@opcua@@YA?AVDataValue@1@AAVClient@1@ABVNodeId@1@W4AttributeId@1@W4TimestampsToReturn@1@@Z) referenced in function "class opcua::Variant __cdecl opcua::services::readValue<class opcua::Client>(class opcua::Client &,class opcua::NodeId const &)" (??$readValue@VClient@opcua@@@services@opcua@@YA?AVVariant@1@AAVClient@1@ABVNodeId@1@@Z)
1>Open62541ppTest.obj : error LNK2001: unresolved external symbol _UA_TYPES
```

</details>

Then i tried adding the ```.lib``` files as additional dependencies:
[image]

This results in:
<details>
<summary>Error List</summary>

```
1>LINK : warning LNK4098: defaultlib 'LIBCMTD' conflicts with use of other libs; use /NODEFAULTLIB:library
1>open62541pp.lib(Client.obj) : error LNK2019: unresolved external symbol _UA_Client_connect referenced in function "public: void __thiscall opcua::Client::connect(class std::basic_string_view<char,struct std::char_traits<char> >)" (?connect@Client@opcua@@QAEXV?$basic_string_view@DU?$char_traits@D@std@@@std@@@Z)
1>open62541pp.lib(Session.obj) : error LNK2019: unresolved external symbol _UA_Server_setSessionParameter referenced in function "public: void __thiscall opcua::Session::setSessionAttribute(class opcua::QualifiedName const &,class opcua::Variant const &)" (?setSessionAttribute@Session@opcua@@QAEXABVQualifiedName@2@ABVVariant@2@@Z)
[... cut ...]
1>open62541.lib(eventloop_posix_select.obj) : error LNK2019: unresolved external symbol __imp__select@20 referenced in function _UA_EventLoopPOSIX_pollFDs
```
</details>

Going onward from here I believe, that the first step was the correct one and all steps afterwards were misleading, because there were more and more errors going onward.

Do you have any suggestions looking at the first error log? (If you do, is it possible that you explain which part of the log is giving you a hint so I may be able to unterstand these logs better?)

Cheers Razuox

lukasberbuer, 2023-12-19:
Hi @Razuox, can you please try to just add following pragmas at the top of your cpp file:

```cpp
#pragma comment(lib, "open62541")
#pragma comment(lib, "open62541pp")
```

Razuox, 2023-12-19:
Hi @lukasberbuer,

it seems like this is just doing exactly the same as ```Then i tried adding the .lib files as additional dependencies:``` from my post above, because it results in the same error list.

lukasberbuer, 2023-12-19:
HI @Razuox, I just installed Visual Studio on my system to reproduce the error.
Following lines fix the linker errors for me:

```cpp
#pragma comment(lib, "open62541")
#pragma comment(lib, "open62541pp")
#pragma comment(lib, "Ws2_32")
```

Razuox, 2023-12-20:
Hi @lukasberbuer,
a big thank you for your quick and effective help and even installing VS to reproduce my problem!
The library is now working and I can finally continue working with it.
You made my week. Merry christmas and have nice holiday :)
Cheers Razuox

closed 2023-12-20 by Razuox

lukasberbuer, 2023-12-20:
I'm glad this helped @Razuox! I wish you merry christmas and happy holidays aswell!!

I guess it would have been way easier with vcpkg and its msbuild integration. That's the next step :)