open62541pp/open62541pp #148: question about changing node values on running server

luminize, 2024-01-04:
Hi,

I'm integrating a OPCUA server in a demo game for the Godot game engine so students can get their feet wet with a SCADA system. The idea is that from a SCADA system we flip a switch, and the game takes over (do some task). After completing the task, the game should change an OPC variable, (which is readable by the SCADA and we can monitor that).
I started out with the open62541.org C code, and then (since the plugin system in Godot uses C++) found your C++ wrapper which helped me getting started quickly. Thanks for your work!

So far I've built an example Godot extension with the example OPC-UA server starting and visible after connecting UAExpert (yay!)

Any thoughts on how to best implement exchange of values with the running server? 

Best,
Bas

luminize, 2024-01-05:
Answering my own question :)
I had fine results with calling `browseChild().writeValueScalar` for variables.
```c++
auto nodePosition = _server.getNode({_ns_i, 20001});
nodePosition.browseChild({{_ns_i, "X"}}).writeValueScalar(vec.x);
nodePosition.browseChild({{_ns_i, "Y"}}).writeValueScalar(vec.y);
```
Thanks!

closed 2024-01-05 by luminize

lukasberbuer, 2024-01-08:
Hi @luminize,
Yes, you found the perfect answer 👍 If the node id of the "X" and "Y" nodes is not known in advance, a `browseChild` call is necessary. I would recommend to store the results of the browse operation (either the `Node` objects or just the `NodeId`s) for better performance.