open62541pp/open62541pp #151: Problems while trying to use opcua::NodeId

Razuox, 2024-01-12:
Hey there,

i am trying to read nodes based on configuration i am reading from an xml file. (```child.attribute...```)

Connection to the server is set using ```client.connect(endPointUrl)```.
Now i am trying to create ```opcua::NodeId``` based on given NodeId Strings -> "ns=2;s=Tag1":
```
// these return "ns=2;s=Tag1"
std::string nodeidstring = child.attribute("VarName").value();
const char* nodeidchar = child.attribute("VarName").value();

std::string tagstring = "Tag1";
std::string_view tagstringview(tagstring.data(), tagstring.size());
opcua::String tagopcstringindirect = opcua::String(tagstringview);

opcua::String tagopcstringdirect = opcua::String("Tag1");

opcua::ByteString tagbytestring = opcua::ByteString(tagstringview);

auto test1 = UA_NODEID(nodeidchar);
auto test2 = opcua::NodeId(2, tagopcstringindirect);
auto test3 = opcua::NodeId(2, tagopcstringdirect);
auto test4 = opcua::NodeId(2, tagbytestring);
```

When inspecting these NodeIds the identifier is not ```Tag1``` like i would expect, but some weird string:
[image]

I believe i am doing something wrong with encoding the strings.

I also tried reading the NodeId using your example [client browse](https://github.com/open62541pp/open62541pp/blob/master/examples/client_browse.cpp) and calling ```auto temp = child.getNodeId().getIdentifier();``` on the child nodes in printNodeTree. This also returns an interesing identifier:
[image]

I am a bit stuck on this. Maybe someone can help me with this.
If you need more information or don't understand my problem please let me know.

Thanks in advance,
Razuox

lukasberbuer, 2024-01-12:
Hi @Razuox,
I guess the screenshots are from your IDE?
The problem might be the representation of strings in open62541:

```c
typedef struct {
    size_t length; /* The length of the string */
    UA_Byte *data; /* The content (not null-terminated) */
} UA_String;
```

`UA_String` is just an array with bytes (`UA_Byte` is an alias for `uint8_t`). But a string in C++ is usually an array of `char`. 
If a `UA_String` is created from a `char` array, it is just casted (`reinterpret_casted) to `UA_Byte`:

```c
static UA_String
UA_STRING(char *chars) {
    UA_String s; s.length = 0; s.data = NULL;
    if(!chars)
        return s;
    s.length = strlen(chars); s.data = (UA_Byte*)chars; return s;
}
```

So your IDE tries to print a casted array of bytes, but it's a array of char. This results in wrong encoding.
If you inspect the `NodeId` with an OPC UA client, it should be as expected. Can you verify this?

Razuox, 2024-01-12:
Hi @lukasberbuer ,

yes these are screenshots from my IDE.
So I tried writing to to the nodes using the weird looking identifiers and it works (confirmed by looking at the server using UaExpert). So I believe it was working the whole time and I only thought my problem were the weird looking identifiers, when in reality my problem is reading the node values correctly.
With this I am a bit confused which read function I have to use in order to access the data correctly.

I tried reading the data according to your example using ```const auto dt = node.readValueScalar<opcua::DateTime>();```
Can you give some example or a link where I can see how to read different datatypes, because when trying to use ```const auto dt = node.readValueScalar<opcua::DataTypeId::Int16>();``` the datatype doesn't match the argument list.

Just like problem while installing I believe this should be a basic thing and I am just missing out on something.

Cheers,
Razuox

lukasberbuer, 2024-01-12:
The `Node::readValueScalar<T>` method does a lot of stuff:
1. Read request
2. Get `DataValue` within read response
3. Get the value -> `Variant`
4. Try to read a scalar value from the `Variant`
5. (Try to convert the scalar value to the provided return type `T` if necessary)

The template parameter `T` is just your desired return type:
- If you want to read an integer: `readValueScalar<int>`
- If you want to read a string: `readValueScalar<std::string>` (conversion needed) or `readValueScalar<opcua::String>` (no conversion necessary)

I hope this helps.

Razuox, 2024-01-13:
Thank you again. Your responses helped me a lot.

Cheers,
Razuox

closed 2024-01-13 by Razuox