open62541pp/open62541pp #161: Could not find a package configuration file provided by "open62541pp::open62541pp"

oliver, 2024-01-16:
I've used the example code from https://github.com/open62541pp/open62541pp#integrate-as-a-pre-compiled-library to add open62541pp to my CMake project:

```
find_package(open62541pp::open62541pp CONFIG REQUIRED)
```

That gave this CMake error 

```
...
CMake Error at test/CMakeLists.txt:35 (find_package):
  Could not find a package configuration file provided by
  "open62541pp::open62541pp" with any of the following names:

    open62541pp::open62541ppConfig.cmake
    open62541pp::open62541pp-config.cmake

  Add the installation prefix of "open62541pp::open62541pp" to
  CMAKE_PREFIX_PATH or set "open62541pp::open62541pp_DIR" to a directory
  containing one of the above files.  If "open62541pp::open62541pp" provides
  a separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
```

I have then changed the `find_package` line to
```
find_package(open62541pp CONFIG REQUIRED)
```
(i.e. removed the `open62541pp::` part) and that worked.

**Do I need to change anything else in my project to make the original syntax work? Or is this an error in the README file?**

I'm using CMake 3.20.5 under Debian 12. I have compiled and installed open62541pp to `/tmp/pp1/`, and have added
```
set(CMAKE_PREFIX_PATH "${CMAKE_PREFIX_PATH};/tmp/pp1/")
```
to my CMakeLists.txt to make the lib available.

lukasberbuer, 2024-01-16:
Hi @oliver,

you are absolutely right, it should be `find_package(open62541pp CONFIG REQUIRED)`.
I will change it in the README. Thanks for pointing this out 👍

Just out of curiosity: Why don't you integrate open62541pp as a submodule?

oliver, 2024-01-16:
Thanks for the quick response!

PR #162 (2024-01-16): docs: fix find_package command for pre-compiled library integration
Closes #161.