open62541pp/open62541pp #164: Setting an array with type bool does not work (assertion error or segfault)

lbckmnn, 2024-01-17:
# Problem
The very unfortunate specialization for `std::vector<bool>` is causing some troubles with Variant.setArray():
Consider the following example:

``` C++
int main() {
  Variant variant;
  std::vector<bool> vector{false, true, true, false};
  //variant.setArrayCopy(vector); // does not compile
  variant.setArrayCopy(vector.begin(), vector.end()); // fails with an assertion / segfaults
}
```
This code fails with the assertion:
```
constexpr T opcua::detail::copy(const T&, const UA_DataType&) [with T = std::_Bit_reference; UA_DataType = UA_DataType]:
```
In release build it even fails with a segfault.

# Investigation
The underlying problem is, that accessing an iterator of a `std::vector<bool>` yields a `Bit_reference` and not a bool.
With e.g. an `std::vector<int>` accessing an iterator yields an int.

https://github.com/open62541pp/open62541pp/blob/7e75bd39080832837cce87b09bc076500a783adb/include/open62541pp/types/Variant.h#L428-L436

The lambda in `std::transform` receives a Bit_reference and passes that into `detail::copy`, which expects a bool.

# Other observations
Using an `std::array<bool>` works just fine and `Variant.getArrayCopy` is also unproblematic. 
However, `Variant.setArrayCopy(std::vector<bool>)` does not compile because it is (fortunately) not convertiable to a `Span`

lbckmnn, 2024-01-17:
I can also take a further look at this and help or provide a fix if wanted.

lukasberbuer, 2024-01-17:
Great catch and investigation, thank you! 👍

As you pointed out, the problem is the `std::transform` of a `vector<bool>` in the `getArrayCopyImpl` method.
The type of the `value` parameter of the lambda is actually a `std::vector<bool>::reference`: https://en.cppreference.com/w/cpp/container/vector_bool/reference

Luckily, the reference type can be converted to a bool with a simple `static_cast`. So this simple modification should fix the problem:

```cpp
    std::transform(first, last, native.get(), [&](auto&& value) {
        return detail::copy(static_cast<ValueType>(value), dataType);
        //                  ^^^^^^^^^^^^^^^^^^^^^^
    });
```

It shouldn't have any consequences for other types. Did I miss something?
Would you provide a patch with a comment why we changed this?

lbckmnn, 2024-01-17:
> It shouldn't have any consequences for other types. Did I miss something?

I can't think of any. 

>  Would you provide a patch with a comment why we changed this?

sure

What about the other overload of setArrayCopy?
``` C++
template <typename ArrayLike, typename = EnableIfNoSpan<ArrayLike>>
void setArrayCopy(ArrayLike&& array) noexcept {
    setArrayCopy(Span{std::forward<ArrayLike>(array)});
 }
```
I stumbled across this issue because I used `setArrayCopy` inside a template, which then could not be instantiated with bool. It is possible to work around that, but it is annoying. 

We could go for something like
``` C++
template <typename ArrayLike, typename = EnableIfNoSpan<ArrayLike>>
void setArrayCopy(ArrayLike&& array) noexcept {
    if constexpr(std::is_same_v<ArrayLike, std::vector<bool>>) {
        setArrayCopy(array.begin(), array.end())
    }
    else {
        setArrayCopy(Span{std::forward<ArrayLike>(array)});
    }
}
```

However, I have not thought this through, or even tried yet and it is certainly not pretty...

lukasberbuer, 2024-01-18:
We could make it more generic and check if a `Span` can be constructed with the `ArrayLike` type. If not, fall back to the iterator approach:

```cpp
template <typename ArrayLike, typename = EnableIfNoSpan<ArrayLike>>
void setArrayCopy(ArrayLike&& array) noexcept {
    using ValueType = typename std::remove_reference_t<ArrayLike>::value_type;
    using SpanType = Span<const ValueType>;
    if constexpr(std::is_constructible_v<SpanType, ArrayLike>) {
        setArrayCopy(std::span{std::forward<ArrayLike>(array)});
    } else {
        setArrayCopy(array.begin(), array.end());
    }
}
```

Minimal example here: https://godbolt.org/z/eeK8Eb1e9

Btw, the overload should not be `noexcept`...

lukasberbuer, 2024-01-18:
The same logic would be needed for the `Variant::fromArray` methods:

```cpp
    /// @overload
    template <typename ArrayLike, typename = EnableIfNoSpan<ArrayLike>>
    [[nodiscard]] static Variant fromArray(ArrayLike&& array) {
        return Variant::fromArray(Span{std::forward<ArrayLike>(array)});
    }

    /// @overload
    template <typename ArrayLike, typename = EnableIfNoSpan<ArrayLike>>
    [[nodiscard]] static Variant fromArray(ArrayLike&& array, const UA_DataType& dataType) {
        return Variant::fromArray(Span{std::forward<ArrayLike>(array)}, dataType);
    }
```

Any idea to avoid the code duplication?

lbckmnn, 2024-01-18:
> We could make it more generic and check if a Span can be constructed with the ArrayLike type. If not, fall back to the iterator approach:

This is way better.

> Any idea to avoid the code duplication?

I'm having a hard time grasping all the `Variant::fromArray` overloads and what they do in which case (copy, no copy.). 
But I wonder if we can somehow use the overloads of `Variant::setArray` in a generic way.

lbckmnn, 2024-01-22:
I added a PR to already fix the setArrayCopy overloads.
The `Variant::fromArray` are a little bit harder.

lukasberbuer, 2024-01-23:
Thanks @lbckmnn for the PR!

I am also not happy with the loads of overloads. Let me try to explain some reasons for the status quo:

`UA_Variant` only supports contiguous C-style arrays so `Span` is a good fit for that.
- `Span<T>` the array can be assigned to the variant
- `Span<const T>` the array must be copied to the variant

If there was only a method `Variant::setScalar(Span<T> array)`, we can not directly pass e.g. a `std::vector<int>` or `std::array<int, 3>` to the method although a span can be constructed with those. Why? The template type `T` can not be deduced: https://godbolt.org/z/bYx4qvEbe
The `Variant::setScalar(ArrayLike&&)` fixes this issue. You can argue, that the span overload isn't need anymore in the public interface. I left it there mainly for documentation purposes how this methods can be used.

If the container is not contiguous, assignment is not possible and the copies are more complex (alloc new contiguous array, copy elements element-wise).

All those checks have to happen in the `Variant::fromArray` method to choose the most efficient way to store a container in a variant. I tried to make a flow chart to summarize all decisions:

[image]

lukasberbuer, 2024-01-26:
Hi @lbckmnn,

I refactored and simplified the `Variant::fromArray`, `Variant::setArray` and `Variant::setArrayCopy` methods.
The explicit `Span` overloads are now fully replaced by the `Array&&` methods.

One key was the implementation of the trait to check if an container is contiguous. Unexpectedly, `std::vector<bool>` was classified as contiguous by checking the `data()` and `size()` methods with GCC 11.
I stripped down the case in compiler explorer: https://godbolt.org/z/hdE9xPa9d

The solution was to also check if the return type of `data()` is a pointer and not a proxy object:
```cpp
std::is_pointer<decltype(std::declval<Container>().data())>
```

Please review and comment the PR #174.

lbckmnn, 2024-01-27:
Hey @lukasberbuer 

Thanks for the explanation, that helped a lot. 
At first glance, the PR looks very good!

I will review the PR but I might need a few days.

lbckmnn, 2024-01-29:
@lukasberbuer 
This is not really in scope of this issue, but it think it might be a good idea to revisit the fromArray/fromScalar methods in general. 
It is not completely obvious for the user, if these methods copy the data or use it directly. 
E.g. a `std::vector<bool>` will be copied, while a `std::vector<int>` will be used directly. 
This can cause lifetime issues and other bugs.

lukasberbuer, 2024-01-29:
> @lukasberbuer This is not really in scope of this issue, but it think it might be a good idea to revisit the fromArray/fromScalar methods in general. It is not completely obvious for the user, if [...]

That's true. My intend was to make those as efficient as possible and to only copy the data if necessary. What is your suggestion to make the internal behavior more explicit?

lukasberbuer, 2024-01-29:
One idea could be to add a `VariantPolicy` template type to specify the behaviour, e.g.:

```cpp
template <typename ArrayLike, typename VariantPolicy>
[[nodiscard]] Variant Variant::fromArray(ArrayLike&& array);

struct VariantPolicyAssignOnly {};
struct VariantPolicyCopyOnly {};
struct VariantPolicyAssignIfPossible {};
```

The policy could be provided as a template-only parameter or an additional argument:
```cpp
template <typename ArrayLike, typename VariantPolicy>
[[nodiscard]] Variant Variant::fromArray(ArrayLike&& array, VariantPolicy policy);
```

closed 2024-02-10 by lukasberbuer