open62541pp/open62541pp #165: Delete Findopen62541.cmake?

traversaro, 2024-01-18:
Thanks a lot of the work on this library! 

In the last days I worked with it, and during the build I was confused by the presence of the `cmake/Findopen62541.cmake` file in the repo.

In particular, it seems that the file is never actually used, as:
* When building the software, the `Findopen62541.cmake` is placed in the `CMAKE_MODULE_PATH` in https://github.com/open62541pp/open62541pp/blob/7e75bd39080832837cce87b09bc076500a783adb/CMakeLists.txt#L24, but then it is not used as `CMAKE_FIND_PACKAGE_PREFER_CONFIG` is set to `ON` in https://github.com/open62541pp/open62541pp/blob/7e75bd39080832837cce87b09bc076500a783adb/CMakeLists.txt#L99 .
* The file is also installed, but then the directory in which it is contained is never added to the `CMAKE_MODULE_PATH`

I am missing some case in which the file is actually useful, for example some specific case in which open62541 is installed without CMake config files, or the file could be deleted?

lukasberbuer, 2024-01-18:
Hi @traversaro,

The original idea behind the `Findopen62541.cmake` was to support just copied open62541 headers and binaries. 
In previous versions of open62541, pre-compiled libraries were included in the releases: https://github.com/open62541/open62541/releases/tag/v1.0

I don't have any good argument to keep the file. And it definitely should not be installed by open62541++. Do you agree?

traversaro, 2024-01-23:
Thanks for the fast reply!

> The original idea behind the Findopen62541.cmake was to support just copied open62541 headers and binaries.
In previous versions of open62541, pre-compiled libraries were included in the releases: https://github.com/open62541/open62541/releases/tag/v1.0

Thanks, I have no idea about this. 

> I don't have any good argument to keep the file. And it definitely should not be installed by open62541++. Do you agree?

I think this make indeed sense, unless there is somewhere in particular still interested in building form that binaries. The tricky aspect indeed is that if someone builds using those binaries, then `find_package(open62541pp)` will not work, as the `find_package(open6241)` call inside the `open62541pp-config.cmake` would fail (and trying to get to work I guess was the rationale behind installing `Findopen62541.cmake). So I totally agree in removing that file, if you like I can prepare a PR doing this.

lukasberbuer, 2024-01-23:
Ok let's go for it 👍 
I'll be happy if you prepare a PR.

closed 2024-01-23 by lukasberbuer