open62541pp/open62541pp #182: logger

Stoffels, 2024-02-03:
I used the normal instructions for compiling, but it fails with this :
>>>
.....
[ 45%] Built target open62541-object
Scanning dependencies of target open62541
[ 47%] Linking C static library ../../bin/libopen62541.a
[ 47%] Built target open62541
Scanning dependencies of target open62541pp
[ 48%] Building CXX object CMakeFiles/open62541pp.dir/src/AccessControl.cpp.o
[ 48%] Building CXX object CMakeFiles/open62541pp.dir/src/Client.cpp.o
/workspace/programming/OPCUA/open62541pp/src/Client.cpp: In constructor ‘opcua::Client::Connection::Connection()’:
/workspace/programming/OPCUA/open62541pp/src/Client.cpp:150:39: error: ‘UA_ClientConfig’ {aka ‘struct UA_ClientConfig’} has no member named ‘logger’
  150 |           logger_(getConfig(client_)->logger) {
      |                                       ^~~~~~
gmake[2]: *** [CMakeFiles/open62541pp.dir/build.make:95: CMakeFiles/open62541pp.dir/src/Client.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:409: CMakeFiles/open62541pp.dir/all] Error 2
gmake: *** [Makefile:182: all] Error 2
>>>

How can I fix this.
Thank you
Herman

lukasberbuer, 2024-02-03:
Hi @Stoffels, can you please provide your CMake configuration output from `cmake -L`?
Are you using open62541 v1.4? Some breaking changes were introduced there, including the logger handling.

Stoffels, 2024-02-03:
Dear Sir Berbuer
I am using v1.3 at this time of th open62541 opc library
and the CMake configuration:
cmake -L
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory
"/home/herman/workspace/programming/OPCUA/open62541pp/build" does not
appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
-- Cache values
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
DOCTEST_NO_INSTALL:BOOL=OFF
DOCTEST_USE_STD_HEADERS:BOOL=OFF
DOCTEST_WITH_MAIN_IN_STATIC_LIB:BOOL=ON
DOCTEST_WITH_TESTS:BOOL=OFF
UAPP_BUILD_DOCUMENTATION:BOOL=OFF
UAPP_BUILD_EXAMPLES:BOOL=ON
UAPP_BUILD_TESTS:BOOL=ON
UAPP_BUILD_TESTS_AUTORUN:BOOL=OFF
UAPP_ENABLE_CLANG_TIDY:BOOL=OFF
UAPP_ENABLE_COVERAGE:BOOL=OFF
UAPP_ENABLE_INCLUDE_WHAT_YOU_USE:BOOL=OFF
UAPP_ENABLE_PCH:BOOL=OFF
UAPP_ENABLE_SANITIZER_ADDRESS:BOOL=OFF
UAPP_ENABLE_SANITIZER_LEAK:BOOL=OFF
UAPP_ENABLE_SANITIZER_MEMORY:BOOL=OFF
UAPP_ENABLE_SANITIZER_THREAD:BOOL=OFF
UAPP_ENABLE_SANITIZER_UNDEFINED_BEHAVIOR:BOOL=OFF
UAPP_INTERNAL_OPEN62541:BOOL=ON
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=OFF
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=OFF
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=OFF
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_FORCE_WERROR:BOOL=OFF;OFF
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=REDUCED

Kind Regards
Herman

lukasberbuer, 2024-02-03:
This is weird. Can you please have a look at the file `3rdparty/open62541/include/open62541/client.h`, especially the definition of `UA_ClientConfig`. It should look like this:

```c
typedef struct {
    void *clientContext; /* User-defined pointer attached to the client */
    UA_Logger logger;    /* Logger used by the client */
    UA_UInt32 timeout;   /* Response timeout in ms */
   
    // ...
} UA_ClientConfig;
```

Stoffels, 2024-02-03:
I have tried it on 2 machines. The one is running ubuntu, and the other
debian.

My client.h:

struct UA_ClientConfig {
    void *clientContext;       /* User-defined pointer attached to the
client */
    const UA_Logger *logging;  /* Plugin for log output */

    /* Response timeout in ms (0 -> no timeout). If the server does not
answer a
     * request within this time a StatusCode UA_STATUSCODE_BADTIMEOUT is
     * returned. This timeout can be overridden for individual requests by
     * setting a non-null "timeoutHint" in the request header. */
    UA_UInt32 timeout;

Strange, it is 'logging'. weird.

Kind regards
Herman

lukasberbuer, 2024-02-03:
It seems that there is an issue with the recursive cloning.
You can manually checkout to the right version of open62541 v1.3.9.

Before, please verify the checked out version:
```shell
cd 3rdparty/open62541
git status
```

On my machine, the output is:

> PS C:\Dev\open62541pp\3rdparty\open62541> git status
> HEAD detached at v1.3.9
> nothing to commit, working tree clean

Then manually checkout v1.3.9:
```shell
cd 3rdparty/open62541
git checkout v1.3.9
git submodule update --init --recursive
```

And try again to compile...

Stoffels, 2024-02-03:
Thank you Sir

lukasberbuer, 2024-02-03:
Did it work?

lukasberbuer, 2024-02-03:
Would be great if you can show your `git status` output for further analysis / understanding.

Stoffels, 2024-02-03:
I run the git status and the output is:

***@***.***:~/open62541pp/3rdparty/open62541$ git status
HEAD detached at 70ff3501d
nothing to commit, working tree clean
***@***.***:~/open62541pp/3rdparty/open62541$ git checkout v1.3.9
HEAD is now at 70ff3501d refactor(build): Bump version to v1.3.9

***@***.***:~/open62541pp/3rdparty/open62541$ git submodule update --init
--recursive
***@***.***:~/open62541pp/3rdparty/open62541$ git status
HEAD detached at 70ff3501d
nothing to commit, working tree clean

I

Stoffels, 2024-02-03:
No, i did reply with all the information you requested...sorry, just did
not press the send button.

My apologies
Herman

lukasberbuer, 2024-02-05:
So it did not work?
Can you check the `client.h` file again as described here: https://github.com/open62541pp/open62541pp/issues/182#issuecomment-1925202924

Can you try to checkout another open62541 version, e.g. `git checkout v1.3.8` and check the file again?

Stoffels, 2024-02-05:
Ok, i got it right. I did all from the start and this time i made sure that 3rdparty/open62541/include/open62541/client.h shows the "logger" and not "logging" variable.  I somehow messed up, and it don't know what i did wrong. But once the correct open62541 library was there, everything compiled correctly. Thanks for the help, i would have struggled to sort this out without your help.

lukasberbuer, 2024-02-05:
Glad it worked @Stoffels :)

closed 2024-02-05 by lukasberbuer