open62541pp/open62541pp #193: Public `Result` class encapsulating a result value and a status code

lukasberbuer, 2024-02-13:
A `detail::Result` class was introduced in #111 to handle the results of requests and discussed with @chandryan

The design is inspired by:
- Rust's `Result`: https://doc.rust-lang.org/std/result/
- C++ 23's `std::expected`: https://en.cppreference.com/w/cpp/utility/expected
- Swift's `Result`: https://developer.apple.com/documentation/swift/result
- Boost Outcome's `result`: https://boostorg.github.io/outcome/tutorial/essential/result.html

This could be a new way of handling exceptions / bad status codes in a more explicit manner.
The `Result` class should be battle-tested internally before making it public.

@chandryan, would you like to work on the design?

chandryan, 2024-02-16:
> A `detail::Result` class was introduced in #111 to handle the results of requests and discussed with @chandryan
>
> The design is inspired by:
>
> * Rust's `Result`: https://doc.rust-lang.org/std/result/ [...]

Yes I would like to 👍

chandryan, 2024-02-21:
Thanks for the link, that's some interesting values :) Though I think that link considers including only `<optional>` by itself and nothing else, right?

I conducted a test to check the LOC difference of `src/Server.cpp` as a pre-processed file by using GCC's `-save-temps` and checking the generated `src/Server.cpp.ii`:
- without `<optional>`: 126880 LOC
- with `<optional>`: 126881 LOC

Apparently `<optional>` was already included 😄 . I see it in some headers, for example in `Variant.h` and `DataValue.h`. 
I did an additional test with an empty file including only Result.h with and without the additional `<optional>`:
- without `<optional>`: 87130 LOC
- with `<optional>`: 88370 LOC

So in a best case including `<optional>` doesn't make a difference LOC, worst case it leads to ~1200 more, which comes close to the un-preprocessed `<optional>` with ~1300 LOC. And it would just be about 1.3% more LOC for a vanilla inclusion of `Result.h` without additional header files. It's probably not a measurable impact on compile time :-)

---

My idea to get around using `<optional>` was the following:
Use some custom implementation of a static memory wrapper with std::aligned_storage, placement new & co to re-implement what we could get from `<optional>`, just without an internal boolean to determine whether the `<optional>` has a value. We could replace the existence of this boolean by a check like `if (!code().isBad())`.

However, that comes with the cost of having to re-invent most of what `<optional>` already provides for us. In the end we would probably end up to some LOC close to what the STL has come up with, while the STL version is thoroughly tested and proven. That's why I would prefer to use `<optional>` and spend the additional boolean.

lukasberbuer, 2024-02-22:
Ok, got it LOC if not an argument 👍

Maybe I don't have the picture where you are planning to go. You still need to store `std::optional<T>` and `StatusCode`, correct? And you want to use `std::optional<T>` to explicitly mark, that a value is provided?

chandryan, 2024-02-22:
Yes that part I wanted to clarify with you, I would keep the StatusCode and replace `T` with `std::optional<T>` 👍

lukasberbuer, 2024-02-22:
I'm fine with the `std::optional<T>` approach 👍

closed 2024-03-21 by lukasberbuer