open62541pp/open62541pp #194: Problems with clang-tidy

messi1, 2024-02-15:
Hi everyone.
in the clang-tidy file you have permantently activated WarningAsErrors.
This cause troubles when building.
cppcoreguidelines-avoid-const-or-ref-data-members and misc-use-anonymous-namespace which are implicitly activated are complaining because of references as class members and static functions which are not included in nameless namespaces.

Also in your CMakeLists.txt you added the option WarningAsErrors.

I would recommand to change the following:
-WarningsAsErrors: "*"
+WarningsAsErrors: ""

And adding
-  cppcoreguidelines*, -cppcoreguidelines-avoid-magic-numbers,
+  cppcoreguidelines*, -cppcoreguidelines-avoid-magic-numbers, -cppcoreguidelines-avoid-const-or-ref-data-members,
   misc*,
+  -misc-use-anonymous-namespace

Then fixing the problems one by one.

I would also recommand to set the default of the CMakeLists.txt WarningAsErrors to ON.

If you like I  can prepare a pull request.

lukasberbuer, 2024-02-15:
Hi @messi1, totally agree. I had the issue recently as well by switching between operating systems and clang-tidy versions. Newer clang-tidy versions include more rules which might lead to new warnings/errors.

But I also would like to keep clang-tidy in the CI pipeline and let it fail, if there are any warnings: https://github.com/open62541pp/open62541pp/blob/master/.github/workflows/ci.yml#L26
The clang-tidy version in the CI pipeline is fixed (Ubuntu 22.04 -> clang-tidy 15.0.7).

How can we achieve both, warnings during compilation and errors in the CI pipeline?

Please provide a PR, that would be awesome 👍

messi1, 2024-02-15:
Of course clang-tidy should stay in the pipeline.
I would remove the WarningAsError from clang-tidy and would handle it via cmake.

Further up in the yml file you have the cmake call:
          cmake -G Ninja \
            -DCMAKE_BUILD_TYPE=${{ matrix.build_type }} \
            -DUA_ENABLE_ENCRYPTION=OPENSSL \
            -DUAPP_BUILD_TESTS=ON \
            -DUAPP_BUILD_EXAMPLES=ON \
            **-DUA_FORCE_WERROR=ON \**
            ${{ matrix.config.flags }} 

This should solve the problem. You can enable/disable WarningAsError via cmake command line and it is alwasy active via build pipeline.

I also recommand to upgrade to clang-16.
I will create a PR on my fork with my suggestion.

lukasberbuer, 2024-02-15:
Sounds good 👍
The `UA_FORCE_WERROR` comes from open62541. I would prefer to add similar option for open62541pp, e.g. `UAPP_WARNINGS_AS_ERRORS`.

messi1, 2024-02-15:
I'm so stupid. Grr.
There is no possebility to enable/disable clang-tidy "warning as error" via cmake.
We have to exclude the clang-tidy checks or we fix the code.
clang-tidy 15.0.7 is the last supported version.
I will test the project with this version.

lbckmnn, 2024-02-15:
We could also invoke clang tidy separately in CI using `run-clang-tidy`. 
That's what I do normally.

lukasberbuer, 2024-02-16:
`clang-tidy` can be invoked with the command line argument `--warnings-as-errors=*`.
So we can remove `WarningsAsErrors: "*"` from the `.clang-tidy` and optionally treat warnings as errors by passing the CLI argument. I sketched the solution in #198.

lukasberbuer, 2024-02-20:
The option `UAPP_WARNINGS_AS_ERRORS` is now merged into master and activated in the CI workflow to enforce the no warnings policy.

Nevertheless, fixes for newer clang-tidy versions should be integrated in open62541pp. Please feel free to create issues and PRs.

lukasberbuer, 2024-02-22:
@messi1 does this fix it for you?

closed 2024-03-11 by lukasberbuer