open62541pp/open62541pp #197: Coding style

messi1, 2024-02-15:
I have a question, why do you use "get" as prefix for member getter methods.
Last time I used that style was in 2010. Nowadays you write as an example:

class Server
{
  public: 
      Server(...);
      Server::Status status();
      void setStatus( Server::Status status ); // Depending on the obj you can also pass it as const ref 
   private:
       Server::Status status_; 
}

Is there a certain reason to write the "get" prefix?

lukasberbuer, 2024-02-15:
I have to wrap a lot of C structs, for example `UA_DataValue`:

```c
typedef struct {
    UA_Variant    value;
    UA_DateTime   sourceTimestamp;
    UA_DateTime   serverTimestamp;
    UA_UInt16     sourcePicoseconds;
    UA_UInt16     serverPicoseconds;
    UA_StatusCode status;
    UA_Boolean    hasValue             : 1;
    UA_Boolean    hasStatus            : 1;
    UA_Boolean    hasSourceTimestamp   : 1;
    UA_Boolean    hasServerTimestamp   : 1;
    UA_Boolean    hasSourcePicoseconds : 1;
    UA_Boolean    hasServerPicoseconds : 1;
} UA_DataValue;
```

Some member variables could be accessed without any problems (integral types). The `value` member variable should be accessed as an wrapped `opcua::Variant`, so there is a need for a getter and setter method.

```cpp
class DataValue : public TypeWrapper<UA_DataValue, UA_TYPES_DATAVALUE> {
public:
    Variant& getValue() & noexcept;
    const Variant& getValue() const& noexcept;
    Variant&& getValue() && noexcept;
    const Variant&& getValue() const&& noexcept;
    DateTime getSourceTimestamp() const noexcept;
    DateTime getServerTimestamp() const noexcept;
    uint16_t getSourcePicoseconds() const noexcept;
    uint16_t getServerPicoseconds() const noexcept;
    StatusCode getStatus() const noexcept;

    void setValue(const Variant& value);
    void setValue(Variant&& value) noexcept;
    void setSourceTimestamp(DateTime sourceTimestamp) noexcept;  // NOLINT
    void setServerTimestamp(DateTime serverTimestamp) noexcept;  // NOLINT
    void setSourcePicoseconds(uint16_t sourcePicoseconds) noexcept;
    void setServerPicoseconds(uint16_t serverPicoseconds) noexcept;
    void setStatus(StatusCode status) noexcept;
};
```

I was also thinking about removing the prefixes `get` and `set`. In my opinion, it's too ambiguous to have overloaded member function which may act as setters or getters:

```cpp
class DataValue : public TypeWrapper<UA_DataValue, UA_TYPES_DATAVALUE> {
    // ...
    // getter
    Variant& value() & noexcept;
    const Variant& value() const& noexcept;
    Variant&& value() && noexcept;
    const Variant&& value() const&& noexcept;
    // setter
    void value(const Variant& value);
    void value(Variant&& value) noexcept;
    // ...
};
```

How would you solve this @messi1?

messi1, 2024-02-15:
I would keep the "set" prefix for distingushment between getter and setter.
All frameworks which I use professionaly follow the same pattern.

I think we have to take one step after the other.
A friend of mine also wrote a C++ wrapper for the open62541 for a customer. I will ask him how he solved it.