open62541pp/open62541pp #206: clangd has trouble finding compile_commands.json with new CMakePresets.json

chandryan, 2024-02-27:
Hi,

I'm using clangd for code completion and have stumbled across the following issue since the introduction of the new CMakePresets.json:
clangd is not able to find the compile_commands.json anymore. It expects it to be under `build/compile_commands.json`, but the new configuration puts it to `build/${presetName}/compile_commands.json`. I couldn't find a way to dynamically forward the path to compile_commands.json.

The easiest way to resolve this issue would be to always configure the binaryDir to `${sourceDir}/build` instead of `${sourceDir}/build/${presetName}`. Would this be a feasible solution or do you see the need to keep multiple configure presets at the same time and be able to switch between them easily? Or maybe you see another solution?

Kind regards

lukasberbuer, 2024-02-29:
Hi @chandryan,

I personally use different presets actively during development. I like to have a pretty fast build during development and additional checks and sanitizers on demand. This can be easily achieved by switching configure presets. If they share the same build directory, you have to reconfigure and rebuild everything from scratch.

May I propose you an alternative? You have the option to define your own `CMakeUserPresets.json` to inherit and overwrite settings, e.g.

```json
{
    "version": 3,
    "cmakeMinimumRequired": {
        "major": 3,
        "minor": 21,
        "patch": 0
    },
    "configurePresets": [
        {
            "name": "my-dev-preset",
            "inherits": [
                "dev",
                "examples",
                "doc"
            ],
            "binaryDir": "${sourceDir}/build"
        }
    ]
}
```

Is this an option for you?

Btw., I don't understand that you select a preset but the build directory is not propagated to clangd. Which IDE are you using?

chandryan, 2024-03-05:
Hi @lukasberbuer ,

thanks for the tip with CMakeUserPresets.json, yes that one is a good option :-)

I'm using VS Code, I assume it does not fully integrate with 3rd party code completion from clangd. There is a setting that can be used to tell clangd where to find the compile commands directory as an argument to clangd, but that field does not seem to support macro expansion. So an argument like `-compile-commands-dir=${workspaceFolder}/build/${configurePresets.binaryDir}` does not get resolved to the actual path.

Are you using VS Code with the standard code completion from Microsoft or some other IDE?

lukasberbuer, 2024-03-11:
Hi @chandryan,

I am using VS Code as well. The new CMake tools works really great in combination with `CMakePresets.json` / `CMakeUserPresets.json`. The `compile_commands.json` is automatically discovered in the build directory of the chosen preset.
Overall, I use following main extensions:
- https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools (comes with clang-tidy, clang-format, ...)
- https://marketplace.visualstudio.com/items?itemName=ms-vscode.cmake-tools
- https://marketplace.visualstudio.com/items?itemName=jeff-hykin.better-cpp-syntax

Following documents helped me to fine-tune the setup:
- https://github.com/microsoft/vscode-cmake-tools/blob/main/docs/cmake-presets.md
- https://github.com/microsoft/vscode-cmake-tools/blob/main/docs/debug-launch.md

closed 2024-03-11 by lukasberbuer