open62541pp/open62541pp #246: async sendRequest using open62541 v1.4.0

lms-hfr, 2024-03-21:
Hello,

I'm experimenting with the open62541 v1.4.0 . But I ran into a problem.

The new eventloop only works with async calls. So I would like to use the async send request call.

In Attribute.h the method

```
template <typename CompletionToken = DefaultCompletionToken>
auto readAttributeAsync(
```

calls `sendRequest()` .

There is a async and a sync overload implementation of this method.

The call `detail::sendRequest` uses the **sync** sendRequest method. 

I think this is because 
`DefaultCompletionToken()` is `struct UseFutureToken {};` from async.h, so it is an empty struct. That is the same as 
```
/// Completion token for sync client operations.
struct SyncOperation {};
```
from ClientService.h . So the overload method has the same signature.

I can 'fix' this by changing the call from:

```
return detail::sendRequest<UA_ReadRequest, UA_ReadResponse>(
        client,
        request,
        [](UA_ReadResponse& response) { return detail::getReadResult(response); },
        std::forward<CompletionToken>(token)
    );
```
to
```
detail::sendRequest<UA_ReadRequest, UA_ReadResponse, std::function<DataValue(UA_ReadResponse&)>, CompletionToken>(
        client,
        request,
        [](UA_ReadResponse& response) { return detail::getReadResult(response); },
        std::forward<CompletionToken>(token)
    );
```

But now `readAttributeAsync()` return a 'void' and this does not compile in `DataValue readAttribute<Client>(` from Attribute.cpp .

The return value of `void` actually comes from async.h `inline auto asyncInitiate(Initiation&& initiation, CompletionToken&& token, Args&&... args)`

I did try to use `readAttributeAsync()` the same way as is used in `../tests/Services.cpp TEST_CASE_TEMPLATE("Attribute service set write/read", T, Server, Client, Async<Client>) {`.

However here I would need a future as return value. 

Is this still in progress or do I miss something?

Thank you for your support.

greating Harald

lukasberbuer, 2024-03-21:
Hi Harald,

glad to see, that you are already using the newest features. The async model is currently only available in the master branched and will be published in the next release. Until then, there is some (minor) work to be done. @chandryan is working on the public `Result<T>` in #207 to encapsulate both the result and the status code, similar to `std::expected` or Rust's Result type.

The same is true for open62541 v1.4 support. I am currently working on this in a (private) branch.

May I ask, which features make you use the v1.4-rc version of open62541?
How is the async model working for you? Any remarks, ideas, feature requests are highly appreciated and might end up in the next release 😊

I would like to delay this issue until v1.4 is finally released and the API frozen. Or is this urgent for you?

lukasberbuer, 2024-03-21:
Btw., I was expecting, that this is an issue related to open62541 v1.4. Does it work with v1.3?

lukasberbuer, 2024-10-01:
The latest open62541 versions are now supported with [v0.15.0](https://github.com/open62541pp/open62541pp/releases/tag/v0.15.0).

closed 2024-10-01 by lukasberbuer

lms-hfr, 2024-10-02:
Exelent. Thank you.
I'll have a look at the new version.