open62541pp/open62541pp #277: ValueBackendDataSource example without lambdas

piperoc, 2024-04-19:
I would like to implement the read/write callbacks in separate functions instead of lambdas (helps with debugging and clutter).

In the open62541 this is easy, since they offer context arguments like SessionContext and NodeContext to pass user defined info to the callback.

I am not sure how to do this in open62541pp.

For example, in the [server datasource example](https://github.com/open62541pp/open62541pp/tree/master/examples/server_datasource.cpp) we have

```cpp
dataSource.read = [&](opcua::DataValue& value, const opcua::NumericRange&, bool timestamp) {
        // Increment counter before every read
        counter++;
        value.getValue().setScalar(counter);
        if (timestamp) {
            value.setSourceTimestamp(opcua::DateTime::now());
        }
        std::cout << "Read counter from data source: " << counter << "\n";
        return UA_STATUSCODE_GOOD;
    };
```

I would prefer to have something that can handle some extra info (like I would in open62541). 

```cpp

...

StatusCode onRead(DataValue& value, const NumericRange& range, bool timestamp, void * userContext)
{
   ...
}
...
dataSource.read = onRead;
...
```

Thank you for replying.

lukasberbuer, 2024-04-20:
Hi @piperoc,

I don't want to use `void*` pointers to pass user-defined data to callbacks.. it's very easy to shoot yourself in the foot. If you really want to use those, you can always use the open62541 functions within open62541pp.

Using capturing lambdas is the easiest way in my opinion. If you don't like or want to use lambdas, you can use any callable. If you need to pass custom data to those callables, functors may be a good fit. As an example, I rewrote the datasource example with functors:

```cpp
#include <iostream>

#include "open62541pp/open62541pp.h"

template <typename T>
class DataSourceOnRead {
public:
    DataSourceOnRead(T& value, opcua::NodeId id)
        : value_(value), id_(std::move(id)) {}

    opcua::StatusCode operator()(opcua::DataValue& dv, const opcua::NumericRange&, bool timestamp) {
        dv.getValue().setScalar(value_);
        if (timestamp) {
            dv.setSourceTimestamp(opcua::DateTime::now());
        }
        std::cout << "Read value from data source: " << value_ << "\n";
        return UA_STATUSCODE_GOOD;
    }

private:
    T& value_;
    opcua::NodeId id_;
};

template <typename T>
class DataSourceOnWrite {
public:
    DataSourceOnWrite(T& value, opcua::NodeId id)
        : value_(value), id_(std::move(id)) {}

    opcua::StatusCode operator()(const opcua::DataValue& dv, const opcua::NumericRange&) {
        value_ = dv.getValue().getScalarCopy<int>();
        std::cout << "Write value to data source: " << value_ << "\n";
        return UA_STATUSCODE_GOOD;
    }

private:
    T& value_;
    opcua::NodeId id_;
};

int main() {
    opcua::Server server(4840);

    // Counter variable as an example of a simple data source
    int counter = 0;

    // Add variable node
    opcua::NodeId id(1, 1000);
    auto nodeCounter = server.getObjectsNode().addVariable(
        id,
        "Counter",
        opcua::VariableAttributes{}
            .setAccessLevel(UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE)
            .setDataType<int>()
            .setValueScalar(counter)
    );

    // Define data source with its read and write callbacks
    opcua::ValueBackendDataSource dataSource;
    dataSource.read = DataSourceOnRead(counter, id);
    dataSource.write = DataSourceOnWrite(counter, id);

    // Define data source as variable node backend
    server.setVariableNodeValueBackend(nodeCounter.id(), dataSource);

    server.run();
}
```

piperoc, 2024-04-21:
Thank you @lukasberbuer, 
the functor solution works for my immediate needs.

I would have some questions on data source strategies, but those are for the discussion forum. 
This issue can be considered resolved.

I appreciated your reply during a weekend.

closed 2024-04-21 by piperoc