open62541pp/open62541pp #293: Plan for historical data

cuongkjm, 2024-05-27:
Hi, 
I want to use historical data feature on server side.
Do you have any plans for it already? or some code in a draft-branch perhaps

lukasberbuer, 2024-05-28:
Hi @cuongkjm, thanks for bringing that up. Is has been on the todo list quite a while now.
Currently, you can implement historical data access using open62541's C API, similar to the example here: https://github.com/open62541/open62541/blob/master/examples/tutorial_server_historicaldata.c

Following steps are needed to bring it to open62541pp:
1. Create wrapper base classes for
    - `UA_HistoryDataGathering`
    - `UA_HistoryDataBackend`
2. Function to set historizing backend `HistoryDataGathering` and settings (`Server` member function?)
3. Function to register nodes for historizing (`Server` member function?)

Do you have any special requirements?

Onwcan, 2026-09-11:
Hi @lukasberbuer , I've opened a PR for a first step here.

It covers points 2 and 3 of your plan — installing open62541's default history
database and registering nodes for historizing with the in-memory backend — so
historical data access is usable from C++ without dropping to the C API:

```cpp
Server server;
useHistoryDatabase(server);
historizeNode(server, variableId);