open62541pp/open62541pp #298: Subscription Issue After Re-connection

Fish0403, 2024-07-04:
Hi,
I have some question about the sample file client_subscription.cpp:
1. In each iteration of the loop, calling client.connect each time, wouldn't this cause any issues?
2. Should subscriptions be established after client.connect and before client.run?
3. I've seen issue 51, which suggests that disconnecting might delete subscriptions. I'm not sure if I understood it correctly. However, after reconnecting and subscribing again, I encounter errors when I run client.run. If I don't subscribe, I don't receive notifications when values change.

Could you please review the code snippet and provide any insights or suggestions on how to properly implement automatic reconnection and subscription？
```
#include <iostream>
#include <thread>
#include "open62541pp/open62541pp.h"

opcua::Client client;
std::string_view endpointUrl = "opc.tcp://W4320-YRJ:4840/";
bool isConnected = false;

void  MyOpcNotificationCallback(uint32_t subId, uint32_t monId, const opcua::DataValue& value) {
    const opcua::MonitoredItem item(client, subId, monId);
    int32_t intValue = value.getValue().getScalar<int32_t>();
    std::cout << "Received value: " << intValue << std::endl;
}

void subscribeToVariable(opcua::Client& client, const opcua::NodeId& nodeId, opcua::AttributeId attributeId, opcua::DataChangeNotificationCallback callback) {
    std::cout << "----------subscribeToVariable-------------" << std::endl;
    auto sub = client.createSubscription();
    opcua::SubscriptionParameters subscriptionParameters{};
    subscriptionParameters.publishingInterval = 1000;
    sub.setSubscriptionParameters(subscriptionParameters);
    sub.setPublishingMode(true);

    auto mon = sub.subscribeDataChange(
        nodeId,
        attributeId,
        callback
    );

    opcua::MonitoringParametersEx monitoringParameters{};
    monitoringParameters.samplingInterval = 100;
    mon.setMonitoringParameters(monitoringParameters);
    mon.setMonitoringMode(opcua::MonitoringMode::Reporting);
}

void disconnectedCallback() {
    std::cout << "----------disconnectedCallback-------------"<< std::endl;
    isConnected = false;

}
void connectedCallback() {
    std::cout << "----------connectedCallback-------------"<< std::endl;
    isConnected = true;
}

int main() {

    opcua::NodeId id = { 1, "TheAnswer" };
    client.onDisconnected(disconnectedCallback);
    client.onConnected(connectedCallback);

    while (true) {
        try {
            if (!isConnected) {
                client.connect(endpointUrl);
                subscribeToVariable(client, id, opcua::AttributeId::Value, &MyOpcNotificationCallback);
            }
            client.run();
        }
        catch (const opcua::BadStatus& e) {
            client.disconnect(); 
            std::cout << "Error: " << e.what() << "\nRetry to connect in 3 seconds\n";
            std::this_thread::sleep_for(std::chrono::seconds(3));
        }
        Sleep(10000);
    }
}

```

lukasberbuer, 2024-07-08:
Hi @Fish0403, did you resolve the issue? Could you share some information?

Fish0403, 2024-07-10:
Hi @lukasberbuer 
Sleep for a while between connecting and subscribing, then there won't be any errors.
I encountered a new issue yesterday. After connecting to the PLC, I received the following warning:
‘‘‘
----------subscribeToVariable-------------
Received value: 55
[2024-07-10 08:39:13.924 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:39:13.924 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:39:13.924 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:39:13.924 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:39:13.925 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:39:13.925 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:39:13.925 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:39:13.925 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:39:13.925 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:39:13.926 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:39:13.926 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:39:13.926 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:39:13.927 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:39:13.927 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:39:13.927 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:40:14.075 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:40:14.075 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:40:14.076 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:40:14.077 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:40:14.077 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:40:14.077 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:40:14.078 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:40:14.078 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:40:14.078 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:40:14.079 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:40:14.079 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:40:14.079 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:40:14.080 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:40:14.080 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:40:14.080 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:41:14.211 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:41:14.211 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:41:14.212 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:41:14.213 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:41:14.213 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:41:14.213 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:41:14.214 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:41:14.214 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:41:14.214 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:41:14.215 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:41:14.215 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:41:14.215 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:41:14.216 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:41:14.216 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:41:14.216 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:42:14.349 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:42:14.356 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:42:14.357 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:42:14.358 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:42:14.358 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:42:14.358 (UTC+0800)] warn/client        Received Timeout for Publish Response
[2024-07-10 08:42:14.359 (UTC+0800)] info/client        Received a ServiceFault response
[2024-07-10 08:42:14.359 (UTC+0800)] info/client        The ServiceResult has the StatusCode BadTimeout
[2024-07-10 08:42:14.359 (UTC+0800)] warn/client        Received Timeout for Publish Response
[... cut ...]
[2024-07-10 08:52:37.430 (UTC+0800)] info/channel       Connection 856 | SecureChannel 15545 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-07-10 08:52:37.430 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
----------connectedCallback-------------
[2024-07-10 08:52:37.437 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
’’’
Could you please review the warn and provide any insights or suggestions？

Fish0403, 2024-07-10:
```
----------connectedCallback-------------
[2024-07-10 09:47:03.952 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2024-07-10 09:47:18.904 (UTC+0800)] warn/channel       Connection 988 | SecureChannel 15585 | Receiving the response failed with StatusCode BadConnectionClosed
[2024-07-10 09:47:18.904 (UTC+0800)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
[2024-07-10 09:47:18.906 (UTC+0800)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
----------disconnectedCallback-------------
[2024-07-10 09:47:18.927 (UTC+0800)] info/channel       Connection 996 | SecureChannel 15586 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-07-10 09:47:18.928 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
----------connectedCallback-------------
[2024-07-10 09:47:18.938 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2024-07-10 09:47:33.919 (UTC+0800)] warn/channel       Connection 996 | SecureChannel 15586 | Receiving the response failed with StatusCode BadConnectionClosed
[2024-07-10 09:47:33.919 (UTC+0800)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
[2024-07-10 09:47:33.920 (UTC+0800)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
----------disconnectedCallback-------------
[2024-07-10 09:47:33.947 (UTC+0800)] info/channel       Connection 980 | SecureChannel 15587 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-07-10 09:47:33.948 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
----------connectedCallback-------------
[2024-07-10 09:47:33.955 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2024-07-10 09:47:48.942 (UTC+0800)] warn/channel       Connection 980 | SecureChannel 15587 | Receiving the response failed with StatusCode BadConnectionClosed
[2024-07-10 09:47:48.943 (UTC+0800)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
[2024-07-10 09:47:48.943 (UTC+0800)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
----------disconnectedCallback-------------
[2024-07-10 09:47:48.964 (UTC+0800)] info/channel       Connection 968 | SecureChannel 15588 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-07-10 09:47:48.964 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
----------connectedCallback-------------
[2024-07-10 09:47:48.971 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2024-07-10 09:48:03.949 (UTC+0800)] warn/channel       Connection 968 | SecureChannel 15588 | Receiving the response failed with StatusCode BadConnectionClosed
[2024-07-10 09:48:03.949 (UTC+0800)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed
[2024-07-10 09:48:03.950 (UTC+0800)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good
----------disconnectedCallback-------------
[2024-07-10 09:48:03.993 (UTC+0800)] info/channel       Connection 1008 | SecureChannel 15589 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-07-10 09:48:03.993 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
----------connectedCallback-------------
[2024-07-10 09:48:04.001 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
```

Fish0403, 2024-07-10:
[image]

lukasberbuer, 2024-07-18:
Hi @Fish0403, do be honest, I never experienced the `Received a ServiceFault response`. The timeout status code within the response indicates, that the request timed out, maybe due to server connection issues? You could try to adjust the timeout in the client config: https://www.open62541.org/doc/master/client.html#client-configuration

I will try to understand the problem the next days, I'm quite busy at the moment.
By the way, why did you choose not to use the `onSessionActivated` callback kat shown in the [`client_subscription.cpp` example](https://github.com/open62541pp/open62541pp/blob/master/examples/client_subscription.cpp).

lukasberbuer, 2024-10-08:
Hi @Fish0403, do you have any updates on this issue? Did you find a solution?