open62541pp/open62541pp #299: writeValue cause reconnection

matunbaman, 2024-07-15:
Hello @lukasberbuer :
First of all thank you very much for your work on this great lib.
I'm initialize a client with subscribes some variables
```
try {
		this->client.run();
	}catch(...){
	}
	this->client.stop();
	std::this_thread::sleep_for(std::chrono::milliseconds(1000));//workaround
	this->client.disconnect();
```
when I am trying to write some scalar value to server. Like this procedure:
```
if(!this->client.isConnected())
            {
                nRet = ICAM_E_ACCESS_DENIED;
                break;
            }
	    opcua::NamespaceIndex ns_id = this->op_ns;
            opcua::NodeId id = {ns_id, NodeStr};
            opcua::Variant var;
            var.setScalarCopy<int32_t>(value);
            auto result = opcua::services::writeValue(this->client, id, var);
```
the value is updated on server side but this procedure It cause reconnection. 

```
[2024-07-15 15:39:55.494 (UTC+0800)] warn/channel       Connection 2584 | SecureChannel 64 | Receiving the response failed with StatusCode BadConnectionClosed                                                                                  [2024-07-15 15:39:55.494 (UTC+0800)] warn/client        Received Publish Response with code BadSecureChannelClosed      [2024-07-15 15:39:55.494 (UTC+0800)] warn/client        Received Publish Response with code BadSecureChannelClosed      [2024-07-15 15:39:55.494 (UTC+0800)] warn/client        Received Publish Response with code BadSecureChannelClosed      [2024-07-15 15:39:55.495 (UTC+0800)] warn/client        Received Publish Response with code BadSecureChannelClosed      [2024-07-15 15:39:55.495 (UTC+0800)] warn/client        Received Publish Response with code BadSecureChannelClosed      [2024-07-15 15:39:55.495 (UTC+0800)] warn/client        Received Publish Response with code BadSecureChannelClosed      [2024-07-15 15:39:55.495 (UTC+0800)] warn/client        Received Publish Response with code BadSecureChannelClosed      [2024-07-15 15:39:55.495 (UTC+0800)] warn/client        Received Publish Response with code BadSecureChannelClosed      [2024-07-15 15:39:55.496 (UTC+0800)] warn/client        Received Publish Response with code BadSecureChannelClosed      [2024-07-15 15:39:55.496 (UTC+0800)] warn/client        Received Publish Response with code BadSecureChannelClosed      [2024-07-15 15:39:55.496 (UTC+0800)] warn/channel       Connection 0 | SecureChannel 0 | Could not receive with StatusCode BadConnectionClosed                                                                                                  [2024-07-15 15:39:55.496 (UTC+0800)] info/client        Client Status: ChannelState: Closed, SessionState: Created, ConnectStatus: Good                                                                                                         [INF]15:39:55.485339 handler_opcua_write.cpp:56-->here2                                                                 [2024-07-15 15:39:55.587 (UTC+0800)] info/channel       Connection 2584 | SecureChannel 66 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s                         [2024-07-15 15:39:55.588 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good                                                                                                           [2024-07-15 15:39:55.614 (UTC+0800)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good                                                                                                         [INF]15:39:55.615169 handler_opcua.cpp:165-->session connected
```

Can you please point out my mistake or help to fix this issue? Thank you.

lukasberbuer, 2024-07-18:
Hi @matunbaman, where do you execute the second part of the code, writing to the variable node? Is it running in a different thread?

matunbaman, 2024-07-26:
Hi, @lukasberbuer,  yes, because the main thread blocking in this content

```
this->client.run();
```

lukasberbuer, 2024-07-26:
Ok, this seems to be a threading issue then. The client API in open62541 version 1.3 is not (fully) thread-safe. This results in sporadic errors like described in https://github.com/open62541/open62541/issues/3875.

We discussed a solution with locks in this issue: https://github.com/open62541pp/open62541pp/issues/100#issuecomment-1696810299
Please let me know if this solution works in your case.

closed 2024-09-15 by lukasberbuer