open62541pp/open62541pp #369: Provide a unified API for reading and writing values of nodes

chandryan, 2024-10-12:
Hi @lukasberbuer !

I would like to contribute to open62541pp with a simplified API proposal for reading and writing node values 😄 . 

I set up some demo code, please take a look at https://github.com/chandryan/open62541pp/pull/1 and tell me if you're interested in it. If yes, I will change the the PR to point to your repo and continue my work.

Kind regards

lukasberbuer, 2024-10-14:
Great idea @chandryan! This could also result in a universal `Variant` constructor instead of `Variant::fromValue`.
As I noticed in your code, the main problem is to differentiate between a container and a string-like type. I can't think of a generic trait, that could separate for example a `Span<int8_t>` from a `std::string_view`.
BUT we might use the `TypeRegistry`/`TypeConverter` with their traits `detail::IsRegisteredType`/`detail::IsConvertibleType`: If the provided template type is a registered or convertible type, is has to be a scalar type. There are already template specializations for string-like types:
- `opcua::String`
- `std::string_view`
- `std::string`
- `const char*`
- `char[N]`
- `opcua::ByteString`

I'm looking forward to see the results 😊

chandryan, 2024-10-19:
Great! I will try it out if I can make it work :) Thanks for the hint with `detail::IsConvertibleType`, I will see if I can use that to get rid of the explicit check for STL containers 😄

closed 2024-12-06 by lukasberbuer