open62541pp/open62541pp #387: Problem with compiling from source code v0.15.0

wc188996, 2024-10-23:
I have met a error when compile the open62541pp v0.15.0  source code cloned from git. the os is opensuse 15.6 and GCC/G++ version is 13.3.0  .
`/opt/open62541pp/include/open62541pp/types/Variant.h: In instantiation of ‘static opcua::Variant opcua::Variant::fromArray(ArrayLike&&) [with opcua::VariantPolicy Policy = (opcua::VariantPolicy)0; ArrayLike = opcua::Span<const unsigned int>&]’:
/opt/open62541pp/include/open62541pp/types/DataValue.h:64:52:   required from ‘static opcua::DataValue opcua::DataValue::fromArray(Args&& ...) [with opcua::VariantPolicy Policy = (opcua::VariantPolicy)0; Args = {opcua::Span<const unsigned int>&}]’
/opt/open62541pp/include/open62541pp/services/detail/AttributeHandler.h:182:85:   required from here
/opt/open62541pp/include/open62541pp/types/Variant.h:75:30: error: class template argument deduction failed:
             Handler::setArray(var, Span{std::forward<ArrayLike>(array)});
             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/open62541pp/include/open62541pp/types/Variant.h:75:30: error: no matching function for call to ‘Span(<brace-enclosed initializer list>)’
In file included from /opt/open62541pp/include/open62541pp/Client.h:12:0,
                 from /opt/open62541pp/src/Client.cpp:1:
/opt/open62541pp/include/open62541pp/Span.h:204:1: note: candidate: template<class Container> opcua::Span(Container&)-> opcua::Span<typename Container::value_type>
 Span(Container&) -> Span<typename Container::value_type>;
 ^~~~`

the error msg is `error: class template argument deduction failed:
             Handler::setArray(var, Span{std::forward<ArrayLike>(array)});
             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/open62541pp/include/open62541pp/types/Variant.h:75:30: error: no matching function for call to ‘Span(<brace-enclosed initializer list>)’`
and the compile will be canceled!

lukasberbuer, 2024-10-23:
Hi @wc188996,
does this error also occur when you compile open62541pp standalone? GCC 13 on Linux is actually tested in the CI pipelines.
It looks like this error is produced by some call to `Node::write*`. Can you provide the snippet of your code or the full compile output? The actual cause of the template instantiation seems to be missing in this output.
Btw it seems that you are not using v0.15.0 because the header and source structure changed (`Variant.h` is now `types.hpp`, ...).

wc188996, 2024-10-23:
Hi lukasberbuer, I update the error msg in issue.
I've made a mistake  that the version is v0.14.0  ,I changed the version . the version is v0.15.0 now.
Thank you  for your reply!

lukasberbuer, 2024-10-23:
You seem to compile with GCC 7.5:
```
cmake -DCMAKE_BUILD_TYPE=Release -DUAPP_BUILD_EXAMPLES=OFF -DUAPP_BUILD_TESTS=OFF ..
-- The CXX compiler identification is GNU 7.5.0
```

Could you please test a newer compiler with full C++17 support?

wc188996, 2024-10-25:
Thank you!   I have installed both gcc/g++ 7 and gcc/g++13 ,and forgot to  check the gcc env before compiling.
I set the CC and  CXX env ,the default path is /usr/bin and /usr/bin/gcc & g++ version is 7.5.0.
 export CXX=/usr/local/gcc13/bin/g++
 export CC=/usr/local/gcc13/bin/gcc

closed 2024-10-25 by wc188996