open62541pp/open62541pp #455: Missing subscription inactivity callback function

PandaBoby, 2024-11-22:
` auto subscription = _client.createSubscription();
   opcua::SubscriptionParameters subscriptionParameters{};
   subscriptionParameters.publishingInterval = 1000.0;
   subscription.setSubscriptionParameters(subscriptionParameters);
   subscription.setPublishingMode(true);`

When I create a subscription, only the subscription inactivity log is notified after a period of disconnection, I can't find where to configure the subscription inactivity callback function.

lukasberbuer, 2024-11-22:
It's a client-wide callback for a subscriptions defined the the client config:
https://github.com/open62541/open62541/blob/v1.3.14/include/open62541/client.h#L180-L186

We don't have the function exposed in `ClientConfig` currently.
But you can easily use the C API:
```cpp
client.config().handle()->subscriptionInactivityCallback = [](
    UA_Client* client,
    uint32_t subscriptionId,
    void* subContext
) {
    // ...
};
```

We should add `Client::onSubscriptionInactive` in the near future though. Do you want to create a PR?

PandaBoby, 2024-11-22:
Thank you for your reply, I will try to edit the code with the suggestions you provided. Regarding the invitation to create a PR, I apologize, I don't have a good understanding of OPCUA or even C++, I just know how to use it at this stage, and I can't provide a reliable PR for now.

PR #456 (2024-11-22): feat: Client::onSubscriptionInactive
Define a subscription inactivity callback via `Client::onSubscriptionInactive`.

The callback is called when the client does not receive a publish response after the defined delay of `(publishingInterval * maxKeepAliveCount) + UA_ClientConfig::timeout)`.

Closes #455.
closed 2024-11-22 by lukasberbuer