open62541pp/open62541pp #484: Where to specify the "ApplicationURI"?

Mebus, 2024-12-04:
Hallo, 

I am generating a certificate like this:

```
  char CLIENT_URI[] = "urn:mymachine:application";

  opcua::crypto::CreateCertificateResult client_cert = opcua::crypto::createCertificate(
      {
          opcua::String{"C=DE"},
          opcua::String{"O=SB"},
          opcua::String{"CN=client@mymachine"},
      },
      {
          opcua::String{"DNS:mymachine"},
          opcua::String{std::string("URI:").append(CLIENT_URI)},
      },
      opcua::crypto::CertificateFormat::DER);
```

However this results in this error:

```
The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
```

Where and how can I specify the ApplicationURI so that both match?

Mebus

lukasberbuer, 2024-12-04:
Hi @Mebus,
did you set the ApplicationURI of the `Client`? The functionality is currently not exposed in the C++ interface, but there is a simple workaround:
```cpp
opcua::Client client;

// with config() function since v0.16
asWrapper<opcua::ApplicationUri>(client.config().handle()->applicationUri) = opcua::String("...");

// with UA_Client_getConfig before v0.16
auto* config = UA_Client_getConfig(client.handle());
asWrapper<opcua::ApplicationUri>(config->applicationUri) = opcua::String("...");
```

Mebus, 2024-12-04:
Thanks! That seems to have changed something :-)

lukasberbuer, 2024-12-04:
Does it work?
PR #485 adds the missing `ClientConfig::setApplicationUri` function.

lukasberbuer, 2024-12-04:
Ups, I think it's the wrong setting:

```c
typedef struct {
    /* The description must be internally consistent.
     * - The ApplicationUri set in the ApplicationDescription must match the
     *   URI set in the certificate */
    UA_ApplicationDescription clientDescription;

    // ...

    /**
     * If the EndpointDescription has not been defined, the ApplicationURI
     * constrains the servers considered in the FindServers service and the
     * Endpoints considered in the GetEndpoints service.
     *
     * If empty the applicationURI is not used to filter.
     */
    UA_String applicationUri;

    // ...
} UA_ClientConfig;
```

So we need to change the `clientDescription.applicationUri`:
```cpp
auto* config = UA_Client_getConfig(client.handle());
opcua::asWrapper<opcua::String>(config->clientDescription.applicationUri) = opcua::String("...");
```

Mebus, 2024-12-06:
Hey @lukasberbuer,

for some reason it cannot connect to the Beckhoff PLC anymore.

Your first suggestion returns these messages in my program:

```
[2024-12-06 08:42:01.481 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2024-12-06 08:42:01.481 (UTC+0000)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2024-12-06 08:42:01.482 (UTC+0000)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2024-12-06 08:42:01.483 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
Trying to connect to the server.
[2024-12-06 08:42:01.483 (UTC+0000)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-12-06 08:42:01.488 (UTC+0000)] info/channel       Connection 4 | SecureChannel 3746718823 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-12-06 08:42:01.488 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-12-06 08:42:01.490 (UTC+0000)] warn/client        FindServers did not returned a suitable DiscoveryURL. Continue with the EndpointURL opc.tcp://OfficeTestBenchPLC:4840.
[2024-12-06 08:42:01.492 (UTC+0000)] error/client       No suitable endpoint found
[2024-12-06 08:42:01.492 (UTC+0000)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
Could not connect to server: BadInternalError
Trying to run the client.
Could not run the client: BadInternalError
[2024-12-06 08:42:01.492 (UTC+0000)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
````

Your second suggestion returns this:

```
[2024-12-06 08:45:08.840 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2024-12-06 08:45:08.840 (UTC+0000)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2024-12-06 08:45:08.841 (UTC+0000)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2024-12-06 08:45:08.842 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
Trying to connect to the server.
[2024-12-06 08:45:08.842 (UTC+0000)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-12-06 08:45:08.849 (UTC+0000)] info/channel       Connection 4 | SecureChannel 3746718826 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-12-06 08:45:08.849 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-12-06 08:45:08.851 (UTC+0000)] info/client        The initially defined EndpointURL opc.tcp://OfficeTestBenchPLC:4840is valid for the server
[2024-12-06 08:45:08.853 (UTC+0000)] info/client        Selected endpoint 0 in URL opc.tcp://OfficeTestBenchPLC:4840 with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2024-12-06 08:45:08.853 (UTC+0000)] info/client        Selected UserTokenPolicy UserName-[0]-SignAndEncrypt-Basic256Sha256 with UserTokenType UserName and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2024-12-06 08:45:08.853 (UTC+0000)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2024-12-06 08:45:08.853 (UTC+0000)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-12-06 08:45:08.858 (UTC+0000)] info/securitypolicy        The basic256sha256 security policy channel with openssl is created.
[2024-12-06 08:45:09.045 (UTC+0000)] warn/channel       Connection 4 | SecureChannel 0 | Receiving the response failed with StatusCode BadCertificateUseNotAllowed
[2024-12-06 08:45:09.046 (UTC+0000)] info/securitypolicy        The basic256sha256 security policy channel with openssl is deleted.
[2024-12-06 08:45:09.046 (UTC+0000)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadConnectionClosed
Could not connect to server: BadConnectionClosed
Trying to run the client.
Could not run the client: BadConnectionClosed
[2024-12-06 08:45:09.046 (UTC+0000)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
```

And also doesn't connect.

This is a snippet from the code:

```
 char CLIENT_URI[] = "urn:unconfigured:application";
  opcua::crypto::CreateCertificateResult client_cert = opcua::crypto::createCertificate(
      {
          opcua::String{"C=DE"},
          opcua::String{"O=SB"},
          opcua::String{"CN=client@mymachine"},
      },
      {
          opcua::String{"DNS:machine"},
          opcua::String{std::string("URI:").append(CLIENT_URI)},
      },
      opcua::crypto::CertificateFormat::DER);

  opcua::ClientConfig client_config(client_cert.certificate, client_cert.privateKey, {}, {});

  // client_config.setTimeout(1000);
[... cut ...]
  //opcua::asWrapper<opcua::String>(client.config().handle()->applicationUri) = opcua::String(CLIENT_URI);

  auto* config_handle = UA_Client_getConfig(client.handle());
  opcua::asWrapper<opcua::String>(config_handle->clientDescription.applicationUri) =  opcua::String(CLIENT_URI);

  std::cout << "Trying to connect to the server." << std::endl;

  bool success = false;

  try {
    client.connect(config.url);
    success = true;
  } catch (const opcua::BadStatus& e) {
    fmt::print("Could not connect to server: {}\n", e.what());
  }
```

Mebus

lukasberbuer, 2024-12-06:
I modified the test in #488 to reproduce the issue.
The second options works IF the server trusts the client certificate. This can be achieved by:
1. Adding the client *self-signed certificate* to the server's trust list (as done in #488).
2. Signing the certificates by a *certificate authority (CA)* to create *signed certificates*. This procedure is well described here (for server certificates): https://github.com/QUaServer/QUaServer?tab=readme-ov-file#create-certificates

Mebus, 2024-12-06:
Hey Lukas,

the Beckhoff OPC/UA server is set to "trust client certificates automatically":

[image]

Do I still have to add the certificate to the server in this case?

Mebus

Mebus, 2024-12-06:
I have added the security policy "None" to the server:

[image]

And now I do use this code:

``` 
  opcua::ClientConfig client_config;

  client_config.setSecurityMode(opcua::MessageSecurityMode::None);
  client_config.setUserIdentityToken(opcua::UserNameIdentityToken(config.username, config.password));

  opcua::Client client(std::move(client_config));

  std::cout << "Trying to connect to the server: " << std::endl;

  try {
    client.connect(config.url);
  } catch (const opcua::BadStatus& e) {
    fmt::print("Could not connect to server: {}\n", e.what());
  }

```

The result is this:

```
[2024-12-06 13:19:30.731 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2024-12-06 13:19:30.732 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
Trying to connect to the server:
[2024-12-06 13:19:30.738 (UTC+0000)] info/channel       Connection 4 | SecureChannel 1015544216 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-12-06 13:19:30.738 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-12-06 13:19:30.740 (UTC+0000)] info/client        The initially defined EndpointURL opc.tcp://OfficeTestBenchPLC:4840is valid for the server
[2024-12-06 13:19:30.744 (UTC+0000)] info/client        Rejecting endpoint 0: security mode doesn't match
[2024-12-06 13:19:30.745 (UTC+0000)] info/client        Rejecting endpoint 1: security mode doesn't match
[2024-12-06 13:19:30.745 (UTC+0000)] info/client        Rejecting endpoint 2: security mode doesn't match
[2024-12-06 13:19:30.745 (UTC+0000)] info/client        Rejecting UserTokenPolicy 0 in endpoint 3: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256' not available
[2024-12-06 13:19:30.745 (UTC+0000)] error/client       No suitable UserTokenPolicy found for the possible endpoints
[2024-12-06 13:19:30.745 (UTC+0000)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
Could not connect to server: BadInternalError
[2024-12-06 13:19:30.746 (UTC+0000)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
```

So even without the encryption I cannot connect to the server.

Is there something wrong with my source code?

lukasberbuer, 2024-12-06:
This might be some issue with the provided `UserNameIdentityToken`. The token provides a field `policyId`. I didn't encountered any server, that require this to be set. The standard says:
> An identifier for the UserTokenPolicy that the token conforms to.
> The UserTokenPolicy structure is defined in 7.42. Servers that provide a null or empty PolicyId shall accept null or empty and treat them as equal.
> -- https://reference.opcfoundation.org/Core/Part4/v105/docs/7.41.4

Please try to set it to the chosen security policy:
```cpp
opcua::UserNameIdentityToken token(config.username, config.password);
token.getPolicyId() = String("...");
```

The policy id's can be retrieved via the `GetEndpoints`, e.g. in UaExpert:
[image]

Mebus, 2024-12-06:
Hey,

that's most probably the value which you can get via getSecurityPolicyUri().

```
opcua::UserNameIdentityToken token(config.username, config.password);
token.getPolicyId() = opcua::String("username_basic256sha256");

client_config.setUserIdentityToken(token);
```

My Beckhoff OPC/UA server denies any authentication now. Thus I've tried it with the "Prosys OPC UA Simulation Server" and I get similar result:

```
[2024-12-06 17:16:44.515 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2024-12-06 17:16:44.516 (UTC+0000)] info/securitypolicy        The basic256sha256 security policy with openssl is added.
[2024-12-06 17:16:44.516 (UTC+0000)] warn/userland      Could not add SecurityPolicy#Basic256Sha256 with error code BadInvalidArgument
[2024-12-06 17:16:44.516 (UTC+0000)] info/securitypolicy        The Aes128Sha256RsaOaep security policy with openssl is added.
[2024-12-06 17:16:44.519 (UTC+0000)] warn/userland      Could not add SecurityPolicy#Aes128Sha256RsaOaep with error code BadInvalidArgument
[2024-12-06 17:16:44.519 (UTC+0000)] warn/userland      AcceptAll Certificate Verification. Any remote certificate will be accepted.
Trying to connect to the server.
[2024-12-06 17:16:44.519 (UTC+0000)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-12-06 17:16:44.527 (UTC+0000)] info/channel       Connection 4 | SecureChannel 21 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-12-06 17:16:44.527 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-12-06 17:16:44.528 (UTC+0000)] info/client        Use the EndpointURL opc.tcp://LACK:53530/OPCUA/SimulationServer returned from FindServers
[2024-12-06 17:16:44.528 (UTC+0000)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2024-12-06 17:16:44.528 (UTC+0000)] warn/client        The configured ApplicationURI does not match the URI specified in the certificate for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2024-12-06 17:16:44.533 (UTC+0000)] info/channel       Connection 4 | SecureChannel 22 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2024-12-06 17:16:44.533 (UTC+0000)] info/client        Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 0: security mode doesn't match
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 1: security policy not available
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 2: security policy not available
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 3: security mode doesn't match
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 4: security policy not available
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 5: security mode doesn't match
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 6: security policy not available
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 7: security mode doesn't match
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 8: security mode doesn't match
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 9: security policy not available
[2024-12-06 17:16:44.537 (UTC+0000)] info/client        Rejecting endpoint 10: security mode doesn't match
[2024-12-06 17:16:44.537 (UTC+0000)] error/client       No suitable endpoint found
[2024-12-06 17:16:44.538 (UTC+0000)] info/client        Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadInternalError
Could not connect to server: BadInternalError
```

Mebus