open62541pp/open62541pp #489: Alternative to `reference` tag in `Variant` constructor

lukasberbuer, 2024-12-06:
The merged PR #384 added `Variant` constructors to create variants from both scalar values and arrays without using `Variant::fromScalar`, `Variant::fromArray`.
Despite the value category (scalar/array), the value can be either copied to the variant (default) or referenced by it. A value is references by using the `ReferenceTag` as the first constructor argument.

Now as discussed with @chandryan, there is a mismatch between the tag name `reference` and the setter methods `setValue`/`setValueCopy`. The setter methods imply, that setting by reference if the default, the constructor on the other hand implies, that copy is the default.

```cpp
int value = 5;

opcua::Variant(value);  // copy
opcua::Variant(opcua::reference, value);  // reference

opcua::Variant var;
var.setValueCopy(value);  // copy
var.setValue(value);  // reference
```

### Inspiration

As a inspiration to alternative solutions, I looked at the STL. A `std::variant` always copies values. To avoid copies, the `std::reference_wrapper` class can be used:

```cpp
int value = 5;

std::variant<int> var(value);  // copy
std::variant<std::reference_wrapper<int>> var(std::ref(value));  // reference
```

The major difference between `std::variant` and `opcua::Variant` is, that `std::variant` supports a closed set of types provided as template parameters. `opcua::Variant` on the other hand supports an open set of types as long as they can be described with a `UA_DataType`.

### Idea / proposed solution

We could introduce something very similar to a `std::reference_wrapper` in open62541pp.
Let's call it `opcua::Reference` and add a free function `opcua::ref` to create a `Reference` object from an lvalue object:

```cpp
/// Reference wrapper to pass objects explicitly by reference.
template <typename T>
class Reference {
public:
    constexpr explicit Reference(T& value) noexcept : ptr_(&value) {}
    constexpr operator T&() const noexcept { return *ptr_; }
    constexpr T& get() noexcept { return *ptr_; }
private:
    T* ptr_;
};

template <typename T>
constexpr Reference<T> ref(T& value) {
    return Reference(value);
}
```

Now we could change the `Variant` interface to handle those reference types. Whenever a value as passed within a reference wrapper, it is stored as a reference. This approach can be used for both the constructors and the setter methods:

```cpp
int value = 5;

opcua::Variant(value);  // copy
opcua::Variant(opcua::ref(value));  // reference

opcua::Variant var;
var.setValue(value);  // copy
var.setValue(opcua::ref(value));  // reference
```

In my opinion, this would lead to a cleaner and more expressive interface.
Any ideas? What do you think @chandryan?

Remaining open questions if this gets implemented:
- How to name the `Reference` class? Is `Reference` a good name or shall we go with `ReferenceWrapper`
- Stick with the naming `opcua::ref`?
- We might also use `std::reference_wrapper`? Although this has a slightly different meaning in the STL.

chandryan, 2024-12-09:
> The merged PR #384 added `Variant` constructors to create variants from both scalar values and arrays without using `Variant::fromScalar`, `Variant::fromArray`. Despite the value category (scalar/array), the value can be either copied to the variant (default) or referenced by it. A value is referenced by using the `ReferenceTag` as the first constructor argument.
> 
> As discussed with @chandryan, there is a mismatch between the tag name `reference` and the setter methods `setValue`/`setValueCopy`. The setter methods imply, that setting by reference if the default, the constructor on the other hand implies, that copy is the default.
> 
> ```c++
> int value = 5;
> 
> opcua::Variant(value);  // copy
> opcua::Variant(opcua::reference, value);  // reference
> 
> opcua::Variant var;
> var.setValue(value);  // reference
> var.setValueCopy(value);  // copy
> ```
> 
> ### Inspiration
> As a inspiration to alternative solutions, I looked at the STL. A `std::variant` always copies values. To avoid copies, the `std::reference_wrapper` class can be used:
> 
> ```c++
> int value = 5;
> 
> std::variant<int> var(value);  // copy
> std::variant<std::reference_wrapper<int>> var(std::ref(value));  // reference
> ```
> 
> The major difference between `std::variant` and `opcua::Variant` is, that `std::variant` supports a closed set of types provided as template parameters. `opcua::Variant` on the other hand supports an open set of types as long as they can be described by a `UA_DataType`.
> 
> ### Idea / proposed solution
> We could introduce something very similar to a `std::reference_wrapper` in open62541pp. Let's call it `opcua::Reference` and add a free function `opcua::ref` to create a `Reference` object from an lvalue object:
> 
> ```c++
> /// Reference wrapper to pass objects explicitly by reference.
> template <typename T>
> class Reference {
> public:
>     constexpr explicit Reference(T& value) noexcept : ptr_(&value) {}
>     constexpr operator T&() const noexcept { return *ptr_; }
>     constexpr T& get() noexcept { return *ptr_; }
> private:
>     T* ptr_;
> };
> 
> template <typename T>
> constexpr Reference<T> ref(T& value) {
>     return Reference(value);
> }
> ```
> 
> Now we could change the `Variant` interface to handle those reference types. Whenever a value as passed within a reference wrapper, it is stored as a reference. This approach can be used for both the constructors and the setter methods:
> 
> ```c++
> int value = 5;
> 
> opcua::Variant(value);  // copy
> opcua::Variant(opcua::ref(value));  // reference
> 
> opcua::Variant var;
> var.setValue(value);  // copy
> var.setValue(opcua::ref(value));  // reference
> ```
> 
> In my opinion, this would lead to a cleaner and more expressive interface. Any ideas? What do you think @chandryan?
> 
> Remaining open questions if this gets implemented:
> 
> * How to name the `Reference` class? Is `Reference` a good name or shall we go with `ReferenceWrapper`?
> * Stick with the naming `opcua::ref`?
> * We might also use `std::reference_wrapper`? Although this has a slightly different meaning in the STL.

I think it's definitely more expressive than using a reference tag as additional parameter, though it's difficult for me to get my head around this usage because it's, as you already mentioned, conceptually quite different from the STL's reference wrapper. Even if there are dedicated classes for this use case in the `opcua` namespace, my mind keeps wanting to compare it to the STL. 

Your proposal gave me another idea, what do you think about using plain pointer syntax as an alternative?
We could also try to distinguish between setting by value and setting by reference in the following way:

```
int value = 5;

opcua::Variant(value);  // copy
opcua::Variant(&value));  // reference

opcua::Variant var;
var.setValue(value);  // copy
var.setValue(&value);  // reference
```

Kinda reminds me of Rust 😆

lukasberbuer, 2024-12-12:
The more I think about your proposal, the more I like it 😊
I just worked on a unified `Variant::assign` function in #496, that combines all set function `setValue*`, `setScalar*` and `setArray*` in simple overloads. The simple `Variant::assign` overloads with a single parameter could also be accessible via the assignment operator. Now it would be great if the assignment operator would work with the copy and reference version, which is simple if we use pointers for references:

```cpp
int value = 5;
opcua::Variant var;
var = value;   // copy
var = &value;  // reference
```

I really like the way the `Variant` interface has simplified is still evolving, thanks to your great contribution!
The universal constructor has been the building block for this 👍

closed 2024-12-12 by lukasberbuer