open62541pp/open62541pp #511: Robustness of test cases that use `runIterate`

chandryan, 2024-12-15:
Hi @lukasberbuer !

I investigated the triggering issue mentioned in https://github.com/open62541pp/open62541pp/issues/362 and have an interesting discovery to discuss:

I was able to reactivate the commented-out checks that fail with open62541 1.4 by calling the client's `runIterate` function multiple times. I cannot surely say *why* this is required, though I recalled that the event loop has been re-implemented in open62541 1.4 and had a suspicion.

From my point of view the best way to go forward within open62541pp is to not make assumptions that some expected event will happen in a specific call to a client's or server's call to the `runIterate` function. How the event loop within open62541 exactly works is an implementation detail, this could change anytime without notice.

What we can expect from open62541 is that some expected event happens within a certain amount of time. So instead of calling `runIterate` just once and checking fo a condition afterwards, in the tests we could call it multiple times until either the condition or a timeout occurred.

One potential drawback that I see are increased test runtimes, if this practice is also applied for checking for negative conditions: You would need to keep calling `runIterate` until the timeout occurs and set a big-enough timeout that respects latency spikes in CI pipelines to be on the safe side for assertions.
On the other hand, if we don't want to rely on implementation details of the event loop then this would be the only way to test properly: What if the event, that was expected to *not* happen, would happen right after the assertion in the next call to `runIterate`?

I will set up a draft Pull Request with some demo code that resolves the failing test cases as base for further discussions 😄

lukasberbuer, 2024-12-16:
I like your idea with the `runIterateUntil` helper function in #512. This decouples our tests nicely from the open62541 implementation details. In most cases, the tests execution time should not increase, because the condition will be `true` after the first iteration. Only for failing tests, the test duration might increase by the provided timeout. That's a good compromise in my opinion 👍

chandryan, 2024-12-16:
> I like your idea with the `runIterateUntil` helper function in #512. This decouples our tests nicely from the open62541 implementation details. In most cases, the tests execution time should not [...]

Great to hear :-)

Shall I create another Pull Request to apply this principle to all occurrences of the `runIterate` function in the tests, or do you prefer to only use it in case a (sporadic) test failure occurs?

lukasberbuer, 2024-12-16:
I applied this already to the critical tests. If the client just calls a raw service, one call to `runIterate` should be fine.
I wouldn't apply this to all tests, because it makes debugging a little harder:

```cpp
client.runIterate();
CHECK(notificationCount >= 1);  // will output the notificationCount (0 >= 0)

runIterateUntil(client, [&] { return notificationCount > 0; });  // will just output CHECK(false)
```