open62541pp/open62541pp #524: BadNodeIdInvalid when adding same name node to folder

GMellar, 2025-01-05:
I want to structure my nodes using folders where these folders may have variables with the same name as in other folders. When I do that I always end up with BadNodeIdInvalid exception when starting the server. 
```

#include <open62541pp/open62541pp.hpp>

int main(int, const char **) {
	opcua::BuildInfo serverBuildInfo(
		"Server",
		"Server",
		"Server",
		"",
		"",
		opcua::DateTime::now()
	);

	opcua::ServerConfig serverConfig;
	serverConfig.setApplicationName("Server");
	serverConfig.setProductUri("Some Product");
	serverConfig.setBuildInfo(serverBuildInfo);

	opcua::Server server(std::move(serverConfig));

	opcua::Node parentNode(server, opcua::ObjectId::ObjectsFolder);
	opcua::Node folder = parentNode.addFolder(
			{1, "folder"},
			"Folder",
			opcua::ObjectAttributes{});

	{
		opcua::Node var1 = parentNode.addVariable(
			{1, "var1"},
			"Variable 1",
			opcua::VariableAttributes{}
				.setDataType<bool>()
		);
		var1.writeValueScalar(false);
	}

	{
	    opcua::Node var1 = folder.addVariable(
	        {1, "var1"},
	        "Variable 1",
	        opcua::VariableAttributes{}
				.setDataType<bool>()
	    );
	    var1.writeValueScalar(false);
	}

	server.run();
}
```

lukasberbuer, 2025-01-15:
Node ids have to be unique. You are using `{1, "var1"}` multiple times.
You either find a logic to manually define the node ids uniquely or you can auto generate a unique node id with `{1, 0}`.

lukasberbuer, 2025-01-15:
Please comment or reopen if you have further questions.

closed 2025-01-15 by lukasberbuer